tsduck/tsduck #671: [Q] Any suggestions as to how I can speed up channel changes in my Windows tuner app

oviano, 2020-11-29:
So, currently my pseudo code is like this when tuning to a channel, or changing channels:

- initialise COM
- new ts::Tuner
- tuner_config.searchService(channel name)
- tuner_config.serviceToRuning(tuning params)
- tuner->tune()
- tuner->start()
- new ts::Service, SectionDemux etc, and my code then produces a minimal TS stream by only keeping the packets for the channel I am interested int)
- forward this data to FFmpeg AVFormat
- AVformat demuxes, decodes....

This is all done on a server app, which then filters, scales, etc, re-encodes and streams to the player.

The slowest bit in the above, is tuner->start(). This takes 4-5 seconds. After that the TSDuck demux is more or less instant, and it's maybe 0.5s-1.0s for AVFormat to fully demux the retained packets in a format that AVCodec etc can use.

Of course, Ideally I would enhance my usage of the TSDuck demux to directly fill out the AVFormatContext fields with the codec info etc and skip the avformat demux altogether, but that's a larger job I think and by the time I've done it I might ed up having transferred the 0.5s-1.0s avformat demux time to TSDuck demux time without really gaining anything.

Is there anything I can do to speed up the tuner->start() though? 5s is a long time before I've even started demuxing, let alone going through my transcode -> {INTERNET} -> player- decode step and means channel changes are a bit slower than I'd like.

For comparison WinTV 10, is maybe 2-3 seconds all in for tuning and buffering.

Maybe I don't need to tear down the tuner each time I change channels? Would that help if I just tuner->stop(), tuner->tune(), tuner->start()? Or is this just an artefact of the Windows tuner code?

lelegard, 2020-11-29:
Hi @oviano 

There are several tracks of improvements.

The first one is that TSDuck is probably not the right tool to do that on Windows. WinTV, and other proprietary TV-watching applications, are pure DirectShow applications. DirectShow is designed for audio/video streaming. TSDuck, on the other hand, is a transport stream processing toolkit. Roughly, DirectShow is designed to implement device-side applications while TSDuck is designed to implement head-end processing (not only but mostly).

DirectShow is a streaming engine, including the rendering endpoints. TSDuck does not need DirectShow in itself. DirectShow is a pain in the xxx of TSDuck. The problem on Windows is that DirectShow is only standard way of interfacing DVB tuners. TSDuck uses DirectShow for this sole reason. But, then, the problem with DirectShow is that you cannot extract data from it. Audio/video data flow from source (here a tuner) to the target (often a rendering engine) using the underground plumbing of DirectShow without any possibility to escape from it. Moreover DirectShow has a push model while TSDuck has a pull model. So, to do something as simple as reading raw packets from a tuner, TSDuck needs to insert a custom filter inside a DirectShow graph, use that filter to extract data out of the DirectShow plumbing, using asynchronous queues to convert the push model into a pull one. This has a cost in terms of performance, especially the initial latency you are experimenting.

If your application is a "TV-watching" type of app, I would recommend two solutions, depending on what is more important for you.

- If you want to stay on Windows, drop TSDuck and write a pure DirectShow application. Develop your own filters if necessary. Fully adopt the DirectShow push model. The code will be strictly non-portable but optimal in the Windows context. This is the WinTV model, with better performance as you noticed.
- If you prefer to stay with TSDuck and use its C++ code base for your application, move to Linux. The LinuxTV model is simple and straight. Tuning is just a few ioctl()'s followed by read() system calls directly to the drivers. It is much faster. And in terms of TSDuck coding, it is the same because the TSDuck API is system-agnostic.

If your application does not have too many dependencies on Windows, I would suggest to try it on Linux, just to compare the performances. 

Additionally, real TVset or STB middleware have optimised channel switching strategies.
- During the initial scanning, all PID's of all services are stored in an internal database. When switching to a new channel, the device immediately routes the audio and video PID's to the codecs. In parallel, the device classically listens to PAT, SDT, PMT to discover potential changes. In the rare cases of a PID reconfiguration, the first seconds or fractions of second are lost, the selected PID's are ineffective and the actual PID's are configured later (and updated in the database). But in 99.99% of the cases, the channel switching is much faster.
- Modern receivers now implement "smart fast switching". The device requires multiple tuners, at least two for live channels. Based on the fact that most users are "zapping" using P+/P- keys on the remote, pre-switching is implemented. Based on the current channel LCN, the device assumes that the next channel to switch on will be LCN+1 (or LCN-1 if the current channel was selected using the P- key). Thus, the second tuner is pre-programmed to the frequency of channel LCN+1, the signalisation is acquired and all audio/video PID's are immediately ready to be injected in the codecs if the next channel is the anticipated one. Then, the only zapping time is the result of the parallel acquisition of the next I-frame and next ECM (plus ECM processing and descrambling setup). If you have dual codec and descrambler resources, you can even anticipate these, effectively reducing the zapping time to zero. But this is for high-end devices only.

Keep in mind that TSDuck is a toolkit to experiment transport stream processing and transformation. Implementing an optimised receiver uses a very different software architecture. Many frameworks already exist for that: DirectShow on Windows, gstreamer on Linux for instance. But almost none exist for head-end experimentations. This is why TSDuck exist. So, I am just wondering if TSDuck is simply not the right tool for what you wan to do. (I am not a sales guy, I do no try to over-sell inadequate stuff for the customer's need :))

oviano, 2020-11-29:
So the project is a personal streaming server that can take input either from a video capture device (using either libs device dshow or custom Magewell Capture SDK) or from a tuner.

So you then configure various video quality profiles and channel favourites and the AV stream is then transmitted using SRT using a custom container format for the player to display.

It’s basically a higher quality customisable “Slingbox” style system that I wrote when I lived in Turkey and wanted near broadcast quality TV piped from the UK at full broadcast frame rate etc and using the best compression (h265 etc).

I chose TSDuck as it seemed like the only cross platform option and tbh I’m pretty happy with it. I could indeed battle myself with DirectShow directly but I would still like to keep the option open to use both Windows and Linux in future and at the moment there is no specific Windows dependency.

At the same time I don’t want to make it exclusively one platform or the other so Duck does suit.

lelegard, 2020-11-29:
> At the same time I don’t want to make it exclusively one platform or the other so Duck does suit.

So, I would recommend to benchmark your issues on Linux, with same hardware and same tuner if possible to compare comparable configurations. And if the performances are better on Linux, well, just use Linux.

That won't be the first time a single software system would have very different performances on distinct systems.

> Another option I guess would be to delve into the DShow stuff but still retain TSDuck for possible future Linux support but it doesn’t really appeal.

Yes, definitely. Don't use two completely different code bases for the two systems.

> last time I looked that way the whole DShow model hurt my brain.....

I definitely sympathise. Interfacing tuners on Windows was one of my worst software experience. I would recommend anyone to fly away from DirectShow. Unfortunately, Microsoft made it a bottleneck to access tuners...

oviano, 2020-11-30:
Thanks for the suggestions.

I've decided to take a bit of a detour and add HDHomeRun support to my project.

Rather than try and support these tuners when -as you clearly state - on Windows there simply isn't a reliable and properly-documented way to achieve this, it seems HDHR has a pretty simple cross-platform C API using sockets which should be straightforward to hook into.

I'll keep TSDuck in for the direct Tuner approach, as it *almost* works on Windows (depending on which way the wind is blowing), and would probably work well on Linux, but I don't think I want to spend a second more on trying to figure out why the DirectShow FilterGraph randomly delivers a stream containing corrupt packets to TSDuck (and it is, literally, depending on which way the wind is blowing, from what I can tell).

SS-FF, 2020-12-01:
> on Windows there simply isn't a reliable and properly-documented way to achieve this

This is entirely incorrect. It works reliably and it is used by dozens of applications without problems. The documentation is even better than in case of Linux. The only problem is that you need to be familiar with DirectShow **first**. That's what makes it hard to get started.

> I don't think I want to spend a second more on trying to figure out why the DirectShow FilterGraph randomly delivers a stream containing corrupt packets to TSDuck (and it is, literally, depending on which way the wind is blowing, from what I can tell).

DirectShow does not deliver "corrupt packets". That's either a tuning parameters or a programming issue. Maybe not draining data fast enough. I have never seen such issues with tsDuck. I would suggest that you run a regular release of tsDuck and simply output data to a file. Then investigate that file and see whether your "wind blowing" theory still stands ;-)

Slower tuning times than with Linux: yes, that's true. From my experience, it takes about 0.5 to 2s longer. You would need to create your own DirectShow filters and make some other changes to avoid this, like @lelegard had stated. Not an easy task, though.

oviano, 2020-12-01:
Well search my posts on these parts from a while back, i did a ton of testing and trial and error I made a long thread about it.

I couldn’t find any other logical explanation for what I was seeing than an issue with DirectShow, at least as implemented by TSDuck. Certainly TSDuck wasn’t introducing the corruption. You may be right about draining the buffer or something but again I couldn’t see how that was a possibility from the code, I tried everything.

I had hacked in code to grab the raw data from the DS filters before they were passed to TSDuck and saving them to file and they contained corruption. There is a comment on the TSDuck relating to it.

Then I’d start WinTV and get perfect reception. Switch back to TSDuck and I’d get corruption.

Just because you have not seen the issue doesn’t mean it doesn’t exist.

SS-FF, 2020-12-01:
We have done tests with a long list of different tuner models and users from all over the world and never experienced such issues.

Though, my baseline version of tsDuck is about 2 years old and I have made a few modifications...

oviano, 2020-12-01:
Sure, well it’s a known issue because @lelegard even has a comment in the code about it (where it loses synchronisation) and I’ve certainly seen it.

I have no idea how common or rare it is, only that I have seen the exact same thing as @lelegard’s code comment explicitly predicts and tries to mitigate.

SS-FF, 2020-12-01:
Which comment are you referring to?

oviano, 2020-12-01:
https://github.com/tsduck/tsduck/blob/master/src/libtsduck/dtv/windows/tsSinkFilter.cpp

Line 361.

SS-FF, 2020-12-01:
Thanks. Hm, I haven't made any modifications there. I have modifications to graph building and tuning. Also, I'm streaming out via UDP, so in theory it could also be the disk-writing which is causing the streaming to stall in some way.
BTW, I have a HCW Solo as well. The only issue I'm seeing sometimes is that the device (or its driver) is getting into a state where it doesn't work anymore, not even after un- and re-plugging. Only a reboot helps. But others haven't seen this, so it might be something about my local system, like USB hw or driver.

SS-FF, 2020-12-02:
From a quick look at the tsSinkFilter code, I see two things:

- `ReceiveCanBlock` returns S_FALSE, even though there's a lock in the `Receive` method
  This could have the effect that all available pool threads call into the `Receive` method and get stuck at the lock, which could in turn cause all processing at the side of upstream filter to cease and not process some input data.
- The IMemInputPin implementation does not provide an allocator
  That means that the samples are allocated elsewhere. That allocation might not be unlimited. The tsDuck code is putting the samples into a queue and calling ::AddRef, which means that they are not getting released. At some point, the source pin allocation might hit a limit (no more samples can be allocated) causing data to be dropped and never getting sent to the input pin

This is what I would look at.

lelegard, 2020-12-02:
Hi @oviano and @SS-FF,

First, I clearly admit that the TSDuck SinkFilter is far from good. I wrote it 10/12 years ago and never came back to it. At that time, the tuner interface of TSDuck was only working on Linux. I struggled quite a lot to port it on Windows and the discovery of DirectShow was a real nightmare. I do not agree on the fact that DirectShow is well documented. It is awful. There are a lot a small and probably auto-generated docs on each method. But there is absolutely nothing which gives a clear and detailed view of DirectShow in an open manner. You can find many examples of full DirectShow streaming applications. But it gives the idea that DirectShow is some independent world. You are in our out but there is no way to interface the two worlds, which was precisely what TSDuck wanted to do.

On Linux, the LinuxTV tuner API is clear and simple. It interfaces tuners, full stop. Then, if you want to develop some streaming application, you can use frameworks such as gstreamer which are equivalent to DirectShow at architectural level. But you can use LinuxTV tuners alone, without gstreamer.

On Windows, the concept of BDA driver is defined, but essentially on a device standpoint. The BDA documentation focuses on driver development. There is no complete and tuner-agnostic API for BDA drivers on the userland side. A tuner vendor usually develops the driver _and_ the corresponding DirectShow filter. The only "clearly" defined API is inside DirectShow, forcing applications such as TSDuck to dig into DirectShow while we absolutely do not need a streaming framework.

I started to understand how DirectShow actually works when I read the parts of the VLC source code which interface tuners using DirectShow, but certainly not from the Microsoft documentation. I wrote the TSDuck SinkFilter, made it work, later made some modifications such as the corruption issue you mentioned and _never_ modified it.

The corruption issues were not systematical. They occurred on some tuners only. I never really demonstrated if they came from the drivers of some devices, from DirectShow or from the way TSDuck uses it.

Note that DirectShow forces the application to have a push-oriented streaming architecture while LinuxTV is just an API which can be used in any architecture. A full DirectShow application which adopts the DirectShow push model will probably work better than applications such as TSDuck which use a push/pull adaptation layer.

According to my experience, working with tuners has always been more reliable on Linux. TSDuck was initially developed on Linux. Two years ago, it was ported to Windows, _except_ the tuner interface. After a couple of years, I was more or less pushed to port the tuner part on Windows for one simple reason: my Linux machines with tuners were all desktop systems while my only laptop was a Windows office machine. I had to make tests on the terrestrial TV and moved to different regions to analyse local transmitters. So, I needed tuner support on my Windows laptop. This is the sad story of TSDuck on Windows... :)

My understanding of DirectShow improved a bit when you helped implementing ATSC and ISDB. If you have suggestions on how to improve the SinkFilter, please do. I must admit a have some reluctance to dig into it again and make lengthy tests and debug. Not to mention that I now got rid of any personal Windows machine. I just run a Windows VM on my Mac (USB redirection works pretty well for tuners with VMware Fusion). I still use a Windows laptop in my professional activity, but this is an independent domain.

oviano, 2020-12-02:
> Also, I'm streaming out via UDP, so in theory it could also be the disk-writing which is causing the streaming to stall in some way.

As mentioned, the corruption can appear before anything is done with the data such as writing to the disk.

My conclusion was that it was occurring higher up in the code before @lelegard's filter was getting hold of it.

I too have no current intention of diving into this code anytime soon tbh, if that changes then this would be a useful additional reference with your additional suggestions, thanks.

SS-FF, 2020-12-02:
> > Also, I'm streaming out via UDP, so in theory it could also be the disk-writing which is causing the streaming to stall in some way.
> 
> As mentioned, the corruption can appear before anything is done with the data such as writing to the disk.

There is no "before" in a streaming chain that can block upstream data flow.

> DirectShow forces the application to have a push-oriented streaming architecture

DirectShow supports both, push and pull models, even mixed in a single filter graph. Simple example for a pull-model is a file source filter connecting to an audio sink filter (e.g. default sound device). But in case of a BDA source - yes, the data is being pushed, which of course makes sense, because the data is received in realtime as the bytes come in and not when an application decides to consume them.

I think the best solution for the sink filter would be to have some memory stream implementation to which the `Receive` method can write data immediately without locking and avoiding to `AddRef` and store the actual `IMediaSample` objects.

oviano, 2020-12-02:
Thanks, well it’s certainly useful to hear from someone else with knowledge in this area, I could’ve done with your input a couple of years ago when I was buried deep in this issue without much of a clue what I was doing 😊

Anyway thanks for the pointers. If I’m ever tempted to have another look then this will be very useful. Maybe @lelegard won’t be able to resist getting his hands dirty again, such is his love of DirectShow programming....

SS-FF, 2020-12-02:
@oviano - may I ask which personal media server project you're working for?
(I know it's somewhat ignorant to ask, while I'm not allowed to disclose my project at this time..)

oviano, 2020-12-02:
Of course, here is the website

emustream.tv

The site is a bit out-of-date for example where it mentions the thirdparty libs I’m using as I’ve changed a fair bit since then (for example the player no longer uses libVLC and I now use SRT for the stream) but the site explains the basic concept.

It’s basically a sort of customisable Slingbox-style system which I developed when I lived in Turkey and wanted as near-to-broadcast quality TV sent from the UK, ie proper high frame rate for sports viewing and using the best compression methods such as HEVC.

The code is currently on a private Gitea repo but I’ll probably put it out there publically at some point.

SS-FF, 2020-12-02:
Thanks, looks like a nice project and luckily far enough away from mine.

I thought you might be interested in this:

[image]

The time between the 'Run' button click until you see the first picture includes starting up tsp, tuning to the channel, processing the stream (outside of tsduck), filtering out PES streams, creating HLS segments and playlist and starting MPV player to play the HLS stream.

PS: tsDuck was running in a Linux VM for this example.

oviano, 2020-12-02:
Thanks for sharing, yes that looks nice and quick. Good to know!

SS-FF, 2020-12-03:
I got the same with fmpeg video conversion adding only 0.5-1s additional delay. 

It's been quite a way to get to that point and many details needed to get right. But one general advice that I can give is this:
Better use those components out-of-process instead of trying to plumb them together in a single application - especially for server-type applications where a certain level of reliability and error resiliency is required.

SS-FF, 2020-12-03:
@lelegard - One thing I'm wondering about: Why isn't the `DvbInputPlugin` derived from `PushInputPlugin`? Wouldn't that make more sense?

oviano, 2020-12-03:
> I got the same with fmpeg video conversion adding only 0.5-1s additional delay.
>
> It's been quite a way to get to that point and many details needed to get right. But one general advice that I can [...]

I started with that approach but as I became more and more familiar with the FFmpeg libraries I switched to doing exactly what you advise against and embedding the libraries directly in my framework.

I’ve never had a problem, and much prefer doing it this way and learning how the API works.

lelegard, 2020-12-03:
> @lelegard - One thing I'm wondering about: Why isn't the DvbInputPlugin derived from PushInputPlugin? Wouldn't that make more sense?

It probably would if TSDuck was Windows-only.

But TSDuck is portable and, in principle, a tuner is "just" an input device like any other from which it is just natural to call some read() call. This is the case on Linux. And this would also be the case on Windows if there was a uniform userland API for BDA drivers. The actual and probably unique problem, as far as TSDuck is concerned, is that the uniform userland API was moved by Microsoft from BDA drivers to DirectShow filters. A very bad software design decision.

Anyway, the abstract class `PushInputPlugin` is just a buffering adaptation layer which converts the push model into a pull one, just as `SinkFilter` does. Using `PushInputPlugin` with `dvb` on Windows would simply move the push/pull adaptation layer into another class. This wouldn't solve the architectural problem. And we would loose the portability.

Interesting enough, `PushInputPlugin` was developed to cope with HTTP access where the problem is exactly the opposite: Windows uses a simple pull model where `InternetReadFile()` does a blocking read from the connection while libcurl uses a push model. There is one difference though : there are other ways than "curl easy" to use libcurl and there are other libraries too. But this one was the most widely spread and the most easy to use.

SS-FF, 2020-12-03:
> But TSDuck is portable and, in principle, a tuner is "just" an input device like any other from which it is just natural to call some read() call. This is the case on Linux. 

Where is the data buffered on Linux? What happens when you don't read() fast enough? I mean, Linux can't infinitely buffer the data.

> Anyway, the abstract class PushInputPlugin is just a buffering adaptation layer which converts the push model into a pull one, just as SinkFilter does. Using PushInputPlugin with dvb on Windows [...]

I don't see an architectural problem actually. Just an implementation problem in the sink filter, that's all.

SS-FF, 2020-12-03:
> Using PushInputPlugin with dvb on Windows would simply move the push/pull adaptation layer into another class.

But that class would have a different implementation that does not block the DirectShow data flow.

> And we would loose the portability.

Right. It's probably better to fix it in the Sink Filter directly.

SS-FF, 2020-12-03:
@oviano 

> so that the HW decoded output frames are treated correctly on the various platforms

Sure. Client players is not what I meant, though. 

> treat FFmpeg as a blackbox

LOL. Not quite like that... I'm not running a regular version and I'm in fact very familiar with its code.

lelegard, 2020-12-03:
> Where is the data buffered on Linux? What happens when you don't read() fast enough? I mean, Linux can't infinitely buffer the data.

All broadcast / multicast sources share the same principle: if you don't read fast enough, you loose data packets. This is equally true for IP multicast and DVB.

oviano, 2020-12-03:
> @oviano
> 
> > so that the HW decoded output frames are treated correctly on the various platforms
> 
> Sure. Client players is not what I meant, though.
> 
> > treat FFmpeg as a blackbox
> 
> LOL. Not quite like that... I'm not running a regular version and I'm in fact very familiar with its code.

Yes but client players are part of my project, and it's beneficial to share some of the code at least between server and client (demuxing, some parts of decoding etc). You offered advice, I'm explaining that my project is not the same as yours so my requirements are different and I have different priorities.

In terms of FFmpeg being a blackbox, I don't doubt you are very knowledgeable about those libraries, you shouldn't take what I wrote as some sort of slight! I'm merely explaining that my preference was to learn and understand the code; I started off by constructing all sorts of elaborate command line strings to pass to a launched FFmpeg process and in the end I decided I'd rather learn how to do it from the inside because for me, at that point, FFmpeg *was* a black box to me. In this sense, it's as much as personal hobby for me as anything else; similarly I started off with the player using SDL + libVLC, because this was an easy and quick way to get something up and running but over time this has evolved into using my own Windowing and input library, with sokol_gfx as the thin 3D API layer (at one stage I used bgfx, but recently switched to sokol - again because I bgfx was dragging in some extra dependencies and complication which I didn't need, whereas sokol_gfx is about as thin a layer as you can get for a 3D API wrapper), and then using libavcodec etc directly, rather than going through LibVLC.

In terms of stability, I assume you mean that it's better if FFmpeg code crashes if it's an external process, without bringing down your server, and that is a valid point. However, in my experience the code is pretty stable, and on the rare occasion it has crashed or hung I've usually figured out the reason. This process itself has often helped my understanding of the libraries and how everything fits together.

Error resilience is possibly a more valid concern, because ffmpeg.c (I think) contains a fair bit of logic for fixing broken timestamps and stuff like that, which you'd miss out on by linking directly, if you didn't re-implement it yourself but there is only a very limited subset of inputs to my project (essentially raw video captures, or broadcast mpegts) so it's not something that has been a problem for me. I can see if you were taking inputs from anywhere, files, a bigger variety of sources, then these might become bigger advantages.

Anyway, this is wildly off-topic, so I'll stop there. Good luck with your project 👍

SS-FF, 2020-12-03:
I guess I need to apologize because I've been a bit too focused on my own requirements. Your guessing is right about the wide variety of input file, output requirements, hwa methods and platforms that I need to handle.
Another reason that caused me to identify with your efforts is, that when I had started into this a number of years ago, I had a very strong opinion against running ffmpeg in a separate process, as well. Today I'm still seeing user issues regularly where I think how glad I am that we haven't switched to an in-process model.

Apart from this, I agree to everything you said (your reply even includes my due apology-text).

lelegard, 2020-12-03:
> Maybe @lelegard won’t be able to resist getting his hands dirty again, such is his love of DirectShow programming....

Argghhh, why are you doing this to me? 😢 

> I think the best solution for the sink filter would be to have some memory stream implementation to which the Receive method can write data immediately without locking and avoiding to AddRef and store the actual IMediaSample objects.

Some form of locking is needed anyway when you read/write a shared buffer. And AddRef is a negligible overhead.

The actual question is: is there is a fixed-size pool of MediaSample objects in a DirectShow graph? Or a maximum size for it? In that case, it would be beneficial to release the MediaSample objects early and buffer the TS packets in some other private internal pool. But if there is no limit to the number of allocated MediaSample objects (just the good will of everyone to release them when no longer necessary), I do not see the benefit of using another kind of buffer. Memory is just memory and using MediaSample objects may only add some negligible overhead.

SS-FF, 2020-12-03:
> Some form of locking is needed anyway when you read/write a shared buffer. And AddRef is a negligible overhead.

It's not the time overhead of calling AddRef. It's that this will prevent a sample from being re-used until it's released.

SS-FF, 2020-12-03:
> The actual question is: is there is a fixed-size pool of MediaSample objects in a DirectShow graph? Or a maximum size for it?

This depends on the `IMemAllocator `

`IMemAllocator::GetProperties` https://docs.microsoft.com/en-us/windows/win32/api/strmif/nf-strmif-imemallocator-getproperties

returns an `ALLOCATOR_PROPERTIES` structure:

https://docs.microsoft.com/en-us/windows/win32/api/strmif/ns-strmif-allocator_properties

SS-FF, 2020-12-03:
These things are handled here:

https://github.com/tsduck/tsduck/blob/0f6a55f943a7f7b5852b868a67299b44b41e1886/src/libtsduck/dtv/windows/tsSinkFilter.cpp#L730-L746

SS-FF, 2020-12-03:
I have just modified the No like this:

```cpp
STDMETHODIMP ts::SinkPin::NotifyAllocator(::IMemAllocator* pAllocator, ::BOOL bReadOnly)
{
    TRACE(1, u"SinkPin::NotifyAllocator");

    ALLOCATOR_PROPERTIES props;
    pAllocator->GetProperties(&props);

    std::cerr << "SinkPin::NotifyAllocator - buffers: " << props.cBuffers << " buffer size: " << props.cbBuffer  << std::endl;

    return S_OK; // we don't care
}
```

The output I'm getting is this:

`SinkPin::NotifyAllocator - buffers: 24 buffer size: 45120`

while tsDuck uses:

```cpp
static constexpr size_t DEFAULT_SINK_QUEUE_SIZE = 1000;  // media samples
```

That means that the sink filter may queue up to 1000 media samples while DirectShow won't supply more than 24 media samples (each having a max size of 45120 bytes)

lelegard, 2020-12-03:
> That means that the sink filter may queue up to 1000 media samples while DirectShow won't supply more than 24 media samples (each having a max size of 45120 bytes)

Good to know, thanks! This may explain issues on Windows.

There are two options. The simplest one (for a DShow-hater like me), is to copy each MediaSample content into an intermediate object and queue those objects. But there is a copy overhead. The second option is to provide a custom "allocator" which allocates (or reuses) MediaSample objects. But I need to dive back into DShow 😱. Would you have a pointer to a sample MediaSmaple allocator, just to avoid some vomit on my side...

Note that there is already a buffer-copy overhead issue on Windows. `tsp` uses a large global TS packet buffer which is shared by all plugins. No copy ever occurs. On Linux, because LinuxTV is a simple API, packets are directly read from the tuner directly into the `tsp` global buffer. On Windows, because of the push model of DirectShow and its MediaSample objects, the packets are copied several times. This is why I would like to avoid having an additional level of buffering.

SS-FF, 2020-12-03:
> The simplest one (for a DShow-hater like me), is to copy each MediaSample content into an intermediate object and queue those objects.

A much more simple method might be to handle `GetAllocatorRequirements` and return a larger number of samples. 
But I don't really want to suggest this.

Isn't there a way to immediately copy the data to the final "TSP packet buffer"?

SS-FF, 2020-12-03:
> Would you have a pointer to a sample MediaSmaple allocator, just to avoid some vomit on my side...

I don't think that it makes much sense to do a custom implementation of `IMemAllocator`.  I think that this would only make sense when it would be possible to return the final target memory range to which the data should be written to, but I don't think that this is possible at all. (samples can have variable lengths and need to be re-usable).

What's possible is to create the standard memory allocator with a custom set of allocator properties, but maybe this won't have a different result than just handling `GetAllocatorRequirements`.

lelegard, 2020-12-03:
> Isn't there a way to immediately copy the data to the final "TSP packet buffer"?

The queue is the adapter between the push and pull models. We need at least some level of queuing when media samples are pushed and the TSDuck tuner thread is not currently trying to read from the tuner. When the queue is empty and the tuner thread is waiting for read, then it could be possible to copy the MediaSample content from `SinkPin::Receive` but this won't change the number of copy operations. Copying the TS packets directly from the tuner to the final tsp packet buffer would be possible only with a direct access to the BDA driver, as with Linux. The DirectShow push models imposes this additional level of data copy.

SS-FF, 2020-12-03:
Why do we need to wait for the tuner thread in order to write to the packet buffer?

lelegard, 2020-12-03:
> Why do we need to wait for the tuner thread in order to write to the packet buffer?

Because I do not want to break the complete design of TSDuck. For an input thread, waiting for and reading packets is a call to `ts::Tuner::receive()`. This method gets a buffer address and size where to receive packets. If no such call is currently in progress (ie. the input plugin thread is not currently reading), we do not know where to copy the packets.

SS-FF, 2020-12-03:
OK, none of these things will work unfortunately.

- GetAllocator does not get called
- GetAllocatorRequirements does not get called

Both are optional:
https://docs.microsoft.com/en-us/windows/win32/directshow/negotiating-allocators

lelegard, 2020-12-03:
> That means that the sink filter may queue up to 1000 media samples while DirectShow won't supply more than 24 media samples (each having a max size of 45120 bytes)

Each media sample can have up to 420 packets. The total buffering capability is 1 MB (1,082,880 bytes). Even with the fastest DVB-S2 stream, reading 10 times per second is enough. So, I do not really think that would cause packet loss.

> GetAllocatorRequirements does not get called

Yes, the article you mention says:

> Optionally, the output pin calls IMemInputPin::GetAllocatorRequirements.

So, based on the graph below, if the Tee filter does not negociate the allocator requirements, we have no way to resize the number of MediaSamples.
~~~
Tuner --> Receiver --> Tee -+-> SinkFiler (provided by TSDuck)
                            |
                            +-> MPEG-2 demux --> TIF
~~~

SS-FF, 2020-12-03:
> Each media sample can have up to 420 packets

45120 / 188 = 240

SS-FF, 2020-12-03:
> Each media sample can have up to 420 packets. The total buffering capability is 1 MB (1,082,880 bytes). Even with the fastest DVB-S2 stream, reading 10 times per second is enough. So, I do not really think that would cause packet loss.

Another idea: Maybe it's better to block in the receive method and wait until we can write directly to the TS Packet Buffer.

I think, anything is better than `AddRef`ing the samples.

SS-FF, 2020-12-03:
> Each media sample can have up to 420 packets. The total buffering capability is 1 MB (1,082,880 bytes). Even with the fastest DVB-S2 stream, reading 10 times per second is enough. So, I do not really think that would cause packet loss.

Shouldn't this be viewed the other way round? 

When the tuner thread doesn't poll for longer than 0.1s, the buffer is getting exhausted and packet loss occurs.

SS-FF, 2020-12-03:
> So, based on the graph below, if the Tee filter does not negociate the allocator requirements, we have no way to resize the number of MediaSamples.
>
> ```
> Tuner --> Receiver --> Tee -+-> SinkFiler [...]

Good point. For some reason I thought we would be behind the MPEG2 demux, but of course we're not.

In that case - we do not want to increase the sample count anyway. AND we must not block. All this can increase tuning time because the network provider communicates with the MPEG2 demux and the TIF.

Hmmm....could that even be the reason why tuning times are slower on Windows?`
When the Sink Filter blocks at the input side, this could block the Tee splitter in turn and prevent it from sending to the MPEG2 demux.

lelegard, 2020-12-03:
> > Each media sample can have up to 420 packets
>
> 45120 / 188 = 240

Yes, sorry, I have always been a bit dyslexic with the keyboard. I am sure you can find a few inverted letters in the doc, comments or anywhere the compiler doesn't spot such errors...

> Another idea: Maybe it's better to block in the receive method and wait until we can write directly to the TS Packet Buffer.
> I think, anything is better than AddRefing the samples.

I totally disagree.

In the [Microsoft doc here](https://docs.microsoft.com/en-us/windows/win32/api/strmif/nf-strmif-imeminputpin-receive) about `IMemInputPin::Receive` method:

> This method is synchronous and possibly blocking. The pin does one of the following:
>
>   Rejects the sample.
>   Returns immediately and processes the sample in a worker thread.
>   Processes the sample before returning.
>
> In the last case, the method might block indefinitely. If this might happen, the IMemInputPin::ReceiveCanBlock method returns S_OK.
>
> If the pin uses a worker thread to process the sample, it holds a reference count on the sample. In any case, the output pin cannot directly re-use this sample. It must call the IMemAllocator::GetBuffer method to obtain a new sample.

TSDuck is clearly in the case of a _"worker thread"_ and _"holds a reference count on the sample"_. This is much more efficient and "fluid" than blocking. Blocking would definitely increase the risk of packet loss.

In that case, it is totally normal to `AddRef`. The corresponding `Release` is performed in `SinkFilter::Read`, after the content of the media sample is returned to `ts::Tuner::receive()`.

> When the tuner thread doesn't poll for longer than 0.1s, the buffer is getting exhausted and packet loss occurs.

Yes, I meant the same thing, it was just formulated in another way. Given the way `tsp` is designed, there is zero delay between calls to `ts::Tuner::receive()`. Once a bunch of packets is received in the input plugin (whichever it is), the corresponding slice of the global buffer is logically passed to the next plugin and all the free space in the global buffer is immediately reused for the next `ts::Tuner::receive()`. See a [diagram here](https://tsduck.io/doxy/devplugin.html#pdevdesign).

So, in practice, the input thread will read packets much faster than 10 times per second. It is litterally reading packets all the time. The input plugin only blocks when waiting for packets (in `ts::Tuner::receive()` in the case of the `dvb` plugin).

If the input plugin blocks longer between two reads, this can be only because the global buffer is full (the thread then waits for free space in the buffer). In that case, we have much bigger problems anyway. That means that there is one or more plugin which "significantly" blocks the flow. I mean "significantly" because each plugin runs in its own thread, all of them, using all CPU cores in the system. This asynchronous architecture was designed from the beginning, precisely to avoid losing input packets. Since this design was adopted as early as 2005 to read from a Dektec ASI board up to 270 Mb/s on a Pentium IV at that time, I am rather confident that the input plugin will not be the bottleneck.

> When the Sink Filter blocks at the input side, this could block the Tee splitter in turn and prevent it from sending to the MPEG2 demux.

For the same reason, I do not think that this explains the issue, the SinkFilter will likely not block, certainly not in a perceptible way as the obviously long zapping time.

Maybe I would have better implemented the SinkFilter that way:
~~~
Tuner --> Receiver -+-> SinkFiler --> MPEG-2 demux --> TIF
                    |
                    V
                    sink queue to ts::Tuner
~~~
The SinkFilter would have implemented the Tee itself, avoiding the fork in the graph. But, at that time, when I could not make it work and could not find any clear documentation on how to use DirectShow graphs, I used DirectShow Spy to view how proprietary applications such as WinTV built graphs. The TSDuck graph is exactly like those, with the SinkFilter replacing the rendering path.

In fact, functionally speaking, TSDuck does not need any demux or TIF because we just read all TS packets. But, in tha absence of demux and TIF, the graph does not start. I told you, I hate DirectShow!

And honestly, today, I do not feel like redesigning the SinkFilter.

SS-FF, 2020-12-03:
> And honestly, today, I do not feel like redesigning the SinkFilter.

I don't either. I don't think it's time to start into developing something, because all we have right now is a couple of assumptions, but we don't know yet what's happening exactly.
So I would rather start with some logging that regularly prints out

- the queue size
- the number of threads waiting at the locks
- the time span between subsequent Read calls (from the tuner thread)

We should also record and report the maximum values for these values.
Maybe not use the TRACE calls, as that might generate too much noise.

I can't help testing this, because I can't reproduce here.

Meanwhile, I'll try a different route: The plan is to create another DirectShow sink filter that will directly send out all received data via UDP. I hope this will allow me to gain some more insights into the latency issue.

lelegard, 2020-12-03:
> Meanwhile, I'll try a different route: The plan is to create another DirectShow sink filter that will directly send out all received data via UDP. I hope this will allow me to gain some more insights into the latency issue.

This is good idea but I would be careful with this. This is a recipe for packet loss because the UDP reception kernel buffers are probably smaller than TS input. If is worth a try however.

SS-FF, 2020-12-03:
> UDP reception kernel buffers

Why "reception"?

SS-FF, 2020-12-03:
> Why "reception"?

Or better: what do you think would be different compared to using the IP output plugin from tsDuck?

lelegard, 2020-12-03:
I mean on the UDP receiver side. Sending is fine, but someone has to receive it. Receiving traditional UDP multicast (IPTV) means receiving small datagrams (typically 7 TS packets) on a regular basis, at the bitrate of an SPTS. Even if you are not fast enough to read each datagram on time, a few of them can be buffered. On the other hand, DVB MPTS streams have higher bitrates and media samples from the DirectShow filter are maybe larger than 7 TS packets (to be checked) since the buffer size is 240 packets (in the right order this time:)). If you send large UDP datagrams from the input pin's `Receive()` method, you should benchmark the effect on typical UDP receivers (including `tsp`, but not only). And if you cut the media sample is smaller pieces, either you send them all right away and your burst effect can be devastating, or you smooth them and you block in the input pin's `Receive()`, a method I did not recommend in a previous post.

SS-FF, 2020-12-03:
Thanks for the explanation. I'm using UDP from the very first moment on to connect tsDuck to our application and it's working rock-solid. 

But I had almost forgot how painful it was to get there.  I just looked it up and in fact, I'm adjusting the socket receive buffer, so you just hit the right point. (among others)

SS-FF, 2020-12-05:
Today I have created a new `UdpSinkFilter` which - when used instead of the regular SinkFilter - sends out data via UDP directly.

This way, I was able to achieve the same startup times like on Linux. The delay of 1-2s that I have been experiencing on Windows vs. Linux doesn't exist anymore.

To be fair and transparent, here's some details on what I've compared:

- Tuner, antenna and tuning parameters are always the same
- I'm testing Linux in a virtual machine with the USB tuner being redirected from the Windows host
- Received data is transmitted via UDP to a Windows application
- Of course, the UDP Sink Filter is bypassing all TSP processing, while for Linux, the data is going through TSP processing and sent out via the IP output plugin
  (I don't know how much delay the TSP processing itself is adding here)

This confirms what I have suspected above: This is not a DirectShow problem and not a BDA driver problem. It's an implementation problem in tsDuck.

I still need to find a nice way to keep TSP alive even when it doesn't receive anything from the tuner...

oviano, 2020-12-05:
Interesting finding, nice test 👍 

It is certainly surprising if the buffering is adding that much latency...

lelegard, 2020-12-05:
It's great that you found a solution for your application.

I would like to understand if the issue comes from the SinkFilter itself or from the graph setup and tuning. When you use your UDP sink filter, do you use the rest of TSDuck, the code that sets up the graph and sends the tuning request, or do you use a more traditional DirectShow application with only your UDP sink filter as specific piece?

> It is certainly surprising if the buffering/locking/whatever-is-happening is adding that much latency...

You can play with options `--max-input-packets` and `--max-flushed-packets` to alter the buffering. But, with the `dvb` plugin, the default value is 1000 packets, which represent only 0.06 seconds of a 24 Mb/s DVB-T stream or 0.02 seconds of a 70 Mb/s DVB-S2 stream. Locking does not seem a realistic explanation either.

SS-FF, 2020-12-05:
To add some more value rather than just make claims, I have done some more testing and recorded the log timings for the three cases to compare:

#### 1. Normal TSP Run on Windows:

```
11:55:09.031 tsp: TSDuck - The MPEG Transport Stream Toolkit - version...
11:55:09.507 * dvb: tuned to transponder --frequency 370000000 --symbol-rate 6900000 --fec-inner none --spectral-inversion auto --modulation 256-QAM
11:55:09.507 * dvb: starting tuner reception
11:55:10.804 * dvb: signal locked: yes, strength: 100%, quality: 76%
11:55:12.610 All threads started.
11:55:12.640 FirstDataReceived
```

**Total: 3.609s**

#### 2. Normal TSP Run on Linux VM:

```
11:38:21.720 tsp: TSDuck - The MPEG Transport Stream Toolkit - version...
11:38:21.896 * dvb: tuned to transponder --frequency 370000000 --symbol-rate 6900000 --fec-inner none --spectral-inversion auto --modulation 256-QAM
11:38:21.897 * dvb: starting tuner reception
11:38:22.148 * dvb: signal locked: yes, strength: 62%
11:38:23.399 All threads started.
11:38:23.421 FirstDataReceived
```

**Total: 1.701s**

#### 3. TSP Run on Windows using UDP Sink Filter

```
11:43:44.297 tsp: TSDuck - The MPEG Transport Stream Toolkit - version...
11:43:44.820 * dvb: tuned to transponder --delivery-system DVB-C --frequency 370,000,000 --symbol-rate 6,900,000 --fec-inner none --modulation 256-QAM
11:43:44.820 * dvb: starting tuner reception
11:43:46.079 * dvb: tuner reception started
11:43:46.112 * dvb: signal locked: yes, strength: 100%, quality: 76%
11:43:46.156 FirstDataReceived
```

**Total: 1.815s**

Notes:

- The results are very reproducible, I ran each variant multiple times and the variance was below 50ms
- I have a local modification to always print out the version, so this can be considered as the TSP startup time
- All lines are from TSP (as received by my test application), except `FirstDataReceived`, which indicates the first moment when data is received via UDP

SS-FF, 2020-12-05:
> When you use your UDP sink filter, do you use the rest of TSDuck, the code that sets up the graph and sends the tuning request, or do you use a more traditional DirectShow application with only your UDP sink filter as specific piece?

I've done this in a very simple way:

- I've copied the SinkFilter
- Removed the queuing stuff
- Then I copied the relevant parts from the IP Output plugin (except the RTP stuff)
- Changed the graph building to use the new UdpSinkFilter instead
- Added 20s Sleep to the tuner::Read function 
  (to avoid TSP from exiting immediately)

SS-FF, 2020-12-05:
PS: I can share this via PR, but it's just experimental code with the UDP target being hardcoded.

[25 more replies]

closed 2020-12-14 by oviano