tsduck/tsduck #672: [BUG] tstables: XML output not closed?

lars18th, 2020-11-30:
#### Bug description:
The `<tsduck>` tag is not closed at the end of the file.

#### How to reproduce:
`tsp -I ip 224.10.11.12:1000 | tstables --all-once --psi-si --pack-all-sections --pack-and-flush --xml-output data.xml`

#### Expected behavior:
At the end of the output file ("data.xml") will be a `</tsduck>`. So the closing tag is missing.

#### Environment:
 - OS: Windows
 - OS version: Windows 7 x64
 - TSDuck full version: release 3.23-1965
 - Installation type: Regular release full install

lelegard, 2020-11-30:
Hi @lars18th 

I assume that your problem is due to the way you terminate the application. By construction, this command never terminates. There is never any natural end of file in the `ip` input plugin and there is no termination condition in any of the two commands `tsp` or `tstables`.

So, you probably ended it with a Ctrl-C on the console, right? In that case, the interruption is sent by the OS to the two processes you are running: `tsp` and `tstables`.

As you may imagine, `tstables` starts with writing `<tsduck>` in the XML output. Then, all tables are written in XML format. At the end of input,  `tstables` writes `</tsduck>`  before exiting. 

When `tsp` gets the Ctrl-C, it terminates its processing. The output (here a pipe) is properly closed. The pipe receiver (`tstables`) then receives an end-of-file and terminates properly with the final `</tsduck>`. But here the Ctrl-C is dispatched to the two processes at the same time and you kill `tstables` before it has the opportunity to properly terminate. If (from the task manager for instance), you had only killed the `tsp` process, then `tstables` would have had the opportunity to terminate on end-of-file and would have written the final `</tsduck>` .

If you have an endless command that you must explicitly kill, you'd better use this:
~~~
 tsp -I ip 224.10.11.12:1000 -P tables --all-once --psi-si --pack-all-sections --pack-and-flush --xml-output data.xml -O drop
~~~

In that case, there is only one process. The Ctrl-C is handled synchronously by `tsp` which then notifies all plugins to terminate. As part of its termination, the plugin `tables`  writes the final `</tsduck>` .

Of course, all of this is pure speculation based on the hypothesis that you hit Ctrl-C.

lars18th, 2020-11-30:
Hi @lelegard ,

You're absolutely right! I'm using Ctrl-C. And now I see the problem. So this is not a bug, and I close it.

Regards.

closed 2020-11-30 by lars18th