tsduck/tsduck #674: [FEAT] "Merge" values in tables using XML

lars18th, 2020-11-30:
#### Is your feature request related to a problem?
For some small actions it will be preferable to have a direct method to replace/modify/fix some PSI-SI tables instead of save them, edit with one script and reinsert them. I know that TSDuck is _scripting oriented_, and it supports XML files to improve this capability. However, at some times it will be desirable to modify a simple value of a table without requiring the full path: extract, modify, reinject. One example: You want to change the Centre Frequency of a Terrestrial delivery system in the NIT table before sending the content (from a file) to a modulator. As the TSDuck plugins can't support all possible descriptors, the solution (at time) is to generate the XML version of the table, patch/modify them, and reinject.

But I feel this can be speed up!

#### Proposed solution:
My proposed solution is to support an "XML merging mode" for plugins like nit, pat, etc. Using the above example, you can use:
- `tsp ... -P nit --patching-xml merge.xml ... -O hides -f 474000000` and the content of the `merge.xml` is:

```
<?xml version="1.0" encoding="UTF-8"?>
<tsduck>
  <NIT>
    <transport_stream>
      <terrestrial_delivery_system_descriptor centre_frequency="474,000,000"/>
    </transport_stream>
  </NIT>
</tsduck>
```

And as you can see the content of the file is a subset of the entire NIT table that only contains the value to replace.

#### Alternatives:
The only alternative at time, is to extract the table, modify it with scripting and reinject it. But this is slow to do it compared to the proposed solution.

#### Additional information:
Perhaps you know another more simple way to achieve this.

Please, comment about this.
Thank you.

lelegard, 2020-11-30:
Could be possible, let me think about it.

lelegard, 2020-12-02:
Added options `--patch-xml` to plugins `bat`, `cat`, `nit`, `pat`, `pmt`, `sdt` (which all share a common base code). It could be easily applied to other plugins if necessary.

Also added a new tool named `tsxml` to test the XML patch files on specific XML files. See the description of `tsxml` in the user's guide for more details on XML patch files.

It was relatively easy to implement it that way.

lars18th, 2020-12-20:
As it's implemented, I close it.
Thank you!

closed 2020-12-20 by lars18th