tsduck/tsduck #675: Error: no input DVB signal after 5,000 milliseconds

RolsimGit, 2020-12-02:
First of all many thanks for disclosing the sources. This is not a matter of course.

I have the following problem:

Hardware : Hauppauge HVR 4400
Develop : Visual Studio 2019

With the parameters 

**tsp -O file test.ts -P until --seconds 20 -I dvb --delivery-system DVB-S2 --lnb Astra --frequency 11493750000 --polarity horizontal --symbol-rate 21995000 --device-name :1** 

 I get a stream.

but I cannot get a stream in my own program

parameter before the call of

	erg = theApp.m_pTuner->tune(*theApp.m_pModArgs, *theApp.m_pReportFile);

***** ModArgs *****
**-		*theApp.m_pModArgs	{delivery_system={_access=0x014f2678 {DS_DVB_S2 (2)} _data=0x014f2678 "\x2" } frequency={_access=0x014f2684 {...} ...} ...}	ts::ModulationArgs
+		ts::Object	{...}	ts::Object
+		ts::ArgsSupplierInterface	{...}	ts::ArgsSupplierInterface
+		delivery_system	{_access=0x014f2678 {DS_DVB_S2 (2)} _data=0x014f2678 "\x2" }	ts::Variable<enum ts::DeliverySystem>
+		frequency	{_access=0x014f2684 {11493750000} _data=0x014f2684 "ð´\x14­\x2" }	ts::Variable<unsigned __int64>
+		polarity	{_access=0x014f2694 {POL_HORIZONTAL (1)} _data=0x014f2694 "\x1" }	ts::Variable<enum ts::Polarization>
+		lnb	{_access=0x014f26a0 {_name=L"Astra" _alias=L"Enhanced" _bands={ size=1 } } _data=0x014f26a0 "Tñ\x3\x1À\x2|\aA" }	ts::Variable<ts::LNB>
+		symbol_rate	{_access=0x014f2700 {21995000} _data=0x014f2700 "øO\x1... }	ts::Variable<unsigned int>
+		inner_fec	{_access=0x014f270c {FEC_2_3 (2)} _data=0x014f270c "\x2" }	ts::Variable<enum ts::InnerFEC>
+		satellite_number	{_access=0x014f2718 {1} _data=0x014f2718 "\x1" }	ts::Variable<unsigned int>
+		modulation	{_access=0x014f2724 {PSK_8 (27)} _data=0x014f2724 "\x1b" }	ts::Variable<enum ts::Modulation>
+		roll_off	{_access=0x014f2784 {ROLLOFF_35 (3)} _data=0x014f2784 "\x3" }	ts::Variable<enum ts::RollOff>
		_allow_short_options	true	bool**

****** Tuner *****
-		theApp.m_pTuner	0x00ea2538 {_duck={_report=0x00f23550 <Keine Informationen verfügbar. Für tsduck.dll wurden keine Symbole geladen.> ...} ...}	ts::Tuner *
+		_duck	{_report=0x00f23550 <Keine Informationen verfügbar. Für tsduck.dll wurden keine Symbole geladen.> _initial_out=...}	ts::DuckContext &
		_is_open	true	bool
		_info_only	false	bool
+		_device_name	L"Hauppauge WinTV 885 Alt BDA Tuner/Demod"	ts::UString
+		_device_info	L""	ts::UString
+		_device_path	L"@device:pnp:\\\\?\\pci#ven_14f1&dev_8880&subsys_c1080070&rev_04#4&29e9d3c5&0&00de#{71985f48-1ca1-11d3-9cc8-00c04f7971e0}\\{b8917e6f-d893-431b-a41c-33c7dc91ba98}"	ts::UString
		_signal_timeout	5000	__int64
		_signal_timeout_silent	false	bool
		_receive_timeout	0	__int64
+		_delivery_systems	{ size=2 }	ts::DeliverySystemSet
		_guts	0x00eb8db0 {...}	ts::Tuner::Guts *

After calling tuner.Start I get the error message

Error: no input DVB signal after 5,000 milliseconds

When I get the ModArgs parameters with

theApp.m_pTuner->getCurrentTuning(*theApp.m_pModArgs, true, *theApp.m_pReportFile);

i get this:
-		modArgs	<Keine Informationen verfügbar. Für tsduck.dll wurden keine Symbole geladen.>	ts::ModulationArgs
+		ts::Object	{...}	ts::Object
+		ts::ArgsSupplierInterface	{...}	ts::ArgsSupplierInterface
+		delivery_system	{_access=0x0117e22c {DS_DVB_S (1)} _data=0x0117e22c "\x1" }	ts::Variable<enum ts::DeliverySystem>
+		frequency	{_access=0x00000000 {???} _data=0x0117e238 "ÌÌÌÌÌÌÌÌ... }	ts::Variable<unsigned __int64>
+		polarity	{_access=0x00000000 {???} _data=0x0117e248 "ÌÌÌÌ... }	ts::Variable<enum ts::Polarization>
+		lnb	{_access=0x00000000 <NULL> _data=0x0117e254 "ÌÌÌÌÌÌÌÌÌÌÌÌÌÌÌÌÌÌÌÌÌÌÌÌÌÌÌÌÌÌÌÌÌÌÌÌÌÌÌÌÌÌÌÌÌÌÌÌÌÌÌÌÌÌÌÌÌÌÌÌÌÌÌÌÌÌÌÌÌÌÌÌÌÌÌÌ... }	ts::Variable<ts::LNB>
+		symbol_rate	{_access=0x0117e2b4 {20000} _data=0x0117e2b4 " N" }	ts::Variable<unsigned int>
+		inner_fec	{_access=0x0117e2c0 {FEC_3_4 (3)} _data=0x0117e2c0 "\x3" }	ts::Variable<enum ts::InnerFEC>
+		satellite_number	{_access=0x00000000 {???} _data=0x0117e2cc "ÌÌÌÌ... }	ts::Variable<unsigned int>
+		modulation	{_access=0x0117e2d8 {QPSK (20)} _data=0x0117e2d8 "\x14" }	ts::Variable<enum ts::Modulation>
+		roll_off	{_access=0x00000000 {???} _data=0x0117e338 "ÌÌÌÌ... }	ts::Variable<enum ts::RollOff>
		_allow_short_options	true	bool

What I have not understood?

lelegard, 2020-12-02:
In your modulation parameters, I see this:
~~~
satellite_number	{_access=0x014f2718 {1} _data=0x014f2718 "\x1" }	ts::Variable<unsigned int>
~~~
Do you have a DiSEqC switch and do you target the second port?

Otherwise, let this field unset (or set to zero) and retry.

RolsimGit, 2020-12-02:
The same result, even if not initialized, or with value 0 or 2.

lelegard, 2020-12-02:
OK, the other two differences with the command line are:

- inner_fec FEC_2_3 (the default is FEC_AUTO)
- modulation PSK_8 (the default is QPSK for DVB-S/S2)

What are the exact characteristics of this transponder? Which satellite is this?

lelegard, 2020-12-02:
Also try `tsp --verbose` on the command which actually grabs the signal to view the actual parameters.

RolsimGit, 2020-12-02:
D:\TSDuck\bin>tsp --verbose -O file test.ts -P until --seconds 20 -I dvb --delivery-system DVB-S2 --lnb Astra --frequency 11493750000 --polarity horizontal --symbol-rate 21995000 --device-name :1

* dvb: using Hauppauge WinTV 885 Alt BDA Tuner/Demod (DVB-S, DVB-S2)
* dvb: tuned to transponder --delivery-system DVB-S2 --frequency 11,493,750,000 --symbol-rate 21,995,000 --fec-inner auto --polarity horizontal --modulation QPSK --pilots off --roll-off 0.35 --lnb Enhanced --satellite-number 0
* dvb: signal locked: yes, strength: 96%, quality: 94%
* dvb: initial input bitrate is 42,584,548 b/s

and the new parameter:
+		delivery_system	{_access=0x026d7130 {DS_DVB_S2 (2)} _data=0x026d7130 "\x2" }	ts::Variable<enum ts::DeliverySystem>
+		frequency	{_access=0x026d713c {11493750000} _data=0x026d713c "ð´\x14­\x2" }	ts::Variable<unsigned __int64>
+		polarity	{_access=0x026d714c {POL_HORIZONTAL (1)} _data=0x026d714c "\x1" }	ts::Variable<enum ts::Polarization>
+		lnb	{_access=0x026d7158 {_name=L"Astra" _alias=L"Enhanced" _bands={ size=1 } } _data=0x026d7158 "Tñ5" }	ts::Variable<ts::LNB>
+		inversion	{_access=0x026d71ac {SPINV_AUTO (1)} _data=0x026d71ac "\x1" }	ts::Variable<enum ts::SpectralInversion>
+		symbol_rate	{_access=0x026d71b8 {21995000} _data=0x026d71b8 "øO\x1... }	ts::Variable<unsigned int>
+		inner_fec	{_access=0x026d71c4 {FEC_AUTO (0)} _data=0x026d71c4 "" }	ts::Variable<enum ts::InnerFEC>
+		satellite_number	{_access=0x026d71d0 {0} _data=0x026d71d0 "" }	ts::Variable<unsigned int>
+		modulation	{_access=0x026d71dc {QPSK (20)} _data=0x026d71dc "\x14" }	ts::Variable<enum ts::Modulation>
+		pilots	{_access=0x026d7230 {PILOT_OFF (1)} _data=0x026d7230 "\x1" }	ts::Variable<enum ts::Pilot>
+		roll_off	{_access=0x026d723c {ROLLOFF_35 (3)} _data=0x026d723c "\x3" }	ts::Variable<enum ts::RollOff>
+		isi	{_access=0x026d7254 {4294967295} _data=0x026d7254 "ÿÿÿÿ... }	ts::Variable<unsigned int>
+		pls_code	{_access=0x026d7260 {0} _data=0x026d7260 "" }	ts::Variable<unsigned int>
+		pls_mode	{_access=0x026d726c {PLS_ROOT (0)} _data=0x026d726c "" }	ts::Variable<enum ts::PLSMode>
		_allow_short_options	true	bool

lelegard, 2020-12-02:
And then what ? Does the application work with the same tuning parameters as the `tsp` command ?

RolsimGit, 2020-12-02:
[image]

It is Astra 19.2 The parameters are from the NIT

RolsimGit, 2020-12-02:
No Stream with the new Parameter

RolsimGit, 2020-12-02:
I have found the reason.

[image]

When selecting the DeviceName I had the following code:

`void CDlgTuner::OnSelchangeCmbTuner()
{
	int i = 0;
	bool erg;

	CComboBox* cmbBox = (CComboBox*)GetDlgItem(IDC_CMB_TUNER);

	i = cmbBox->GetCurSel();

	if (theApp.m_pTuner->isOpen()) theApp.m_pTuner->close(*theApp.m_pReportFile);

	erg = theApp.m_pTuner->open(m_tunerVector[i]->deviceName(), false, *theApp.m_pReportFile);
	...
}`

A new opening of the tuner is apparently not enough. 
The new functional code:

`void CDlgTuner::OnSelchangeCmbTuner()
{
	int i = 0;
	bool erg;

	CComboBox* cmbBox = (CComboBox*)GetDlgItem(IDC_CMB_TUNER);

	i = cmbBox->GetCurSel();

	ts::UString deviceName = m_tunerVector[i]->deviceName();

	if (theApp.m_pTuner->isOpen())
		theApp.m_pTuner->close(*theApp.m_pReportFile);
	
	delete theApp.m_pTuner;

	theApp.m_pTuner = new ts::Tuner(theApp.m_duck, deviceName, false, *theApp.m_pReportFile);
	erg = theApp.m_pTuner->open(deviceName, false, *theApp.m_pReportFile);
	...
}`

lelegard, 2020-12-03:
> A new opening of the tuner is apparently not enough.

OK, I will check.

lelegard, 2020-12-03:
I have made a test closing / reopening the same `ts::Tuner` object and it works.  I have currently only one tuner to test and close / reopen is done on the same tuner device, so the test a bit different. But looking at the source code, I see no reason why a close / reopen should not work.

Test done on Windows 10 20H2, VS 2019, current TSDuck repo state (3.24-2103).

Which version of TSDuck are you using?

RolsimGit, 2020-12-03:
I'm using TSDuck-Win64-3.23-1965.exe. 

I will install the latest one. I report then.

lelegard, 2020-12-03:
I do not think it will change anything. I was wondering if you were using some very old version. But this is not the case.

RolsimGit, 2020-12-03:
hmmm... look here

https://tsduck.io/download/tsduck/

lelegard, 2020-12-03:
Yes, 3.23 is the latest official version. When I say "latest repo", I mean the current state of the source code in the git repo. 

You can find nightly builds of beta versions here: https://tsduck.io/download/prerelease/

But, again I do not think this will solve your issue. There is no significant difference in the way tuners are opened and closed.