tsduck/tsduck #677: What is the release date of Version 3.24-2106

pdaga2, 2020-12-04:
I can see changes regarding version 3.24-2106 in CHANGELOG.txt file, but cannot find this version in tag section. Is there any release date for Version 3.24-2106?

lelegard, 2020-12-04:
Hi @pdaga2 

TSDuck has never planned formal release dates. Being mostly a single-person project, there is no product management team to take care of releases.

I had in mind to release version 3.24 over the next week-end if I have time and if there is no severe issue which would need more work.

Since I have no team and little time to dedicate to this project, there is an automated continuous integration process which tests the code and maintains the web site always at the current code level. This is why you can see the documentation on the web site at the very latest level. The documentation for a given release is inside the installation package of that release.

Thanks to self discipline in the development and full CI automation, the current state of the source code repo is usually as stable as an official release. Many users just pull the repo and rebuild. This is why the web site is always at the same level as the source code. Windows users who don't dare to rebuild may find nightly prereleases in https://tsduck.io/download/prerelease/

lelegard, 2020-12-05:
Just released during the weekend, as promised :)

pdaga2, 2020-12-06:
Thanks a lot for a quick response and new release.

closed 2020-12-06 by lelegard