tsduck/tsduck #678: [Q] Minimal bitrate removing NULL stuff

lars18th, 2020-12-04:
Hi @lelegard ,

I'm adding at the end of the chain a filter to remove the NULL stuff of a TS created from scratch (as in the exemple from section `5.2.21.2 Creating a transport stream from scratch`). So my command ends with `-P filter -n -p 8191 -O ...`. And all works great with the MPE encapsulation, except when the incoming UDP packets have a low bitrate (the encapsulated stream is VBR). So, my question is this: It's possible to remove the NULL stuff only _over a minimum value_? That's remove all NULL packets, except when the bitrate goes below a defined value.

How to achieve that?
Thank you one more time!

lelegard, 2020-12-14:
>  It's possible to remove the NULL stuff only over a minimum value? 

I don't see an easy way to achieve this inside the tsp framework. The obvious way is to have a global view of the TS between two PCR's and remove extraneous packets in an equally distributed way, fixing PCR's and accumulating partial packets time. This is not possible with tsp which processes one packet at a time and cannot go backward. This would require a dedicated tools, similar to tsstuff, but removing null packets, instead of adding them. Or this could be an additional feature of tsstuff.

But I am not sure to understand what you want. According to your description, the output is VBR but with a minimum guaranteed bitrate. It this really what you want? Or do you want a CBR output at the specified value?

lars18th, 2020-12-14:
Hi @lelegard ,

Perhaps you're right and what I need is a new function inside tsstuff. But, I'm not sure. Let me to explain the use case:

> But I am not sure to understand what you want. According to your description, the output is VBR but with a minimum guaranteed bitrate. It this really what you want? Or do you want a CBR output at the specified value?

My objective is this: I'm using the plugin `mpeinject` (with PCR) to encapsulate various RTP streams of different sizes (not TS) inside a TS. What I'm doing is quite simple: `tsp --bitrate <bitrate> -I null ...` to genate a CBR TS, then I inject the required PSI/SI tables, and finally the different `-P mpeinject --local-address ...`. The output is a regular `-O ip <multicast-address> --packet-burst 7 --enforce-burst`.

As you can imagine, the maximum data that can be encapsulated is the CBR of the TS minus the overhead of the tables and the encapsulation (around 1-2%). And in order to not waste space, at the end of the chain I add `-P filter -n -p 8191` before the output plugin. With this configuration, the outcoming is a VBR TS based on the varible birate of the RTP incoming streams.

But the problem is this: when the RTP streams are disabled, or they consume very few UDP datagrams, then the resulting VBR TS has a very low bitrate. And this generates some troubles with the jitter. So what I want is to set a _minimum bitrate_.

Perhaps you can propose this solution (the one that initially I have considered): Increase the PSI/SI tables to generate sufficient traffic to raise the minimum bitrate. And yes, this really works. But you're wasting a lot of data that can be consumed for valid payload. So my request is for this.

Any idea?

lelegard, 2021-01-11:
Please try the latest version of plugin `reduce` (RTFM as usual). In addition to the original rudimentary feature of removing X null packets every Y input packets, it can now target a given bitrate.

To do this, I added a new mode of working for packet processing plugins: the "packet window mode". The traditional mode is the "individual packet mode" and remains the preferred one for reasons which [are explained here](https://tsduck.io/doxy/classts_1_1_processor_plugin.html#details).