tsduck/tsduck #679: Remove Windows "Hack"

SS-FF, 2020-12-05:
https://github.com/tsduck/tsduck/blob/3b50845686528e8c99c2a0c2dc14bdc312353ce8/src/libtsduck/dtv/windows/tsTunerGuts.cpp#L475-L480

This should be removed. I'm not sure how you came to add this, but - based on the assumption that this wasn't about a re-used filtergraph but rather happened across separate TSP process runs - this can only be result of an incorrect driver implementation.

When you think this would still be needed, it should at least be optional but not default behavior.

lelegard, 2020-12-05:
Hi @SS-FF 

This date back to a while in the past. I do not remember exactly which was the tuner model, its driver and the version of Windows. But I do remember that there were inconsistent packets right after a stop/start cycle. I lost a long time tracking that issue and those inconsistent packets were finally from the previous stream (I remember having caught a PAT or PMT packet that was obviously from the previous stream).

I agree with you that this is the result of some incorrect implementation, driver or filter, I cannot decide. But this horrible hack was necessary to make things work (and I will not come back on how much I hate this environment...)

So, if I remove this hack, there is a risk that something goes wrong with someone else using the same tuner (by experience, those drivers are never updated and this kind of bug remains forever).

I know that you are chasing tuning time on #671 but come on, this is only 50 ms, 1/20 second, while you are looking for a 1 or 2 seconds improvement.

At best, I can make it optional and actionable on the command line. But users having issues will probably never have the idea to correlate their imperceptible problems with that option.

SS-FF, 2020-12-05:
I'm more concerned about what the .Flush might do rather than the Sleep. (I haven't debugged and analyzed this yet).

> So, if I remove this hack, there is a risk that something goes wrong with someone else using the same tuner (by experience, those drivers are never updated and this kind of bug remains forever).

Driver updates are very much dependent on the manufacturer. Many are still releasing regular updates, but yes, others don't.
But this bug would manifest in every other DVB application as well, so I can't really believe that it would still exist. 
50ms sounds little - but these things sum up (e.g., BDA device enumeration is expensive and something that I'm going to tweak - another 150ms...).

Finally, let me ask the following question: Would you find a 50ms hack acceptable when it would be in the Linux implementation? 
😆

lelegard, 2020-12-05:
> Would you find a 50ms hack acceptable when it would be in the Linux implementation?

I would put it the other way round. Waiting or polling is always ugly. Cases where ugly stuff is necessary are a symptom of ugliness of the underlying platform. Similarly, for something as simple as reading data from an input device, if you need to develop something as complicated as DirectShow filter, this is the symptom of something going wrong with the platform. I do not prefer Linux _in principle_ (Linux has its own flaws elsewhere). I just notice by experience and experimentation that doing something simple and reliable is much more complicated on Windows than Linux.

SS-FF, 2020-12-05:
That's what I mean: You blame it on DirectShow, even though the blame is on the driver. You consider the 50ms acceptable based on your attitude against DirectShow. That's why I asked, whether you would make the same decision on Linux: Make all use cases suffer from a 50ms delay just because there exists a single faulty driver where you don't even remember the brand and model..

SS-FF, 2020-12-05:
(I'm just challenging you a bit for fun - it's been a long stint for me 😄 )

closed 2022-09-23 by lelegard