tsduck/tsduck #680: [Q] Linux Build: How to disable Dektec and others?

SS-FF, 2020-12-05:
After catching up to the latest version, excluding DekTec doesn't work anymore.

This is about static building on Linux.

I used to have ` STATIC=1 NOPCSC=1 NOCURL=1 NODTAPI=1 NOTEST=1` in the make command, but this does not longer work to exclude DekTec.

How can I exclude DekTec?

Also, there are two plugin folders now:

- src/libtsduck/plugin
- src/tsplugins

I know I can exclude plugins from the latter by adding something like

        export TSPLUGINS := $(filter-out tsplugin_aes, $(TSPLUGINS))

to Makefile.tsduck

How can I exclude plugins from the first folder?

lelegard, 2020-12-06:
> After catching up to the latest version, excluding DekTec doesn't work anymore.

What do you mean by that? NODTAPI means do not use the DTAPI and do not depend on it. But the tools and plugins are still built. Using them reports an error:
~~~
$ tsdektec 
tsdektec: This version of TSDuck was compiled without Dektec support
$ 
~~~

> Also, there are two plugin folders now:
>
> - src/libtsduck/plugin
> - src/tsplugins

The plugins folder is still `src/tsplugins`. Everything that is under `src/libtsduck` is the TSDuck library. A long time ago, it was a flat directory which ended up with more that 1000 files.  So, it was reorganised as a tree of files. The subdirectory `plugin` contains classes pertaining to the plugin management. Please read the `libtsduck/README.txt` file.

If you worked on a very old base line, the code may have substantially changed since then.

SS-FF, 2020-12-06:
> If you worked on a very old base line, the code may have substantially changed since then.

I have started "from scratch" with the latest tsDuck code and I have manually re-applied all the changes that I had made over time.

> What do you mean by that? NODTAPI means do not use the DTAPI and do not depend on it. 

I build with NODTAPI, but then I run `tsp -lall`, I get this:

```
List of tsp input plugins:

  dektec ... Receive packets from a Dektec DVB-ASI or demodulator device
  dvb ...... DVB receiver device input
  file ..... Read packets from one or more files
  fork ..... Fork a process and receive TS packets from its standard output
  hls ...... Receive HTTP Live Streaming (HLS) media
  ip ....... Receive TS packets from UDP/IP, multicast or unicast
  null ..... Generate null packets
  srt ...... Receive TS packets from Secure Reliable Transport (SRT)

List of tsp output plugins:

  dektec ... Send packets to a Dektec DVB-ASI or modulator device
  drop ..... Drop output packets
  file ..... Write packets to a file
  fork ..... Fork a process and send TS packets to its standard input
  hls ...... Generate HTTP Live Streaming (HLS) media
  ip ....... Send TS packets using UDP/IP, multicast or unicast
  srt ...... Send TS packets using Secure Reliable Transport (SRT)

List of tsp packet processor plugins:

  analyze .. Analyze the structure of a transport stream
  file ..... Write packets to a file and pass them to next plugin
  filter ... Filter TS packets according to various conditions
  fork ..... Fork a process and send TS packets to its standard input
  pes ...... Analyze PES packets
  stats .... Report various statistics on PID's and labels
  timeshift  Delay transmission by a fixed amount of packets
  until .... Copy packets until one of the specified conditions is met

```

SS-FF, 2020-12-06:
Just in case you missed it: I'm doing a STATIC build.

lelegard, 2020-12-06:
Roughly, nothing changed. Compiling with NODTAPI, NOCURL, etc. never removed the wrapper C++ classes on top of these external libraries and never removed the corresponding plugins. The idea of the NOxxx options is to remove dependencies on external libraries if you do not want to install them or if you have a problem to get or compile them on such rare flavour of Linux (it happens). But there is also a requirement that applications using the TSDuck library (tsduck tools, plugins or third-party applications) continue to compile normally. For instance, an application using the class `ts::DektecDeviceInfo` shall continue to compile. Otherwise, that would create a mess. But, if the TSDuck library was compiled with NODTAPI, when you run the application, you get the error message I mentioned in my previous post.

Now, one little thing has changed a couple of versions ago about the way plugins are _registered_. From the beginning, for purely technical link-edit reasons, the code of some plugins such as the `dektec` or `srt` plugins was already _inside_ the TSDuck library. The shared library `tsplugin_dektec.cpp` or `tsplugin_srt.cpp` contained nothing but the macro for the registration of the plugin. This was a bit ridiculous to build and install a shared library just to register code which was otherwise already present inside `tsduck.dll` or `libtsduck.so`. So, now, the macro for the registration of those plugins are inside the code of the plugins, in the TSDuck library. The corresponding `tsplugin_xxx` have been removed. 

So, in practice, we now have "built-in self registered plugins" which are always there and traditional plugins from shared libraries which are registered only when the corresponding shared library is loaded (or object file is linked in the case of a static build). Note that the effect of `tsp -l` is to scan and load all `tsplugin_xxx` to list them.

What does this change for you? I understand that you explicitly remove the `tsplugin_dektec` shared library (or static object file in your case). So, previously, you removed the _registration_ of the `dektec` plugin. But the code was there anyway and continues to be there now, except that the registration is now performed internally in the TSDuck library and you can see it with `tsp -l`. Anyway, since you compiled with NODTAPI, using this plugin will result in the error message. And static vs. dynamic build does not change anything regarding the plugin registration.

SS-FF, 2020-12-06:
Hi Thierry, thank you very much for the detailed explanation.

It also explains why there are those two "plugin dirs" now. I had just thought looked at this in a very simple way: Previously, Dektec was not listed as an available plugin and now it is listed => something must be wrong.
But when you say that it's just listed now, even though it's not included, then it's just a bit weird, but I think I can live with it ;-)
Thanks again!

closed 2020-12-06 by SS-FF