tsduck/tsduck #687: [BUG] lib64 error with SRT

enricovittorini, 2020-12-07:
#### Bug description:
when running tsp command this error is retunred:
/tsduck # tsp --version
Error loading shared library libsrt.so.1: No such file or directory (needed by /usr/lib/libtsduck.so)

#### How to reproduce:
get haivistion GIT sources and compile it
deploy alpine docker image, get TSDUCK from git and compile it. then run 

#### Expected behavior:
tsp should run looking at /usr/lib64/ directory

#### Errors and logs:

~~~
/tsduck # tsp --version
Error loading shared library libsrt.so.1: No such file or directory (needed by /usr/lib/libtsduck.so)
Error relocating /usr/lib/libtsduck.so: srt_recvmsg2: symbol not found
Error relocating /usr/lib/libtsduck.so: srt_create_socket: symbol not found
Error relocating /usr/lib/libtsduck.so: srt_bind: symbol not found
Error relocating /usr/lib/libtsduck.so: srt_close: symbol not found
Error relocating /usr/lib/libtsduck.so: srt_setsockflag: symbol not found
Error relocating /usr/lib/libtsduck.so: srt_startup: symbol not found
Error relocating /usr/lib/libtsduck.so: srt_connect: symbol not found
Error relocating /usr/lib/libtsduck.so: srt_getsockflag: symbol not found
Error relocating /usr/lib/libtsduck.so: srt_listen: symbol not found
Error relocating /usr/lib/libtsduck.so: srt_getlasterror_str: symbol not found
Error relocating /usr/lib/libtsduck.so: srt_cleanup: symbol not found
Error relocating /usr/lib/libtsduck.so: srt_send: symbol not found
Error relocating /usr/lib/libtsduck.so: srt_accept: symbol not found
~~~

#### Environment:
 - OS: Linux, Alpine Docker image Latest
 - OS version: 
NAME="Alpine Linux"
ID=alpine
VERSION_ID=3.12.1
PRETTY_NAME="Alpine Linux v3.12"
 - TSDuck full version: 3.24.2109
 - Installation type: built form binary

lelegard, 2020-12-08:
> Error loading shared library libsrt.so.1: No such file or directory (needed by /usr/lib/libtsduck.so)

> tsp should run looking at /usr/lib64/ directory

> PRETTY_NAME="Alpine Linux v3.12"

There is no `/usr/lib64` on Alpine Linux, only `/usr/lib`.
On my Alpine VM:
~~~
$ cat /etc/alpine-release
3.12.1
$ uname -a
Linux vmalpine 5.4.72-0-virt #1-Alpine SMP Mon, 19 Oct 2020 06:22:29 UTC x86_64 GNU/Linux
$ ls -ld /usr/lib*
drwxr-xr-x 46 root root 36864 Nov 14 20:19 /usr/lib
drwxr-xr-x  8 root root  4096 Oct 11 21:14 /usr/libexec
$
~~~

Alpine is like Debian and Ubuntu, it uses `/usr/lib`. The Red Hat family (CentOS, Fedora), on the other hand, uses `/usr/lib64`.

enricovittorini, 2020-12-09:
if you compile srt from https://github.com/Haivision/srt.git, libraries are in /usr/local/lib64
the issue can be solved by setting 
export LD_LIBRARY_PATH=$LD_LIBRARY_PATH:/usr/local/lib64/

lelegard, 2020-12-09:
Hi @enricovittorini 

LD_LIBRARY_PATH can always be used to redirect anywhere. But I think that this an issue with the SRT installation procedure. On a given 64-bit Linux distro, libraries should be installed on the right directory for that specific distro.

This is not as easy as it seems. There are many possible configurations about `/usr/lib64`:
- Does not exist (Alpine).
- Exists but contains a few symbolic links only, `/usr/lib` remaining the standard directory (Ubuntu)
- Contains everything (Fedora).

In TSDuck, this is solved that way in `Makefile.common`, considering that the directory which contains the libc is the reference one for libraries:
~~~
LOCAL_OS   := $(shell uname -s | tr A-Z a-z)
MAIN_ARCH  := $(shell uname -m | $(SED) -e 's/i.86/i386/' -e 's/arm.*/arm/')
SYSPREFIX  ?= $(if $(MACOS),/usr/local,/usr)

USELIB64  := $(if $(subst linux-x86_64,,$(LOCAL_OS)-$(MAIN_ARCH)),,$(if $(wildcard /usr/lib64/libc.so*),true,))
USRLIBDIR ?= $(if $(USELIB64),$(SYSPREFIX)/lib64,$(SYSPREFIX)/lib)
~~~

closed 2022-09-23 by lelegard