tsduck/tsduck #688: [FEAT] Compact XML parameter

lars18th, 2020-12-09:
#### Is your feature request related to a problem?
When running TSDuck over _remote_ hosts you need to pass _all_ the stuff in a command. You have other alternatives, but when passing only a full (large) command line your work is more simple. The problem is when you want to use XML files inside your tsduck command line. Even when the content have a small length, for example in this case: `tsp ... --patch-xml file.xml ...` and "file.xml" contains only:
```
<?xml version="1.0" encoding="UTF-8"?>
<tsduck>
  <NIT>
    <transport_stream>
      <terrestrial_delivery_system_descriptor centre_frequency="474,000,000"/>
    </transport_stream>
  </NIT>
</tsduck>
```

#### Proposed solution:
My proposed solution is to support for all cases when XML files are supported, to read the content of the XML file from the command line in a full string. More or less in this way:
```
tsp ... --patch-xml="<?xml version="1.0" encoding="UTF-8"?><tsduck><NIT><transport_stream><terrestrial_delivery_system_descriptor centre_frequency="474,000,000"/></transport_stream></NIT></tsduck>" ...
```

As you can see the difference is that the parameter supports one space indicating the path of the file, or in another hand an equal character to pass the content of the file. For sure, it will be necessary to support escape encodings besides the UTF-8 to don't have troubles with quotes or other signs.

#### Alternatives:
I don't see any other alternative to this.

#### Additional information:
Scaping support as described here should be necessary: https://en.wikipedia.org/wiki/XML#Escaping

lelegard, 2020-12-09:
I keep this one on the list of enhancements.

Just two remarks:

The equal sign won't make any difference since the two syntaxes `--patch-xml foo.xml` and `--patch-xml=foo.xml` are already accepted and equivalent. But this is not a issue. An XML content shall start with `<?xml`. Thus, if the specified option value starts with `<?xml`, this is an XML content, otherwise this is the name of an XML file.

There is no need for special escapes. Just use the syntax of the shell, keeping in mind that simple and double quotes are equally accepted by the shells and XML attributes. So, any of the two following syntaxes are possible and equivalent, without specific escape:
~~~
--patch-xml='<?xml version="1.0" encoding="UTF-8"?>< ... centre_frequency="474,000,000"/>...</tsduck>' ...
--patch-xml="<?xml version='1.0' encoding='UTF-8'?>< ... centre_frequency='474,000,000'/>...</tsduck>" ...
~~~

lelegard, 2020-12-10:
Done

lars18th, 2020-12-11:
Thank you! :+1:

closed 2020-12-11 by lars18th