tsduck/tsduck #689: Default Region for ATSC

SS-FF, 2020-12-09:
Wouldn't it make sense to set the default region to "US" in case of ATSC tuning?

Of course, there is a very small number of additional regions where ATSC is used, but one thing is for sure: the regular default region is "Europe", and that's in no case an appropriate choice for VHF and UHF band configuration when ATSC tuning is requested.

SS-FF, 2020-12-09:
Maybe just print a warning that ATSC and Europe don't go together..

lelegard, 2020-12-09:
Options such as `--atsc` or `--isdb` indicate a standard, typically for PSI/SI. I prefer to add a synthetic option `--usa` which triggers several options, including ATSC and HF region. This is more consistent with similar options such as `--japan` or `--brazil`.

I keep this on the enhancement list.

SS-FF, 2020-12-09:
I don't care personally. I just thought that it's not ideal for starters when there are too many parameters to set in order to achieve some successful result..

closed 2020-12-09 by SS-FF