tsduck/tsduck #690: [Q] fork plugin without packet streaming

lars18th, 2020-12-09:
Hi @lelegard ,

I want to execute another subprocess when running tsp. This subprocess reads from the multicast TS generated by the tsp process, so the PIPE stream of the packets will never be readed. So, I ask if this can be done at time or not.

Thank you for this impresive project!
Regards.

lelegard, 2020-12-10:
If the subprocess has no direct interaction with tsp (except reading an IP stream), then there is no reason to run it from tsp. Just run it in the background, eg:
~~~
tsp -I ip 230.1.1.1:1234 ... &
tsp .... -O ip 230.1.1.1:1234 
~~~

lars18th, 2020-12-10:
Hi @lelegard ,

> If the subprocess has no direct interaction with tsp (except reading an IP stream), then there is no reason to run it from tsp. Just run it in the background, eg:
> 
> ```
> tsp -I ip 230.1.1.1:1234 ... &
> tsp .... -O ip 230.1.1.1:1234 
> ```

Copied from the Documentation:
```
So, this plugin is redundant with the shell pipe features. However, this plugin is useful when the tsp
process is created from another native application (not a shell script). In that case, it is much easier for
this application to create a simple binary process rather than a shell and its commands.
```

That's my case: From another process I need to launch a tsp process, that requires to interact with another one... _using a multicast stream, and not PIPES_. And what I want is that the child subprocess of the tsp will end when the tsp process ends. All automatically. So, my bet is to use the `fork` plugin to execute the childprocess. However, the current problem is that the plugin "sends" the entire TS over the PIPE. My request is to support to disable this copy of packets.

I know that this _indirect_ use of fork can be considered unorthodox. But from my point of view can be a valid (an easy) method to control child processes from the tsp process.

What you think?

lelegard, 2020-12-10:
When the doc says _"this plugin is redundant with the shell pipe features"_, it means the _pipe_ feature, not simply creating a process. The pipe is the purpose of the `fork` plugin. Without pipe, this plugin is useless.

In fact the real use of the `fork` plugin is when you use it at a packet processing plugin because you can pipe the stream several times at different stages, which is not possible using the shell syntax. The output plugin `fork` is just a copy-cat of the packet processing plugin variant, so it was cheap to provide. Otherwise, it would not be there.

Similarly, the input plugin variant is useless with tsp because the shell syntax can be used. I added the input variant when tsswitch was introduced. This one may require several input pipes, which is not possible with the shell syntax.

> interact with another one... using a multicast stream, and not PIPES. 

This is not an interaction in a sense of the shell. Both processes are completely independent and interact with the network, without any kind of "connection". A pipe is a special interaction because it has no explicit name (it must be created and inherited from inside the partner processes) and is a strong connection.

> And what I want is that the child subprocess of the tsp will end when the tsp process ends. 

That won't work. On the contrary, tsp will wait for the subprocess to terminate (unless you specify `--nowait`). The natural scenario is that tsp passes the stream to the pipe, closes the pipe on end of stream and _waits for the termination of the subprocess_. The subprocess does what it has to do with the input stream and most probably terminates after the end of input. At that moment, the parent tsp will complete. The `--nowait` option is used when, for some reason, the subprocess is not expected to terminate on end of input. But, in no circumstances, tsp will force the subprocess to terminate.

To force the subprocess to quit when tsp terminates, I would suggest something like this:
~~~
tsp -I ip 230.1.1.1:1234 ... &
subpid=$!
tsp .... -O ip 230.1.1.1:1234 
kill $subpid 
~~~

To build any kind of simple to complex scenario, you need a shell. tsp is not a shell, it is not there to create processes without "transport stream added value". Oh, and by the way, it does make coffee either :)

lars18th, 2020-12-11:
Hi @lelegard ,

Thank you for your detailed explanation. I can see your extensive experience. And I not doubt that you have a lot of reason.

Anyway, the problem that I want to solve is this: Make a call _from another process_ (mainly from a remote machine) to start tsp _and_ another process at the same time without using a shell. And when the tsp process ends (or it's killed) then the other process is then stoped. I'm sure you can see the use case. So, any option to achieve this with TSDuck? If not, please do not bother with that question. I'm just looking for options, and your feedback is very valuable.

Regards.

lelegard, 2020-12-11:
> Make a call from another process (mainly from a remote machine) to start tsp and another process at the same time without using a shell.

So, you want to play with some combination of processes and interactions between them. This is the exact description of a shell. So, you want tsp to implement the features of a shell. My answer is no, use a real shell.

> And when the tsp process ends (or it's killed) then the other process is then stoped.

This is exactly what is done in the example in my previous post:
- The subprocess is run in the background (and we keep its pid in a variable).
- The tsp process is run synchronously (wait for its termination).
- When tsp terminates (voluntarily or by accident), the subprocess is terminated.

Just encapsulate this sequence in a shell script and run it remotely. If you use SSH, use an scp command first to push the script, then an ssh command to run it. Or maybe it could be run directly as input to ssh but I am not sure, you have to test.

lars18th, 2020-12-11:
Hi @lelegard ,

I appreciate your response and accept your point of view. However, I do not want, and I can't, use a shell for this purpose. I will look for another option. In any case, there seems to be no sense in reinventing the wheel. :wink: 

Regards.