tsduck/tsduck #692: [Q] About SRT

lars18th, 2020-12-12:
Hi,

I have some questions about SRT in TSDuck. Please, help to found the correct response:

- The TSDuck, when compiled with the SRT input/output modules (`tsplugin_srt.so` exists), is compiled with the library statically o it uses the `libsrt.so` installed in the system? I don't know how to select static or dynamic linking of this library when compiling TSDuck.
- How to know the version of the library used inside TSDuck? I feel the help of the module will need to print the version, but I don't see it.
- When using the SRT as the _input_ for the TSDuck chain, but requiring to start the connection to a remote server (the "caller" mode with the "srt-live-transmit" tool), which parameters I need to use? I can't found the solution in the documentation, as it indicates only the "listener mode" for the input and the "caller mode" for the output.

If anyone can answer, I would be very grateful to him.

lelegard, 2020-12-12:
TSDuck is built with whatever form of SRT is present on the system on which it is built.

On Linux and macOS where SRT comes as a binary package from the official repos, it is built against the shareable library by default, as result of the `-lsrt` option. Using the `-static` option will probably try to use `libsrt.a` if present on the system.

Using the lastest version on Fedora:
~~~
$ ldd /usr/lib64/libtsduck.so | grep srt
	libsrt.so.1 => /lib64/libsrt.so.1 (0x00007f071c425000)
~~~

On Windows, to avoid the usual DLL hell of open source libraries, SRT is linked statically.

The version of libsrt which is used can be seen as follow: 
~~~
$ tsp --version=all
tsp: TSDuck - The MPEG Transport Stream Toolkit - version 3.24-2109
Built Dec  4 2020 - 22:43:22
Using GCC 10.2.1, C++ std 2011.03
Web library: libcurl: 7.71.1, ssl: OpenSSL/1.1.1h-fips, libz: 1.2.11
SRT library: libsrt version 1.4.1
Dektec: DTAPI: 5.43.0.147
~~~

Concerning how to use SRT connections, I let others respond. I am not an expert in SRT. The SRT plugins were a contribution from a user who needed SRT. I am just integrating it and make sure it compiles on new versions.

lars18th, 2020-12-12:
Hi @lelegard ,

Thank you! I'll use the cmd `tsp --version=all`. It's the command that I searching for it.

Regarding the use of the SRT plugins inside TSDuck I hope @spnngl want to comment something.

Regards.

spnngl, 2021-03-23:
Hi

the srt implementation in tsduck for now supports only listener mode for input and caller mode for output as documented in https://tsduck.io/download/docs/tsduck.pdf 

So if you want to transmit something with caller mode use tsduck srt output plugin.

A third mode is also available, rendezvous, which kinda mix both of previous ones.

lars18th, 2021-03-23:
Hi @spnngl ,

> the srt implementation in tsduck for now supports only listener mode for input and caller mode for output
> So if you want to transmit something with caller mode use tsduck srt output plugin.
> A third mode is also available, rendezvous, which kinda mix both of previous ones.

What I personally want to do with SRT and TSDuck is operate in this scenario: 
In a central network point (the datacenter) with full Internet connection , I want to run a `srt-live-transmit` daemon with TWO open end-points listening. One end-point is the INPUT stream and the second is the OUTPUT stream. So the idea is that a "remote" TSDuck _sends_ to the central datacenter; while another second "ingest receiver" TSDuck _receives_ from the central datacenter. This architecture is to enforce strict security policies, as only TSDuck clients can connect to the SRT listener server in the central point. However, in this case one TSDuck uses the SRT output module for _sending_. And that's ok because it uses the caller mode. But for the other, it needs to connect to the central server using the SRT input module, but in this case with the caller mode too.

Any solution for this environment?
Thank you!

spnngl, 2021-03-24:
It is possible. Most of the intelligence of SRT in tsduck is in tsSRTSocket.{h,cpp}, you'd have just to add support for missing modes in tsSRT{In,Out}putPlugin and add an input `--mode` to those plugins.

PS: I inverted modes in my previous response, tsSRTOutputPlugin supports listener & rendezvous, tsSRTInputPlugin supports caller & rendezvous

lars18th, 2021-03-24:
Hi @spnngl ,

Thank you for the explanation. I hope someone will update then the tsduck SRT plugins to support all three modes.

Regards.

lelegard, 2021-03-26:
> I hope someone will update then the tsduck SRT plugins to support all three modes.

I did... The input and output `srt` plugins now use the same options: `--caller` and `--listener`. Specifying both means rendezvous mode. The previous parameters and `--rendezvous` option are still there for compatibility but no longer documented in the user's guide.

Please test this, I have not tested everything.