tsduck/tsduck #693: [Q] How accurate is the info obtained from ContinuityAnalyzer?

bertothunder, 2020-12-14:
Sorry if the question itself looks weird, but I'm having a small issue using the ContinuityAnalyzer in a plugin we are developing to measure ETR 290 priority 1 variables (some of them, at least).

 To measure Continuity Errors, ETR standard mentions:

- Incorrect packet order
- a packet occurs more than twice
- lost packet

Checking `feedPacketInternal` from ContinuityAnalyzer, this sounds like exactly what it's doing. So I added one into the plugin, passed the PIDs to monitor, and each time the plugin receives a new packet, it's feeding the continuityAnalyzer instance, checking the result from ContinuityAnalyzer.feedPacket() and, if false, increasing an internal statistic.

 So far, so good, everything seems to work as expected, getting tsp to listen to a stream, going through the plugin, and we have an about average 3 continuity errors each 60-90 secs of video.  Getting the stream through `analyze` plugin, shows a similar number of unexpected discontinuities.

 However, and here comes the weird part, we have two probes running in the same network, monitoring ETR 290 (Brightec) and one StreamXpert instance pointed out to the same stream. And none of them shows any continuity errors at all, not an alarm, nothing.

 After checking twice source code for `ContinuityAnalyzer`, I'm quite sure the code is right and according to the standard, but I can't explain completely why is this happening. To me, the problem could be coming in the exact part that the analyzer is checking the cc order, this one:

```
            // Compute expected CC for this packet.
            const uint8_t good_cc_in = has_payload ? ((last_cc_in + 1) & CC_MASK) : last_cc_in;
            const uint8_t good_cc_out = has_payload ? ((state.last_cc_out + 1) & CC_MASK) : state.last_cc_out;

            if (cc != good_cc_in) {
                if (_display_errors) {
                    // Display a specific message depending on the error.
                    if (!has_payload && cc == ((last_cc_in + 1) & CC_MASK)) {
                        _report->log(_severity, u"%s, incorrect CC increment without payload", {linePrefix(pid)});
                    }   
                    else {
                        _report->log(_severity, u"%s, missing %d packets", {linePrefix(pid), MissingPackets(last_cc_in, cc)});
                    }   
                }   
                _error_count++;
                result = false;
            }   
```

I wonder if we might have a network issue in the machine running tsp that could be causing the packets to arrive in a wrong order? Any ideas to check or something I might be doing wrong, missing additional checks that I'm not doing, maybe by bad assumption?

lelegard, 2020-12-14:
Hi @bertothunder 

The CC checking rules are sometimes complicated by weird specific cases. It is possible that the ContinuityAnalyzer class has missed one such weird rule that professional monitoring tools implement correctly. If the CC errors appear as often as every 60 or 90 seconds, it should be easy to catch a sample. Just add a `-P file` plugin immediately before your plugin and stop it when you see the error. Then, use `-P continuity` on the captured file to check that you still see a discontinuity. Then investigate this specific one (or post here a TS fragment containing the discontinuity).

> I wonder if we might have a network issue in the machine running tsp that could be causing the packets to arrive in a wrong order? 

Maybe. If you receive UDP multicast, you should have the usual 7 TS packets per UDP datagram (you can easily check this using Wireshark). Using `-P continuity` or your plugin, you can see the number of "missing packets". A missing count of 7 packets almost always indicate the loss of one UDP datagram.

bertothunder, 2020-12-14:
Hi @lelegard ,

 thanks for the tips. `-P continuity` running on the same stream shows this:

```
tsp -I ip x.x.x.x:8208 -P continuity -O drop
* continuity: packet index: 502,047, PID: 0x03E9, missing 6 packets
* continuity: packet index: 502,048, PID: 0x05DD, missing 5 packets
* continuity: packet index: 502,049, PID: 0x07D1, missing 7 packets
* continuity: packet index: 502,051, PID: 0x0FA1, missing 8 packets
* continuity: packet index: 502,052, PID: 0x0DAD, missing 4 packets
* continuity: packet index: 502,059, PID: 0x0515, missing 1 packets
* continuity: packet index: 502,090, PID: 0x07DC, missing 1 packets
* continuity: packet index: 502,098, PID: 0x1004, missing 1 packets
* continuity: packet index: 502,099, PID: 0x0BC2, missing 1 packets
* continuity: packet index: 502,110, PID: 0x06A5, missing 1 packets
* continuity: packet index: 502,117, PID: 0x03EA, missing 1 packets
* continuity: packet index: 502,128, PID: 0x05E0, missing 1 packets
* continuity: packet index: 502,143, PID: 0x0DAE, missing 1 packets
* continuity: packet index: 502,152, PID: 0x07D4, missing 1 packets
* continuity: packet index: 502,159, PID: 0x0579, missing 1 packets
* continuity: packet index: 502,160, PID: 0x09CF, missing 1 packets
* continuity: packet index: 502,228, PID: 0x0DB1, missing 1 packets
* continuity: packet index: 523,817, PID: 0x0FA1, missing 2 packets
* continuity: packet index: 523,819, PID: 0x1004, missing 2 packets
* continuity: packet index: 523,820, PID: 0x05DD, missing 11 packets
* continuity: packet index: 523,821, PID: 0x07D1, missing 9 packets
* continuity: packet index: 523,822, PID: 0x03E9, missing 3 packets
* continuity: packet index: 523,823, PID: 0x0DAD, missing 4 packets
* continuity: packet index: 523,824, PID: 0x0579, missing 2 packets
* continuity: packet index: 523,825, PID: 0x0BC2, missing 2 packets
* continuity: packet index: 523,830, PID: 0x044D, missing 2 packets
* continuity: packet index: 523,835, PID: 0x03ED, missing 1 packets
* continuity: packet index: 523,838, PID: 0x0641, missing 2 packets
* continuity: packet index: 523,841, PID: 0x0012, missing 2 packets
* continuity: packet index: 523,842, PID: 0x0DB0, missing 1 packets
* continuity: packet index: 523,843, PID: 0x0515, missing 2 packets
* continuity: packet index: 523,853, PID: 0x0DB1, missing 1 packets
* continuity: packet index: 523,855, PID: 0x1005, missing 2 packets
* continuity: packet index: 523,863, PID: 0x04B1, missing 1 packets
* continuity: packet index: 523,869, PID: 0x05DF, missing 2 packets
* continuity: packet index: 523,874, PID: 0x07DA, missing 2 packets
* continuity: packet index: 523,875, PID: 0x03EB, missing 1 packets
* continuity: packet index: 523,882, PID: 0x0DAF, missing 1 packets
* continuity: packet index: 523,886, PID: 0x07D4, missing 1 packets
* continuity: packet index: 523,896, PID: 0x07DC, missing 3 packets
* continuity: packet index: 523,897, PID: 0x07D2, missing 2 packets
* continuity: packet index: 523,899, PID: 0x07D3, missing 2 packets
* continuity: packet index: 523,900, PID: 0x0DAE, missing 2 packets
* continuity: packet index: 523,901, PID: 0x05E0, missing 2 packets
* continuity: packet index: 523,902, PID: 0x03EA, missing 3 packets
* continuity: packet index: 523,904, PID: 0x09CF, missing 2 packets
* continuity: packet index: 523,926, PID: 0x05E1, missing 1 packets
* continuity: packet index: 523,943, PID: 0x06A5, missing 2 packets
* continuity: packet index: 524,043, PID: 0x03EC, missing 1 packets
```

Since this stream is multiprogram, I just went over a video-only PID.

````
tsp -I ip x.x.x.x:8208 -P continuity -p 1501 -O drop
* continuity: packet index: 224,587, PID: 0x05DD, missing 1 packets
* continuity: packet index: 871,477, PID: 0x05DD, missing 15 packets
* continuity: packet index: 1,184,831, PID: 0x05DD, missing 6 packets

````

lelegard, 2020-12-14:
Could you capture the stream in a file? Thus we could verify that the discontinuities are real. It they are real and the professional tools do not see them, then UDP datagrams must be lost on your machine.

lelegard, 2020-12-14:
Looking at the packet index of the discontinuities, it seems that several PID's report a discontinuity at the same point in the TS at regular intervals. Quite a lot in fact, more that every 60-90 seconds. This is most probably an issue of UDP packet loss on the system. You could try to increase the UDP buffer size in the input ip plugin. Is this a Linux or Windows machine? Is it particularly loaded?

bertothunder, 2020-12-14:
Yes, I was doing exactly that, and it shows the discontinuities are real:

```
tsp -I file /opt/data/output.ts -P continuity -p 1501 -O drop
* continuity: packet index: 72,496, PID: 0x05DD, missing 2 packets
* continuity: packet index: 111,195, PID: 0x05DD, missing 4 packets
* continuity: packet index: 275,377, PID: 0x05DD, missing 11 packets
* continuity: packet index: 402,025, PID: 0x05DD, missing 3 packets
```

I just recalled that I never mentioned that I was testing this inside a docker container connected to the stream, I'll re-check from a bare metal machine, I'm also suspecting the UDP packet loss must be playing a crucial role hjere.

bertothunder, 2020-12-14:
Ok, so best news @lelegard . Right now, I'm able to confirm that I can't see any further continuity errors. Even running tsp from inside a docker container.

The tips you gave me above helped a lot to focus myself in the UPD / networking side, and started checking the behavior with different network configuration settings. Right now, this works for us:

LINUX: 

net.core.rmem_max = 33MB
net.core.rmem_default = 1MB
net.core.somaxconn = 128

ETHERNET (via ethtool):

Used ethtool to check the maximum value for RX ring buffer (`ethtool -g <iface>`). In this case, the host system had a maximum value of 2047, so I set it up with such a value (`ethtool -G <iface> rx <max_value>`).

As said, this is now working like a charm, I don't see any further continuity errors after an hour (it was above 300 before). I'll leave it running for some more hours.

bertothunder, 2020-12-14:
As an update, 33M packets afterwards, only 1 continuity error, which is consistent with the rest of the monitoring elements checking the same stream. 

The ticket can be closed. Thanks again @lelegard .

closed 2020-12-14 by bertothunder