tsduck/tsduck #695: [Q] Build error on Centos 7.9

antoktm, 2020-12-15:
Hi, 

I've encountered compilation error on Centos 7, on latest version and 3.24-2109, as following : 

```
...
  [CXX] dtv/tables/tsAIT.cpp
dtv/tables/tsAIT.cpp: In constructor ‘ts::AIT::AIT(ts::DuckContext&, const ts::BinaryTable&)’:
dtv/tables/tsAIT.cpp:62:9: error: no matching function for call to ‘ts::AIT::AIT()’
     AIT()
         ^
dtv/tables/tsAIT.cpp:62:9: note: candidates are:
In file included from dtv/tables/tsAIT.cpp:30:0:
dtv/tables/tsAIT.h:112:9: note: ts::AIT::AIT(ts::DuckContext&, const ts::BinaryTable&)
         AIT(DuckContext& duck, const BinaryTable& table);
         ^
dtv/tables/tsAIT.h:112:9: note:   candidate expects 2 arguments, 0 provided
dtv/tables/tsAIT.h:98:9: note: ts::AIT::AIT(const ts::AIT&)
         AIT(const AIT& other);
         ^
dtv/tables/tsAIT.h:98:9: note:   candidate expects 1 argument, 0 provided
make[2]: *** [/home/foo/tsduck-3.24-2109/bin/release-x86_64-localhost/objs-libtsduck/tsAIT.o] Error 1
make[1]: *** [default] Error 2
make: *** [default] Error 2
````
gcc version is 4.8.5
binutils version is 2.27-44
kernel 3.10.0-1160.6.1.el7.x86_64

Am I missing something?

lelegard, 2020-12-15:
Hi @antoktm 

Strange. I no longer have any CentOS 7.x (the latest one is 8.3) and no compiler as old as gcc 4.8. So, I cannot reproduce this problem now. 

But TSDuck used to compile on CentOS 7 and definitely with gcc 4.8.5 and the `AIT` class is not recent, it used to compile. Second, the "no matching function for call" error is incorrect because if obviously matches the default constructor which is just above in the same source file.

This C++ construction is not particularly unusual, it is used everywhere in TSDuck, it used to compile with gcc 4.8.x and it currently compiles on many different versions of gcc, clang and Visual Studio.

Did you modify the source code in any way? Can you build it on other Linux versions?

lars18th, 2020-12-15:
Hi @antoktm ,

In my debian 7 I compile TSDuck with the GCC 4-9 backported without troubles. Perhaps you can try to update the compiler from the 4.8 to 4.9.

Here my command to compile TSDuck with the not stock compiler (you can side install it):

`./configure CC=/usr/lib/gcc-4.9-backport/bin/gcc GCC=/usr/lib/gcc-4.9-backport/bin/gcc CXX=/usr/lib/gcc-4.9-backport/bin/g++ AR=/usr/lib/gcc-4.9-backport/bin/gcc-ar`
`make -j 4`

I hope it helps.
Regards.

lelegard, 2020-12-15:
Hi,

I quickly setup a CentOS 7.9 VM and notice the same issue. Weird. I am sure to have compiled TSDuck in the past on CentOS 7.x (maybe not 7.9) and gcc 4.8.x (maybe not 4.8.5). So, it must be a very local gcc bug which must have been fixed right away and only 4.8.5 has it. We don't see any issue with other versions of gcc, neither with clang nor Visual Studio.

Anyway, it was simple and not too dirty to slightly change the source code to avoid that issue. It now compiles on CentOS 7.9 with gcc 4.8.5. You have to pull the latest repo or get an archive of it using the link https://tsduck.io/download/source

antoktm, 2020-12-16:
> Hi,
>
> I quickly setup a CentOS 7.9 VM and notice the same issue. Weird. I am sure to have compiled TSDuck in the past on CentOS 7.x (maybe not 7.9) and gcc 4.8.x (maybe not 4.8.5). So, it must be a [...]

Hi,

Confirmed that the latest repo builds fine. 

Thanks !

closed 2020-12-16 by antoktm

dp289m, 2021-01-14:
Same error, same gcc, same latest repo.

```
[tsduck-tsduck-3b50845]$ cat /etc/redhat-release
CentOS Linux release 7.9.2009 (Core)

[tsduck-tsduck-3b50845]$ gcc --version
gcc (GCC) 4.8.5 20150623 (Red Hat 4.8.5-44)
Copyright (C) 2015 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```
```

[tsduck-tsduck-3b50845]$ make
...
  [CXX] dtv/tables/tsAIT.cpp
dtv/tables/tsAIT.cpp: In constructor ‘ts::AIT::AIT(ts::DuckContext&, const ts::BinaryTable&)’:
dtv/tables/tsAIT.cpp:62:9: error: no matching function for call to ‘ts::AIT::AIT()’
     AIT()
         ^
dtv/tables/tsAIT.cpp:62:9: note: candidates are:
In file included from dtv/tables/tsAIT.cpp:30:0:
dtv/tables/tsAIT.h:112:9: note: ts::AIT::AIT(ts::DuckContext&, const ts::BinaryTable&)
         AIT(DuckContext& duck, const BinaryTable& table);
         ^
dtv/tables/tsAIT.h:112:9: note:   candidate expects 2 arguments, 0 provided
dtv/tables/tsAIT.h:98:9: note: ts::AIT::AIT(const ts::AIT&)
         AIT(const AIT& other);
         ^
dtv/tables/tsAIT.h:98:9: note:   candidate expects 1 argument, 0 provided

```

lelegard, 2021-01-14:
You are not using the latest repo. The intially reported error was on the same source code line. This line is correct and compiled by all other compilers and more recent versions of GCC. I _modified_ this source code line as a workaround for the GCC 4.8 bug. Your report shows the _previous_ source code, not the _modified_ one. This is why you do not have the latest repo.

dp289m, 2021-01-14:
tsduck-tsduck-3b50845 << that was the latest I got from this url: https://tsduck.io/download/source

lelegard, 2021-01-14:
This URL points to the source code of the _latest official release_: `tsduck-tsduck-v3.24-2109-0-g3b50845.zip`, release 3.24-2109 as the archive name says. You need to get the latest git _repo_. There, you will get the intermediate version 3.25-2172 as of today.

The source code workaround for the GCC bug was done after the latest official release.

dp289m, 2021-01-14:
That worked, thanks!