tsduck/tsduck #696: [FEAT] tspcontrol: suspend input/output plugins

lars18th, 2020-12-16:
#### Is your feature request related to a problem?
That's part of the tsp command that I'm using:
- `tsp --bitrate <bitrate> -I null ... -P mpeinject --local-address ... -O srt <remote-address>:<port>`

And the objective of this command is quite simple: I want to use TSDuck to encapsulate some RTP streams inside a TS that will be sended to a remote server using the SRT protocol.

However, I want to _control_ this processing from an external orchestrator. As here the TS with the encapsulated data will work as a _tunneller_. Then I have defined these STATES for the TSP process:

- RUNNING: When encapsulating the RTP streams inside the TS and sending it over the SRT connection.
- ACTIVE: When the SRT is stablished (the output plugin is "on") but no packets are created (the input plugin is "off").
- IDLE: When all plugins are "off", but the entire TSDuck chain is instantiated and the process is active and not running.

So, my idea is quite simple: when the orchestation starts the `tsp` process is launched and it pass to the _IDLE_ state. Then at some point it receives a message to pass to _ACTIVE_, waiting for the point to start for streaming when receiving the _RUNNING_ message. Futhermore, multiple status changes are possible between IDLE<-->ACTIVE<-->RUNNING until at some time the orchestrator will stop/kills the process.

#### Proposed solution:
Two issues limits this use of the `tsp` process with the `tspcontrol` tool:
1. The Input plugin can't be suspended. So I suggest to add the funcitonality to interrupt it and flush all packets until they arrive to the output (or optionally clear all packets). After that the entire processing chain pass to an idle o sleep mode, until the input plugin is restarted.
2. The Output plugin can only be "on" or "off", but not in "prepared" state. So I suggest to add a third state that indicates: "the plugin is ready, but not accepting/passing packets".

#### Alternatives:
The only alternative is to instantiate/kill the entire `tsp` process from the orchestrator. But this generates a huge overhead if you start/stop in small periods. So it's preferable a more fine-grained control.

#### Additional information:
I hope these enhancements can be implemented without huge changes in the architecture and/or the source code.

lelegard, 2020-12-16:
> The Input plugin can't be suspended. So I suggest to add the funcitonality to interrupt it and flush all packets until they arrive to the output (or optionally clear all packets). After that the entire processing chain pass to an idle o sleep mode, until the input plugin is restarted.

In the current implementation, the input plugin cannot be suspended for a reason: there is no unique satisfactory definition of "suspending" the input plugin. Several alternatives were considered and none was without trouble:
- Stop reading packets. Problem: connection-oriented plugins would block the source with potential cascading effects.
- Continues reading packets but dropping them. Problem: CPU-bound input plugins (generators) or disk-bound plugins (files) would saturate the CPU and/or the disk.
- Stop the input plugin and restart it later. Problem: off-line sources such as files would restart at the beginning with is not the definition of suspend / resume.

The actual issue is that the input plugin is the one which creates the tempo. Other plugins just see the passing packets and may ignore them without affecting the whole process.

> The Output plugin can only be "on" or "off", but not in "prepared" state. So I suggest to add a third state that indicates: "the plugin is ready, but not accepting/passing packets".

When the output plugin is suspended, it is ready (started, opened or initialized whatever needs to be) but does not accept packets (they are dropped).

lars18th, 2020-12-16:
Hi @lelegard ,

Thank you for your response. I comment the two topics in reverse order:

> When the output plugin is suspended, it is ready (started, opened or initialized whatever needs to be) but does not accept packets (they are dropped).

That's de _current_ behaviour... all incoming packets to the output are dropped. However, what I need is the control over the "connected" and "disconnected" states of the output. In this case the SRT output. To understand what I request, please think on an hypotetical HTTP output plugin (non existent at time) to illustrate the use case. This output plugin can listen for incoming TCP connections, and when some client connects it acts as a simple HTTP server that to any request responds with an infinite data streaming with a MIME type `video/MP2T`. And the streaming data are the TS packets of the chain. And if more than one client connects, they receive a copy of the same packets. Until now you can see that this HTTP output plugin can be "listening" for incoming connections or not. The first is the "on" state and the second the "off" state, with independence of the running status of the chain. If you extrapolate this example to the SRT output plugin, what I want is to control the "connection status" of the SRT protocol. That's all.

> In the current implementation, the input plugin cannot be suspended for a reason: there is no unique satisfactory definition of "suspending" the input plugin. Several alternatives were considered [...]

Before writing my request I feel that this will be not trivial to implement, because then you've already implemented it. Nevertheless, what I request is the _thrid_ option. This solves my problem at all. If it generates some side-effects, then almost two options can be considered: 1) Document the limitations. For example, you can comment that with files inputs the file pointed is reseted to the begin when the input plugin is restarted. 2) The other option is target the edge cases with some logic. For example, with file inputs you can save the offset of the file, and when restarting the input plugin then reopen the file, set the reading buffer pointer, and if the file is not changed and the pointer is valid, then start reading from this point (and if not, then reset to the start of the file).

So please, consider to add this enhacement. Or almost comment here about the different options.
Thank you!

lelegard, 2020-12-16:
For the output plugin, what you explain is specific to the SRT plugin. It cannot be described in tsp packet management terms.

For the input plugin, I explained that 3 options were initially considered but then rejected because of their respective troubles. Your answer is I want option number 3 and please document the corresponding troubles as limitations.

In both cases, these are requests which may be legitimate in the context of your own project but which can hardly be understood as beneficial for the TSDuck project and its users in general.

lars18th, 2020-12-16:
Hi @lelegard ,

> For the output plugin, what you explain is specific to the SRT plugin. It cannot be described in tsp packet management terms.

OK. I'll need to experiment more with this output then.

> For the input plugin, I explained that 3 options were initially considered but then rejected because of their respective troubles. Your answer is I want option number 3 and please document the corresponding troubles as limitations.

Thank you. But it's quite difficult to see the reason to have rejected the option 3. I feel the side-effects are minimal... or not?

> In both cases, these are requests which may be legitimate in the context of your own project but which can hardly be understood as beneficial for the TSDuck project and its users in general.

I agree that perhaps it's difficult to see my request in a different context that will be beneficial for other users. However, please note this:
- The remote control, without the input plugin control is in fact doing the solution of the option 2 that you described. With the help of other plugins (`filter`) it's trivial to filter out all packets with remote control (you only need to put it after the input, and filter all pids, and disable it with the `tspcontrol` tool when you want to pass all traffic). So this option exists at time!

So what I ask is to add the functionality of "enable option 3" _optionally_. If the effort to do so is enormous, then I understand your position. But if not, please, then consider it.

In any case, thank you for this invaluable toolkit!