tsduck/tsduck #697: [Q] Compiling with non-default compiler: and libraries?

lars18th, 2020-12-18:
Hi @lelegard ,

I see the enhancement of defining the `/bin` directory for the compiler in a recent commit: https://github.com/tsduck/tsduck/commit/90864002d59c788ca88eebf032242185a3d9a56d
This will make more easy the compilation process as you (we) don't need to pass the full path of the compiler/linker/etc.
Thank you!

And now I ask: It will be posible to compile (optionally) fixing the path of the used libraries? My problem is this: when compiling inside Debian 7 with the backport GCC-4.9, the libaries of the shared objects used by the non-standard compiler are installed in a different path that the stock libraries. Then to execute my compiled TSDuck tools it's required to first set the LD_LIBRARY_PATH and the PATH. So it will be possible to set in the `configure` a non-default path for the libraries that use the TSDuck binaries?

This will simplifies a lot the use of different versions of the TSDuck inside the same computer.

Thank you in advance for your response!

lelegard, 2020-12-18:
I did this after issue #695, a build error with an old compiler (just six _major_ versions behind the current one...) I quickly installed an old CentOS 7 VM to characterize the issue. But I cannot keep too many old VM's and I do not want to spend time reinstalling new ones all the time. There is not always a pre-built back-port of GCC on a given distro. So, the natural way is to rebuild specific old versions of GCC on a given system and keep them. I used my regular CentOS 8.3 VM,, the most recent CentOS but the distro with the oldest software components, and rebuilt GCC 4.8.5. In case of other issue, I suppose I will be able to rebuild any other oldish version of GCC.

Having built GCC 4.8.5 in `/opt/gcc-4.8.5`, I build TSDuck that way:
~~~
GCCVERSION=4.8.5
GCCROOT=/opt/gcc-$GCCVERSION
GCCBIN=$GCCROOT/bin
GCCLIB=$GCCROOT/lib64

make BINDIR_SUFFIX=-gcc485 NOSRT=1 \
     CPP=$GCCBIN/cpp CC=$GCCBIN/gcc GCC=$GCCBIN/gcc CXX=$GCCBIN/g++ LD=$GCCBIN/gcc AR=$GCCBIN/gcc-ar \
     LDFLAGS_EXTRA="-L$GCCLIB -Wl,-rpath -Wl,$GCCLIB" \
     -j10 test
~~~
The `NOSRT` option is required because the standard libsrt on this distro uses a level of std C++ ABI which is not available with gcc 4. SRT should be recompiled with gcc 4 to be able to be used from that build of TSDuck.

The rpath linker option keeps track of the library location:
~~~
$ ldd tsp | grep /opt
	libstdc++.so.6 => /opt/gcc-4.8.5/lib64/libstdc++.so.6 (0x00007f9808bd2000)
	libgcc_s.so.1 => /opt/gcc-4.8.5/lib64/libgcc_s.so.1 (0x00007f9808212000)
$ 
~~~

lars18th, 2020-12-18:
OK. Thank you!

In any case in Debian 7 exists the backport of the GCC 4.9. I installed in this way, as the default 6.3 is not supported by TSDuck. And please, try to continue supporting the compilation in older kernels and/or with old compilers. The "rolling release model" used in current Linux distributions is a no-go in a lot of scenarios.

I close this issue.

lelegard, 2020-12-19:
I just rebuilt gcc 6.3.0 on my CentOS 8 VM (same principle as for 4.8.5) and TSDuck compiles with it.

lelegard, 2020-12-19:
If anyone is interested in scripts to rebuild old versions of GCC, mine are here: https://github.com/lelegard/gcc-build

lars18th, 2020-12-19:
Hi @lelegard ,

Some comments and one question.
But before: Thank you for continuning the support of old/ancient/non-recent compilers/kernels! :+1:

- I need to confirm that current TSDuck compiles fine with GCC 6.3 in Debian 7.7 with kernel 3.2.
- It compiles too with minimal changes in Debian 7.11 with GCC 4.9 (backported) with the same kernel 3.2.

For the last I need to use this patch (please, merge it):
```
diff --git a/src/libtsduck/dtv/linux/tsTunerGuts.cpp b/src/libtsduck/dtv/linux/tsTunerGuts.cpp
index 6af4e72..2584414 100644
--- a/src/libtsduck/dtv/linux/tsTunerGuts.cpp
+++ b/src/libtsduck/dtv/linux/tsTunerGuts.cpp
@@ -781,7 +781,9 @@ bool ts::Tuner::Guts::getCurrentTuning(ModulationArgs& params, bool reset_unknow
             props.add(DTV_INVERSION);
             props.add(DTV_SYMBOL_RATE);
             props.add(DTV_INNER_FEC);
+#if defined(DTV_STREAM_ID)
             props.add(DTV_STREAM_ID);
+#endif

             if (::ioctl(frontend_fd, ioctl_request_t(FE_GET_PROPERTY), props.getIoctlParam()) < 0) {
                 const SysErrorCode err = LastSysErrorCode();
@@ -789,16 +791,20 @@ bool ts::Tuner::Guts::getCurrentTuning(ModulationArgs& params, bool reset_unknow
                 return false;
             }

+#if defined(DTV_STREAM_ID)
             uint32_t val = 0;
+#endif
             params.inversion = SpectralInversion(props.getByCommand(DTV_INVERSION));
             params.symbol_rate = props.getByCommand(DTV_SYMBOL_RATE);
             params.inner_fec = InnerFEC(props.getByCommand(DTV_INNER_FEC));
             params.stream_id.clear();
+#if defined(DTV_STREAM_ID)
             if ((val = props.getByCommand(DTV_STREAM_ID)) != DTVProperties::UNKNOWN) {
                 // Warning: stream id may be incorrect when returned from the driver.
                 // We should update it when possible with the actual transport stream id from the inner stream.
                 params.stream_id = val;
             }
+#endif
             break;
         }
         case DS_ISDB_T: {
@@ -1224,7 +1230,9 @@ bool ts::Tuner::tune(ModulationArgs& params, Report& report)
             props.addVar(DTV_SYMBOL_RATE, params.symbol_rate);
             props.addVar(DTV_INNER_FEC, params.inner_fec);
             props.addVar(DTV_INVERSION, params.inversion);
+#if defined(DTV_STREAM_ID)
             props.addVar(DTV_STREAM_ID, params.stream_id);
+#endif
             break;
         }
         case DS_ISDB_T: {
```

And I need to use this to compile it:
```
GCCVERSION=4.9-backport \
GCCROOT=/usr/lib/gcc-$GCCVERSION \
GCCBIN=$GCCROOT/bin \
GCCLIB=$GCCROOT/lib \
make BINDIR_SUFFIX=-gcc49-backport \
     CPP=$GCCBIN/cpp CC=$GCCBIN/gcc GCC=$GCCBIN/gcc CXX=$GCCBIN/g++ LD=$GCCBIN/gcc AR=$GCCBIN/gcc-ar \
     LDFLAGS_EXTRA="-L$GCCLIB -Wl,-rpath,$GCCLIB '-Wl,-rpath,\$\$ORIGIN'" \
     ARFLAGS=rc \
     NOSRT=1 NOTEST=1 NOPCSC=1 NODTAPI=1 \
     -j3
```
[... 46 lines cut ...]
        librtmp.so.0 => /usr/lib/x86_64-linux-gnu/librtmp.so.0 (0x00007fbfd403b000)
        libz.so.1 => /lib/x86_64-linux-gnu/libz.so.1 (0x00007fbfd3e24000)
        /lib64/ld-linux-x86-64.so.2 (0x00007fbfd7227000)
        libgcrypt.so.11 => /lib/x86_64-linux-gnu/libgcrypt.so.11 (0x00007fbfd3ba5000)
        libresolv.so.2 => /lib/x86_64-linux-gnu/libresolv.so.2 (0x00007fbfd398f000)
        libsasl2.so.2 => /usr/lib/x86_64-linux-gnu/libsasl2.so.2 (0x00007fbfd3774000)
        libgnutls.so.26 => /usr/lib/x86_64-linux-gnu/libgnutls.so.26 (0x00007fbfd34b4000)
        libkrb5.so.3 => /usr/lib/x86_64-linux-gnu/libkrb5.so.3 (0x00007fbfd31e0000)
        libk5crypto.so.3 => /usr/lib/x86_64-linux-gnu/libk5crypto.so.3 (0x00007fbfd2fb7000)
        libcom_err.so.2 => /lib/x86_64-linux-gnu/libcom_err.so.2 (0x00007fbfd2db3000)
        libkrb5support.so.0 => /usr/lib/x86_64-linux-gnu/libkrb5support.so.0 (0x00007fbfd2baa000)
        libkeyutils.so.1 => /lib/x86_64-linux-gnu/libkeyutils.so.1 (0x00007fbfd29a6000)
        libgpg-error.so.0 => /lib/x86_64-linux-gnu/libgpg-error.so.0 (0x00007fbfd27a3000)
        libtasn1.so.3 => /usr/lib/x86_64-linux-gnu/libtasn1.so.3 (0x00007fbfd2592000)
        libp11-kit.so.0 => /usr/lib/x86_64-linux-gnu/libp11-kit.so.0 (0x00007fbfd2380000)

LD_LIBRARY_PATH="/usr/lib/gcc-4.9-backport/lib" ldd tsp | grep 4.9
        libstdc++.so.6 => /usr/lib/gcc-4.9-backport/lib/libstdc++.so.6 (0x00007f2aded81000)
        libgcc_s.so.1 => /usr/lib/gcc-4.9-backport/lib/libgcc_s.so.1 (0x00007f2ade4c4000)
```
So the question is: Why the linking of the libraries `libstdc++` and `libgcc_s` is not correctly set by default?

Please, help me to solve this problem.
I hope the rest of the information will be useful.
Regards.

lelegard, 2020-12-20:
> For the last I need to use this patch (please, merge it):
> ...
> `+#if defined(DTV_STREAM_ID)`

Yes. Will do...

> The ARFLAGS=rc is required to overcome what is described in #379 & #378.

No. The makefile doesn't really know when this outdated option is needed. If you specify an old compiler, also specify the required options in the `make` command line.

> I suggest to put in one line the flags for modules NOSRT=1 NOTEST=1 NOPCSC=1 NODTAPI=1.

No. Just select the list of options you need. When there is a good technical reason to remove one, I don't see why we should remove them all.

> The GCCLIB is /lib and not /lib64 (I feel that "lib" is the standard in x64 and not "lib64").

It depends on the distro. The Debian family (Debian, Ubuntu, Raspbian) uses `lib` while the Red Hat family (Red Hat, CentOS, Fedora) uses `lib64`. For the regular build of TSDuck, this is computed in `Makefile.common`. The sample commands in a previous post in this thread were run on CentOS. They must be tuned according to the distro.

> Compiling with this line and the GCC 4.9-backport, the linking of the libraries requires to use the LD_LIBRARY_PATH.
> Why the linking of the libraries libstdc++ and libgcc_s is not correctly set by default?

This is solved by `-Wl,-rpath,$GCCLIB` in the `LDFLAGS_EXTRA` in the sample commands above. The two libraries are direct support for the generated code and depends on the compiler version. So, they come with the compiler, not with the system.

Note that in https://github.com/lelegard/gcc-build/blob/master/build-tsduck.sh I also add `'-Wl,-rpath,\$\$ORIGIN'` (beware of quotes and backslashes) in the `LDFLAGS_EXTRA` to avoid redefining `LD_LIBRARY_PATH` for `libtsduck.so` when TSDuck is not "installed" afterward and is only built to be run from some alternate directory (several builds of TSDuck with several compilers).

lars18th, 2020-12-20:
Hi @lelegard ,

Thank you for your comments. I reply inline...

> > For the last I need to use this patch (please, merge it):
> > ...
> > `+#if defined(DTV_STREAM_ID)`
> 
> Yes. Will do...

:+1: 

> > The ARFLAGS=rc is required to overcome what is described in #379 & #378.
> 
> No. The makefile doesn't really know when this outdated option is needed. If you specify an old compiler, also specify the required options in the `make` command line.

That's whats I'm doing. I include in my make command the `ARFLAGS=rc` flag. Without it, the compile fails as the `ar` tool receives the `-u` parameter that is not available in ancient environments. It's all documented and explained in these two threads.

> > I suggest to put in one line the flags for modules NOSRT=1 NOTEST=1 NOPCSC=1 NODTAPI=1.
> 
> No. Just select the list of options you need. When there is a good technical reason to remove one, I don't see why we should remove them all.

I'm just selecting what I need. And for this reason I use this. My suggestions is only to move the `NOSRT=1` that you use in the first line of the `make` to another to make more easy to understand and modify it.

> It depends on the distro. The Debian family (Debian, Ubuntu, Raspbian) uses `lib` while the Red Hat family (Red Hat, CentOS, Fedora) uses `lib64`. For the regular build of TSDuck, this is computed [...]

OK. So it's only a thing to document/comment.

> This is solved by `-Wl,-rpath,$GCCLIB` in the `LDFLAGS_EXTRA` in the sample commands above. The two libraries are direct support for the generated code and depends on the compiler version. So, they [...]

It's my fault! I really need to tweak the command. And you're right. However, I see the reason to use in your testing environment the `'-Wl,-rpath,\$\$ORIGIN'` (and yes, I'm aware about the single quotes), but I don't recomment it in a user environment. I prefer to use `-Wl,-rpath,.`. Then you can copy the `bin/release-*` directory to whatever you want and it continues to work. However, *only* if you change the directory to the one where is the binary. Any idea to use in this way calling from another path with the full path of the tsp tools?

Finally, that's my command to compile TSDuck with a backport compiler. I suggest to include it here (after the simple example `make NOPCSC=1 NOCURL=1 NODTAPI=1`): https://tsduck.io/doxy/building.html
```
export GCCVERSION=4.9-backport
export GCCROOT=/usr/lib/gcc-$GCCVERSION
export GCCBIN=$GCCROOT/bin
export GCCLIB=$GCCROOT/lib
make BINDIR_SUFFIX=-gcc49-backport \
     CPP=$GCCBIN/cpp CC=$GCCBIN/gcc GCC=$GCCBIN/gcc CXX=$GCCBIN/g++ LD=$GCCBIN/gcc AR=$GCCBIN/gcc-ar \
     LDFLAGS_EXTRA="-L$GCCLIB -Wl,-rpath,$GCCLIB -Wl,-rpath,." \
     ARFLAGS=rc \
     NOSRT=1 NOTEST=1 NOPCSC=1 NODTAPI=1 \
     -j3
```

Regards.

lars18th, 2020-12-20:
Hi @lelegard ,

Sorry, but finally you're right and the RPATH requires to be `<lib_path>:$ORIGIN`. So the command should be:
```
export GCCVERSION=4.9-backport
export GCCROOT=/usr/lib/gcc-$GCCVERSION
export GCCBIN=$GCCROOT/bin
export GCCLIB=$GCCROOT/lib
make BINDIR_SUFFIX=-gcc49-backport \
     CPP=$GCCBIN/cpp CC=$GCCBIN/gcc GCC=$GCCBIN/gcc CXX=$GCCBIN/g++ LD=$GCCBIN/gcc AR=$GCCBIN/gcc-ar \
     LDFLAGS_EXTRA="-L$GCCLIB -Wl,-rpath,$GCCLIB '-Wl,-rpath,\$\$ORIGIN'" \
     ARFLAGS=rc \
     NOSRT=1 NOTEST=1 NOPCSC=1 NODTAPI=1 \
     -j3
```

Please, consider to add this example (as a complex example) after the "simple" version `make NOPCSC=1 NOCURL=1 NODTAPI=1` in the documentation in this page: https://tsduck.io/doxy/building.html
It will be very useful for many users. And not forget to mention the address of https://github.com/lelegard/gcc-build too!

Regards.

lelegard, 2020-12-20:
> And not forget to mention the address of https://github.com/lelegard/gcc-build too!

This is just a personal repo, not part of the tsduck organization. All this discussion is just Linux culture and the TSDuck documentation does not pretend to be a Linux lesson.

lars18th, 2020-12-20:
> > And not forget to mention the address of https://github.com/lelegard/gcc-build too!
> 
> This is just a personal repo, not part of the tsduck organization. All this discussion is just Linux culture and the TSDuck documentation does not pretend to be a Linux lesson.

Yes, but... Why not help other users if you can? I feel this "mini-repo" is useful. So a simple line with "more complex compilation commands in this repo: https://github.com/lelegard/gcc-build" could be useful, don't you think? :wink: 

Regards.

lars18th, 2020-12-20:
And before closing this issue, some reference command for people that requires to check the `tsp` linking RPATH: `objdump -x tsp | grep RPATH` . This is useful as the regular `ldd tsp` and the common  `tsp --version=all`.

Thank you!

lars18th, 2021-01-14:
Hi @lelegard ,

I want to close this issue (I prefer to clean up instead of leave open a lot of issues), but I have one related suggestion. In this case are "simple" comments:

1. Can you add to the "tsp --help" this line?
```
  --version
      Display the TSDuck version number. Use '--version=all' for complete compilation information.
```
2. And can you include the make command line in the information?
```
tsp: TSDuck - The MPEG Transport Stream Toolkit - version 3.25-2127
Built Dec 19 2020 - 18:36:26
Using GCC 6.3.0, C++ std 2011.03
With command 'make NOSRT=1 NOTEST=1 NOPCSC=1 NODTAPI=1 -j3'
Web library: libcurl: 7.52.1, ssl: OpenSSL/1.0.2l, libz: 1.2.8
SRT library: This version of TSDuck was compiled without SRT support
Dektec: This version of TSDuck was compiled without Dektec support
```

This will make more easy to check multiple compilations.
Thank you!

lelegard, 2021-01-14:
The predefined common options `--help` and `--version` have many undocumented values. Like any other option with a list of predefined values, just use some invalid value to get the list of valid ones:
~~~
$ tsp --help=x
tsp: invalid value x for option --help, use one of "description", "full", "name", "syntax", "usage"
$ tsp --version=x
tsp: invalid value x for option --version, use one of "all", "compiler", "date", "dektec", "http", "integer", "long", "nsis", "short", "srt"
$
~~~

lars18th, 2021-01-14:
> The predefined common options `--help` and `--version` have many undocumented values. Like any other option with a list of predefined values, just use some invalid value to get the list of valid ones:

Then, I suggest to add some line in the help like this (for help _and_ version): ` bla,bla,bla   . Try --help/version=x for additional options`.

And regarding the add of the `make` line. It's possible to implement it?

lars18th, 2021-06-09:
Hi @lelegard ,

Still waiting for adding the line `Use --help=x or --version=x for a list of undocumented options`.

And also I want to comment that a `config.log` file including _ALL_ the `make` parameters are useful. Then we can (after a lot of time) recompile a new version of TSDuck using the same options.

Perhaps you feel that these two points are irrelevant. However, in some environments they have a lot of sense. So please, consider to implement them if it's easy.
Thank you!

lars18th, 2021-12-02:
Hi @lelegard ,

After checking the commit 556cee60ff1d53de1f73987a3289da5b52b6ca84 I want to remember this request:
- Print the line `Use --help=x or --version=x for a list of undocumented options` in the general help (`-h` or `--help`).
- Add a new `--version=config` entry that prints the configuration used for compiling.

Then will be more easy to work with multiple versions in multiple platforms.

Thank you!

lelegard, 2021-12-02:
The list of options for `--version=...` have been documented (RTFM 3.1.3).

The list of options for `--help=...` are useless to the user. They are technical formatting options which are used in internal scripts to generate [this Web page](https://tsduck.io/commands/) or to support bash auto-completion (new feature) for instance.

The "configuration used for compiling" is basically described in `--version=all`. This is not something that can be unambiguously defined by one string. TSDuck does not use the old (and outdated) "./configure" method. The way to customize the build is different from Windows to Linux/macOS (Visual Studio vs. make). Some options are not even driven by the `make` command line but by the potential unavailability of the corresponding libraries during the build (e.g. SRT or RIST).

So, the best summary of the configuration remains `--version=all`.

lars18th, 2021-12-02:
Hi @lelegard ,

Thank your for the response. From my point of view a simple line in the current help indicating these two command has sense. Nevertheless, I respect your opinion. Don't worry then.

Regarding the compiling info, I'm not suggesting a "one-line-output". Several huge projects, like for example FFmpeg could print the _full_ configuration when compiling. This could be 10-15 lines. And it indicates a very useful information about this specific binary. I know the use of the `--version=all` (pointed by you time ago, thank you!). But my suggestion is to know the compiler environment/parameters too, and environment vars as well. The idea is to know how to recompile an identical binary.

Please, think on this: you have a binary version of TSDuck, and now you want to compile an additional plugin. But you need to know the exact compiling toolchain and configuration. What we can do that withtout this useful information?

Sorry for asking for something that perhaps doesn't have sense for you.
Regards.

lelegard, 2021-12-02:
> Please, think on this: you have a binary version of TSDuck, and now you want to compile an additional plugin. But you need to know the exact compiling toolchain and configuration. What we can do that withtout this useful information?

All required information is in `--version=all`.

And if your choice is wrong, it won't work anyway. The two critical points are 1) version of the libtsduck (the binary C++ API can be incompatible between two versions) and 2) the bitrate representation. These two constraints are enforced through linker symbols. In case of incompatibility, no TSDuck command will start. I may be wrong but I do not see any inconsistency that would not generate an error while activating the command or the plugin.

lars18th, 2021-12-09:
OK, so then I close this issue.

closed 2021-12-09 by lars18th