tsduck/tsduck #698: [Q] /bin/bash: warning: setlocale: LC_ALL: cannot change locale (en_US.UTF-8)

lars18th, 2020-12-19:
Hi @lelegard ,

I see this "message" (repeated hundreds of times) when I compile TSDuck:
```
/bin/bash: warning: setlocale: LC_ALL: cannot change locale (en_US.UTF-8)
/bin/bash: warning: setlocale: LC_ALL: cannot change locale (en_US.UTF-8)
/bin/bash: warning: setlocale: LC_ALL: cannot change locale (en_US.UTF-8)
/bin/bash: warning: setlocale: LC_ALL: cannot change locale (en_US.UTF-8)
```

I know that is my fault, as in this environment...
```
declare -x LANG="en_GB.UTF-8"
declare -x LANGUAGE="en_GB:en"
```

And the locale "en_US.UTF-8" is not installed. However, only for your information, as this doesn't have any impact in the result of the compilation, perhaps you want to silence it.

Regards.

lars18th, 2020-12-19:
Hi @lelegard ,

I silenced this warning with this simple patch:
```
diff --git a/Makefile.common b/Makefile.common
index e5634c1..b770ebe 100644
--- a/Makefile.common
+++ b/Makefile.common
@@ -612,7 +612,8 @@ SHELL := /bin/bash --noprofile --norc

 # Enforce English locale by default.
 export LANGUAGE = en_US.UTF-8
-export LC_ALL = en_US.UTF-8
+#export LC_ALL = en_US.UTF-8
+export LC_ALL = C
 export LANG = en_US.UTF-8

 # Define functions to search commands (see comments in header)
```

Are you sure that's mandatory to use for LC_ALL one specific language and not the generic ASCII-7 output?

lelegard, 2020-12-19:
Hi @lars18th 

This definition was added quite some time ago and I do not recommend to remove it. At some point, I got some weird issue (I unfortunately cannot remember the details) and realised that the Linux system was configured with the French language by default. This messed up the output of some commands. So, this definition was added to make sure that the execution and output remain predictable.

The language `C` is an legacy definition and I remember having had issues when some command reported an error or warning because of the lack of UTF-8 support. Nowadays, all OS are UTF-8 based.

So, generally speaking, even if your default language is not `en_US`, it is safe to never _remove_ it from a Linux system and make that scripts which play with command output enforce it.

That being said, I clearly sympathise with the idea that the _English_ language is, by definition, the language of the _English_ people and not the vernacular dialect of some former British colony over the ocean. 😆

lars18th, 2020-12-19:
Hi @lelegard ,

I agree to not _remove_ this. However, in some environments, where the `en_US.UTF-8` is not installed (and can't be installed), then it will _desirable_ to compile anyway. And using the definition `export LC_ALL = C` actually works. So why almost not include a simple method to redefine it (if the users want to do it)? For example with `LC_ALL_EXTRA=C make ...`

Regards.