tsduck/tsduck #699: [FEAT] Limit the NULL input

lars18th, 2020-12-19:
#### Is your feature request related to a problem?
Hi,

I created a trouble in my network with this simple command:
- `tsp -I null -O ip 224.1.1.1:10000`

Obviously, I'm the responsible about what I'm doing, and the TSDuck is a tool for engineers. However, the simple error of not setting the bitrate can be very dangerous.

#### Proposed solution:
I suggest to enforce the explicit use of a bitrate when using the NULL input. Or almost use a default value that will be low. Or if you don't agree with this, request to confirm with a warning.

#### Alternatives:
The alternative is to limit the bitrate to some rasonable value. But I hate the tools that limit what the user (engineer) request to do because the developer think that this is dangerous. So, almost print a warning!

#### Additional information:
My network uses some switches without storm limit and some doesn't support multicast (only broadcast), so if you flow the network with all the packets that the machine can generate you'll be in troubles.

So, think on this risk, please!

lelegard, 2020-12-19:
I do not want to see a warning when I want to generate a file:
~~~
tsp -I null 10 -O file null10.ts
~~~
What you think is dangerous in _your_ environment can be normal in others.

lars18th, 2020-12-20:
Hi @lelegard ,

> I do not want to see a warning when I want to generate a file:
> What you think is dangerous in _your_ environment can be normal in others.

Yes. I aggree with you that you don't need to _enforce_ anything. However, _some_ cases can be potentially dangerous. For example, when writing to the network without a limit of the bitrate. I'm only commenting about the benefits of a simple check. Obviously all these warnings should be disabled with some command like `--no-dangerous-warnings`. But a check like: `if network output and no bitrate and no dangerous warnings then ask user to be sure`.

That's only a suggestion.
Regards.

lars18th, 2020-12-20:
Hi @lelegard ,

I prefer to not leave opened an infinite number of issues. So, as this is already commented, if you agree I close it.

Regards.

closed 2020-12-20 by lars18th