tsduck/tsduck #700: [FEAT] Pull mode VS Push mode

lars18th, 2020-12-20:
#### Is your feature request related to a problem?
The current working mode of the `tsp` tool can be considered PUSH mode. More or less it reads from the input as fast as it can, and then pushes the packets to the output. So, from the point of view of the outcomings the model is PUSH.

Another different working mode can be the PULL model. In this case, the output instead of pushing all the incoming data, it can _consume_ data from the input. So if no one is reading from the output, then no packets are readed in the input. This can be useful for example for a hypotetical HTTP server output plugin. If no clients are connected, then no packets are consumed.

#### Proposed solution:
My proposal is to support such type of PULL outputs. For example a plain `HTTP` output (described here as example https://github.com/tsduck/tsduck/issues/696#issuecomment-746174096), that is very different to the current HLS push output.

#### Alternatives:
I feel there is not alternative at time for a PULL model. Or I'm wrong?

#### Additional information:
That's only an idea. My concern is about how to control the tsp processing in a on-demand model. If a PULL mode will be supported then it will possible to apply this on-demand control.

What you think? This crazy idea has sense?

lelegard, 2020-12-20:
The structure of tsp is the following:

- Each plugin executes in a dedicated thread.
- Each plugin waits for packets from the previous plugin (or waits for packets from the input device, whatever it is).
- When packets are pulled from the previous plugin (resp. input), they are processed and passed to the next plugin (resp. output).

This is the definition of a PULL model. In a push model, the plugins would not necessarily have their dedicated threads. There could be a pull of worker threads or a dedicated single thread using a reactor pattern. The plugin would be called in some undefined thread context when packets would be pushed by the previous plugin, typically from the same thread or using a dynamic queue between worker threads.

> In this case, the output instead of pushing all the incoming data, it can consume data from the input. So if no one is reading from the output, then no packets are readed in the input. This can be [...]

Such a server output plugin would be implemented in the current tsp that way for instance (other implementations are possible): The send() method of the output plugin would wait for a client (listen() system call) and then would write packets to the client connection, once established. When the client disconnects, the send() method would block again on listen(). Thus, packets are indeed read from the input plugin and accumulate in the global buffer. When the buffer is full and no client connects to the output plugin, the whole chain is naturally suspended and the input plugin stops to read (its thread waits on free space in the buffer). When a client connects, packets start flowing. This exactly how the TCP flow control works, with socket buffers in client and server kernels instead of the tsp global buffer. Google for "TCP window size".

lars18th, 2020-12-21:
Hi @lelegard ,

Thank you for your good explanation. In my initial request perhaps I incorrectly used the words PULL and PUSH. However, the idea was quite simple: now the TSDuck model reads from the input and pushes data to the output. The suggestion is to support output modules that requests data from the chain. So in the first scenario the "control" (in the sense who marks the steps) is the input and in the second is the output. Nothing more.

And regarding about the options to implement such HTTP output plugin, I'm sure your suggestion is clearly the best.

Thank you!

lelegard, 2020-12-21:
>  So in the first scenario the "control" (in the sense who marks the steps) is the input and in the second is the output. 

It does not make sense. The flow control does the job in all cases. That is what I explained: if the output plugin blocks (as if there is no http client for instance), then the whole chain is blocked.

This is why I used the TCP example (unlike UDP). Thanks to the TCP window regulation, whether the source "pushes" data to an "always ready" destination or the destination "pulls" data from an "always ready" source, the result is the same.

lars18th, 2020-12-21:
> It does not make sense. The flow control does the job in all cases. That is what I explained: if the output plugin blocks (as if there is no http client for instance), then the whole chain is blocked.

OK. Now I understand it clearly. My mistake was thinking that the output plugin can't block the whole chain.

> This is why I used the TCP example (unlike UDP). Thanks to the TCP window regulation, whether the source "pushes" data to an "always ready" destination or the destination "pulls" data from an "always ready" source, the result is the same.

Perhaps at some point you will want to provide a simple example of a real TCP output plugin (but please, implement only what you think has sense).
Thank you!