tsduck/tsduck #701: [Q] HiDes Modulator bandwidth support

DavidSuignard, 2020-12-21:
Why does the HiDes modulator plugin limit the bandwidth to 5-8 MHz? The HiDes spec on their website allows for a bandwidth between 2-8 MHz.

Source: http://www.hides.com.tw/product_cg74469_eng.html

lelegard, 2020-12-21:
This is the result of a long legacy. I need to change this.

15 years ago, with Linux DVB API v3, when only DVB-S, DVB-C and DVB-T were supported, the bandwidth (DVB-T only) was an enum with a few values. With DVB API v5 and the generalization to other modulations, it became a value in Hz but TSDuck continued to implement bandwith as a set of discrete values.

I mark this as enhancement.

lars18th, 2020-12-21:
Hi @lelegard ,

Great to hear this. However, please remember to maintain (if you can) the compatibility with legacy kernels.

Regards.

lelegard, 2020-12-21:
> However, please remember to maintain (if you can) the compatibility with legacy kernels.

I dropped support for Linux DVB API v3 a long time ago. Supporting v3 and v5 (aka "S2-API") at the same time resulted in an awfully bloated and hardly maintainable source code.

lars18th, 2020-12-21:
> I dropped support for Linux DVB API v3 a long time ago. Supporting v3 and v5 (aka "S2-API") at the same time resulted in an awfully bloated and hardly maintainable source code.

Yes, I know it! The problem is with kernels with DVB API 5.0 and 5.1 support, instead of >=5.2. So, if the Hz steps is supported by all versions, then don't care about it.
Sorry for the noise introduced!
Regards.

lelegard, 2020-12-22:
> > Why does the HiDes modulator plugin limit the bandwidth to 5-8 MHz? The HiDes spec on their website allows for a bandwidth between 2-8 MHz.
> This is the result of a long legacy. I need to change this.

Done, you may rebuild from the repo or try the latest nightly builds.