tsduck/tsduck #702: [Q] Detecting Dropped/Added Component PID with --error-analyze

tahaarian, 2020-12-22:
Hello @lelegard 

I hope you are doing well. 

Before main subject, We are so appreciative of your generous project. 

We are working on some TS stream that will received on IP and we need to detect errors according to ETR 101 290 standard.  
Actually we need to detect which PMT's Component (audio,video and etc) would be dropped or added on real-time stream.

We could detect same Component PID that was dropped and added some seconds later (for example... according below image, we have dropped PID = 0x257 audio component and catch PAT/SDT actual error. after some seconds, we have added it again and these two errors was fixed)

[image]

But when we have dropped some components and added new components with different PIDs , tsduck can not detect any related errors. According to below images, we have dropped pid = 0x0046 and added pid = 0x0047 ... and could detect these modifications with your --service-analysis tag that is a kind of polling based request and we should compare them manually. Actually we need notification based solution like your --error-analysis tag and related error (ETR 101 290 standard)

[image]

[image]

We are working with tsp command according to below script. so we need to detect these kind of add/drop actions that will have different PIDs

`tsp -v -m --synchronous-log -I ip -r 239.0.0.1:2000 -P analyze -i8 --error-analysis -O drop `

We will appreciated if you could help us about the point.

Thank you in advance

Taha

lelegard, 2020-12-22:
Hi @tahaarian 

Thanks for your feedback.

The need for real-time analysis is not addressed by the `analyze` plugin. This plugin and its command counter-part `tsanalyze` are offline analyzers. It only reports a global analysis of a TS over a certain period. For the record, `tsanalyze` was the first TSDuck tool still alive. It existed before `tsp` and I developed it to understand the structure of TS files.

When used with the option `--interval`, the `analyze` plugin is just some kind of offline analyzer over repetitive periods of time, giving a snapshot of the TS during each interval.

Real-time events such as new or deleted PID's and others such as table renewal or crypto-periods can be detected with the `history` plugin. It is less synthetic than `analyze` but can report individual events. The idea was to precisely analyze sequences such as changes in service structure or clear/scrambled transitions. Some events such as deleted PID's are reported out-of-sequence because there is no such event as 'this PID is deleted from now on'. You just realize later that you don't see any more packets from a PID. The user's guide contains basic examples to sort the report in a strict chronological way.

tahaarian, 2020-12-23:
Hi @lelegard 

Thank you for your quick response. 

i have used `history `tag as you said ... it was so great and we could catch changes on PIDs 

Thank you in advance

Taha

tahaarian, 2020-12-23:
Hi again @lelegard 

sorry , i had another question.

could you explain me what kind of events which tsduck will catch by `history `tag ?
actually we have some other points that we need to detect 

maybe you could share event type list that its support them 

Thank you in advance

Taha

lelegard, 2020-12-30:
> maybe you could share event type list that its support them

Just added in the user's guide, in the documentation of the plugin `history`.

Version 3.25-2145, in the [repo](https://github.com/tsduck/tsduck/raw/master/doc/tsduck.pdf) right now and will be automatically updated on the [web site](https://tsduck.io/download/docs/tsduck.pdf) tonight.