tsduck/tsduck #703: [Q] using tsswitch and remote control, not showing the selected input

vandanachadha, 2020-12-22:
I am using tsswitch to switch between various inputs and expecting that as soon as the first input completes, the next input will start. But thats not happening. Then I added the remote control and sending the input id, is not working either.

tsswitch command is as follows
```
tsswitch -v --remote 4444 --delayed-switch --cycle 2  \
-I file /home/lenovo/vandana/tvinabox/videomanagement/server/public/converted/clipcanvas_14348_offline.ts \
-I file /home/lenovo/vandana/tvinabox/videomanagement/server/public/converted/big_buck_bunny_720p_1mb.ts  \
-I fork `tsp -I ip -l  192.168.29.217 :1234 -P until --seconds 20`  \
 -O  ip 192.168.29.217:12345 
```

Any suggestions? 

If i take the output to a file and have switched to another input using echo 1 >/dev/udp/127.0.0.1/4444, and play the output file using mplayer, it gives me a TS_PARSE error after showing the 1st video for some time.

do the inputs to the tsswitch command, need to be similar in some properties?

lelegard, 2020-12-22:
>  expecting that as soon as the first input completes, the next input will start. But thats not happening.

Are you sure? How do you characterise that this is not happening?

Trying something similar works as expected:
~~~
$ tsp -I craft --pid 100 --count 1 -O file in1.ts
$ tsp -I craft --pid 200 --count 2 -O file in2.ts
$ tsswitch -v --remote 4444 --delayed-switch --cycle 2  \
    -I file in1.ts \
    -I file in2.ts \
    -I fork 'tsp -I craft --pid 300 --count 3' \
    | tsdump | grep PID:
  PID: 100 (0x0064), header size: 4, sync: 0x47
  PID: 200 (0x00C8), header size: 4, sync: 0x47
  PID: 200 (0x00C8), header size: 4, sync: 0x47
  PID: 300 (0x012C), header size: 4, sync: 0x47
  PID: 300 (0x012C), header size: 4, sync: 0x47
  PID: 300 (0x012C), header size: 4, sync: 0x47
  PID: 100 (0x0064), header size: 4, sync: 0x47
  PID: 200 (0x00C8), header size: 4, sync: 0x47
  PID: 200 (0x00C8), header size: 4, sync: 0x47
  PID: 300 (0x012C), header size: 4, sync: 0x47
  PID: 300 (0x012C), header size: 4, sync: 0x47
  PID: 300 (0x012C), header size: 4, sync: 0x47
~~~
We can see two cycles of the three inputs in sequence.

> play the output file using mplayer, it gives me a TS_PARSE error after showing the 1st video for some time.
> do the inputs to the tsswitch command, need to be similar in some properties?

`tsswitch` does not make any assumption from input. It just routes TS packets. The problem is the consistency of the output. At the switch point, the content and structure of the output TS suddenly changes without prior notice. When a player reads this TS, it may be confused. Let's take one example: the two files have the same structure, same TS id and service id, same component PID's but different video encoding (AVC and HEVC for instance). At the switch point, the player is not notified of the change (why should it be?) and the AVC codec suddenly receives HEVC data. We may find many similar examples leading to inconsistencies.

When you switch from one input to another, you need to make sure that the change is either properly notified (distinct TS id, service ids, PAT and PMT version) or fully transparent (exact same TS structure, PAT, PMT, PID's and codecs).

vandanachadha, 2020-12-23:
Thanks @lelegard for answering and clarifying doubts. 
The issue was with the disparate ts structured files. Once thats was fixed, tsswitch worked correctly for me, and kept showing the next input. Thanks

lelegard, 2020-12-29:
> These did not work, but is this even possible? 2 ts files and 1 stream as input to tsswitch

All combinations of inputs are technically possible, this one included. You problem is likely not linked to the input switching but rather to the interpretation of an output stream which is in fact a sequence of distinct streams with similar although not identical structure.

vandanachadha, 2021-01-05:
Thanks for your reply @lelegard,  and wish you a very happy 2021.