tsduck/tsduck #705: Use tsduck in Java with JNI or JNA [Q]

tahaarian, 2020-12-27:
Hello @lelegard 

I hope you are doing well.

Before main subject, We are so appreciative of your generous project and your helpful advices.

Actually we could handle `Drop/Add PIDs` according to your advice. thank you in advance

Now, we need to use `tsduck` as a `DLL` or library in java programming and develop an app (java based) for detecting those `PID` changes. because we need to use this java module in our project.

So when we tried to create `JNI` (java api for using `history` tag) we had some problems and we could not understand your structure. 

As we have checked, you developed some java `JNI` or sample code according to [this issue](https://github.com/tsduck/tsduck/issues/556). 

May i ask if you could help us and share with me some java sample code ? or we will appreciated if you could give me some points for starting to our development 

Thank you in advance

Taha

lelegard, 2020-12-29:
> As we have checked, you developed some java JNI or sample code according to this issue.
>
> May i ask if you could help us and share with me some java sample code ? or we will appreciated if you could give me some points for starting to our development

Indeed, I developed a substantial amount of JNI code about 12 or 13 years ago in a professional context which was not linked to TSDuck. So, I cannot help you directly in your development.

All I can say is that JNI has a much better defined architecture than Python native interface. It is quite rigorous and correctly documented. It is easy to interact with Java objects from C++ (but not the opposite). From my experience, using JNI correctly requires a good knowledge of Java and C++ internals. Being a former compiler developer, I found it quite straightforward. But this is clearly not a task for beginners.

Take great care of the description of the memory usage in the documentation so that you can correctly track and free memory objects when necessary. This is the tricky point. Java programmers don't care about memory allocation. C++ programmers may use tricks to avoid caring about memory allocation (see the `ts::SafePtr` class in TSDuck). But when you interface the two worlds, you have to take care about everything yourself.

tahaarian, 2021-01-04:
Hello again @lelegard 

Thank you for your helpful comments. 

Actually i have developed a java code that has used `tsduck.dll` according to below part 

```java
        String dir = new File(ResourceUtils.getResourcePath("win/tsduck.dll")).getAbsolutePath();
            return dir;
```

Then prepare some `api `like your` python file (lib.py)` same as below :

```java
package tv.samim.iptv.cmts.modules.tsp;

import com.sun.jna.*;
import static java.util.Collections.singletonMap;
interface TspLibrary extends Library {

	TspLibrary INSTANCE = (TspLibrary) Native.loadLibrary(Tsp.getMainLib(), TspLibrary.class, singletonMap(OPTION_FUNCTION_MAPPER, (FunctionMapper) (lib, method) -> {
		return method.getName();
	}));

	int tspyVersionInteger();
    long tspyNewAsyncReport(int severity, boolean sync_log, boolean timed_log, int log_msg_count);
    long tspyNewTSProcessor(long report_pointer);
    boolean tspyStartTSProcessor(long tsp_address,long args,Pointer plugins,int size_of_plugins);

}
```

i could get version with `tspyVersionInteger();` and create` Report & NewTSProcessor settings `with related functions, but i have some problem with `tspyStartTSProcessor`.

when i have checked your python code, you have prepared a` UTF-16 buffer with application names and plugins` that was appended with 0xffff.... 

actually i cant understand this part. I think that we did some wrong style. i have developed `getPluginBytes()` function for these parts as below :

```java
    public byte[] getPluginBytes() throws UnsupportedEncodingException {
        StringBuffer buffer = new StringBuffer();

        buffer.append(tspPlugin.app_name);

        buffer.append(0xFF);
        buffer.append(0xFF);
        buffer.append("-I");
        for(int i= 0;i< tspPlugin.input.length;i++){
            buffer.append(0xFF);
            buffer.append(0xFF);
            buffer.append(tspPlugin.input[i]);
        }

        buffer.append(0xFF);
        buffer.append(0xFF);
        buffer.append("-P");
        for(int j= 0;j< tspPlugin.plugin.length;j++) {
            for (int z = 0; z < tspPlugin.plugin[j].length; z++) {
                buffer.append(0xFF);
                buffer.append(0xFF);
                buffer.append(tspPlugin.plugin[j][z]);
            }
        }
        
        buffer.append(0xFF);
        buffer.append(0xFF);
        buffer.append("-O");
        for(int u= 0;u< tspPlugin.output.length;u++) {
            buffer.append(0xFF);
            buffer.append(0xFF);
            buffer.append(tspPlugin.output[u]);
        }
        return buffer.toString().getBytes("UTF-16LE");
    }
```

Its a simple fucntion that will handle your `Start `function in` tsp.py `file, but we have gotten below error :

[image]

As you could see, it has some error in parsing `input plugin`. i think that we could pass parameters correctly but maybe some wrong sequence of bytes for  `input plugin` part. The function prepare byte array for calling `tspyStartTSProcessor `api according below part :

```java
        tspPlugin = new TspPlugins();
        tspPlugin.app_name = "HistoryTsp";
        tspPlugin.input = new String[]{"ip", "-r", "239.0.0.1:1234"};
        tspPlugin.plugin = new String[][]{{"history", "-c", "--suspend-packet-threshold","60","--ignore-stream-id-change"}};
        tspPlugin.output = new String[]{"drop"};

        byte[] plg = getPluginBytes();
        Pointer ptr = new Memory(plg.length);

        ptr.write(0, plg, 0, plg.length);
        boolean result = TspLibrary.INSTANCE.tspyStartTSProcessor(tsp_address, tsp_address,ptr,plg.length);
```

we will be appreciated if you could help us. just please describe me your sequence of bytes or wrong point. other functions work great. 

I will share with you final java module that will handle tsp command same as your python binder.

Thank you in advance

lelegard, 2021-01-10:
See the new Java bindings, in [a comment](https://github.com/tsduck/tsduck/issues/707#issuecomment-757455698) of issue #707.

closed 2022-09-23 by lelegard