tsduck/tsduck #706: [Q] Bit rate monitoring Alarms

Jazzykool85, 2020-12-29:
Hello,
I am using the excellent way provided by Tsduck to monitor bit rate for some specific PIDs on my side
I am stuck with problem, Kindly help me sort it out.
Here am using the tsp bit rate monitor plug on a specific PID
Currently am trying to trigger and Alarm when the bit rate goes below referance 
tsp -I dektec -d 0 -P bitrate_monitor -a "<<some python script>>" --min 10 -p 10  --pid xxx -O drop

I want to check if i can trigger another alarm when the bit rate recovers ?

Jazzykool85, 2020-12-29:
tsp -I dektec -d 0 -P bitrate_monitor -a "some python script" --min 10 -p 10 --pid xxx -O drop

lelegard, 2020-12-29:
> I want to check if i can trigger another alarm when the bit rate recovers ?

The same alarm command is triggered when the bitrate is back to normal:
~~~
$ tsp -P bitrate_monitor --help
...
  -a 'command'
  --alarm-command 'command'
      Command to run when the bitrate goes either out of range or back to
      normal. The command receives an additional string parameter containing an
      informational message.
~~~

Jazzykool85, 2020-12-29:
Thanks 
Actually i want to trigger 2 different python scripts 
One when it is above threshold and another after the bit rate recovers
Not sure how we can do that

lelegard, 2020-12-29:
As indicated above, the alarm command (your script) receives a message as parameter:

~~~
$ tsp -I ip 1234 -P bitrate_monitor --min 1,000,000 --max 2,000,000 --alarm 'echo "In alarm command:"' -O drop
* Warning: bitrate_monitor: TS bitrate (919,244 bits/s) is lower than allowed minimum (1,000,000 bits/s)
In alarm command: TS bitrate (919,244 bits/s) is lower than allowed minimum (1,000,000 bits/s)
* Warning: bitrate_monitor: TS bitrate (1,097,318 bits/s) is back in allowed range (1,000,000-2,000,000 bits/s)
In alarm command: TS bitrate (1,097,318 bits/s) is back in allowed range (1,000,000-2,000,000 bits/s)
* Warning: bitrate_monitor: TS bitrate (2,000,320 bits/s) is greater than allowed maximum (2,000,000 bits/s)
In alarm command: TS bitrate (2,000,320 bits/s) is greater than allowed maximum (2,000,000 bits/s)
* Warning: bitrate_monitor: TS bitrate (1,565,363 bits/s) is back in allowed range (1,000,000-2,000,000 bits/s)
In alarm command: TS bitrate (1,565,363 bits/s) is back in allowed range (1,000,000-2,000,000 bits/s)
~~~
So, you should use one script that looks for "is lower than allowed", "is greater than allowed" or "is back in allowed" in its first parameter and takes appropriate action.

Admittedly not really pretty but it works.

Jazzykool85, 2020-12-30:
Excellent .will try this
Nothing more than it if it works .Thanks a lot Lelegard for your time

Jazzykool85, 2020-12-30:
Excellent , the bit rate monitoring works perfectly fine for me now with one single script 
But i have a question,
Currently how do i manage if i need to monitor more than one PID at a time ?

lelegard, 2020-12-30:
> Currently how do i manage if i need to monitor more than one PID at a time ?

Use one plugin per PID:
~~~
tsp ... \
    -P bitrate_monitor --pid 101 --min .. --max ... --alarm ... \
    -P bitrate_monitor --pid 102 --min .. --max ... --alarm ... \
    -P bitrate_monitor --pid 103 --min .. --max ... --alarm ... \
    ...
~~~
Since each PID has probably a distinct range of valid bitrates, it does not really make sense to have one single plugin invocation for multiple PID's.

Note that I just pushed a modification of the plugin to pass 5 additional parameters (after the message) which are easier to handle than a plain text message. See the section 5.2.30 as an example in the TSDuck user's guide (pull the latest repo first).

Jazzykool85, 2020-12-30:
Ok..i will try .Thanks Lelegard

Jazzykool85, 2021-01-12:
Sorry i was on a holiday , Thats why response i delayed.

Tried now , Indvidually its working well , But as a single command like below it is not :(
Can you validate the below.,Thanks a Lot 
 tsp -I dektec -d 0 \
     -P bitrate_monitor  -t 20 --pid xx --min 1000 --alarm "myscript"\
     -P bitrate_monitor  -t 20 --pid yy--min 1000 --alarm "myscript"\
     -P bitrate_monitor  -t 20 --pid yy--min 1000 --alarm "myscript"\
     -O drop

Jazzykool85, 2021-01-12:
w

Jazzykool85, 2021-01-12:
Works well ;)

closed 2021-01-12 by Jazzykool85