tsduck/tsduck #707: Developing Java Binder for TSDUCK[FEAT]

tahaarian, 2021-01-04:
Hello again @lelegard 

Thank you for your helpful comments. 

Actually i have developed a java code that has used `tsduck.dll` according to below part 

```java
        String dir = new File(ResourceUtils.getResourcePath("win/tsduck.dll")).getAbsolutePath();
            return dir;
```

Then prepare some `api `like your` python file (lib.py)` same as below :

```java
package tv.samim.iptv.cmts.modules.tsp;

import com.sun.jna.*;
import static java.util.Collections.singletonMap;
interface TspLibrary extends Library {

	TspLibrary INSTANCE = (TspLibrary) Native.loadLibrary(Tsp.getMainLib(), TspLibrary.class, singletonMap(OPTION_FUNCTION_MAPPER, (FunctionMapper) (lib, method) -> {
		return method.getName();
	}));

	int tspyVersionInteger();
    long tspyNewAsyncReport(int severity, boolean sync_log, boolean timed_log, int log_msg_count);
    long tspyNewTSProcessor(long report_pointer);
    boolean tspyStartTSProcessor(long tsp_address,long args,Pointer plugins,int size_of_plugins);

}
```

i could get version with `tspyVersionInteger();` and create` Report & NewTSProcessor settings `with related functions, but i have some problem with `tspyStartTSProcessor`.

when i have checked your python code, you have prepared a` UTF-16 buffer with application names and plugins` that was appended with 0xffff.... 

actually i cant understand this part. I think that we did some wrong style. i have developed `getPluginBytes()` function for these parts as below :

```java
    public byte[] getPluginBytes() throws UnsupportedEncodingException {
        StringBuffer buffer = new StringBuffer();

        buffer.append(tspPlugin.app_name);

        buffer.append(0xFF);
        buffer.append(0xFF);
        buffer.append("-I");
        for(int i= 0;i< tspPlugin.input.length;i++){
            buffer.append(0xFF);
            buffer.append(0xFF);
            buffer.append(tspPlugin.input[i]);
        }

        buffer.append(0xFF);
        buffer.append(0xFF);
        buffer.append("-P");
        for(int j= 0;j< tspPlugin.plugin.length;j++) {
            for (int z = 0; z < tspPlugin.plugin[j].length; z++) {
                buffer.append(0xFF);
                buffer.append(0xFF);
                buffer.append(tspPlugin.plugin[j][z]);
            }
        }
        
        buffer.append(0xFF);
        buffer.append(0xFF);
        buffer.append("-O");
        for(int u= 0;u< tspPlugin.output.length;u++) {
            buffer.append(0xFF);
            buffer.append(0xFF);
            buffer.append(tspPlugin.output[u]);
        }
        return buffer.toString().getBytes("UTF-16LE");
    }
```

Its a simple fucntion that will handle your `Start `function in` tsp.py `file, but we have gotten below error :

[image]

As you could see, it has some error in parsing `input plugin`. i think that we could pass parameters correctly but maybe some wrong sequence of bytes for  `input plugin` part. The function prepare byte array for calling `tspyStartTSProcessor `api according below part :

```java
        tspPlugin = new TspPlugins();
        tspPlugin.app_name = "HistoryTsp";
        tspPlugin.input = new String[]{"ip", "-r", "239.0.0.1:1234"};
        tspPlugin.plugin = new String[][]{{"history", "-c", "--suspend-packet-threshold","60","--ignore-stream-id-change"}};
        tspPlugin.output = new String[]{"drop"};

        byte[] plg = getPluginBytes();
        Pointer ptr = new Memory(plg.length);

        ptr.write(0, plg, 0, plg.length);
        boolean result = TspLibrary.INSTANCE.tspyStartTSProcessor(tsp_address, tsp_address,ptr,plg.length);
```

we will be appreciated if you could help us. just please describe me your sequence of bytes or wrong point. other functions work great. 

I will share with you final java module that will handle tsp command same as your python binder. i hope it would be helpful

Thank you in advance for your great project

Taha

lelegard, 2021-01-07:
HI,

The functions tspy... have been added to interact with Python. The TSDuck UString class is physically identical to Java strings. So, a dedicated approach should be used for Java. Moreover, while Python needs some hack to interact with C++, JNI has a cleaner architecture, without dirty hack. I am working on a JNI interface to TSDuck (high level features only, same as Python). Be patient, I cannot work full time on TSDuck.

tahaarian, 2021-01-09:
Hello again @lelegard 

Thank you in advance for your respond. 
We could fix it. actually we should add a `16-UTF char (0xFFFF)` according bellow code 

```
    public byte[] getPluginBytes() throws UnsupportedEncodingException {
        StringBuffer buffer = new StringBuffer();

        buffer.append(tspPlugin.app_name);

        // Append with the content of a string or list of strings.
        // Strings are separated with an FFFF sequence.

        // INPUT
        buffer.append("\uFFFF");
        buffer.append("-I");
        for(int i= 0;i< tspPlugin.input.length;i++){
            buffer.append("\uFFFF");
            buffer.append(tspPlugin.input[i]);
        }

        // PROCESSOR
        buffer.append("\uFFFF");
        buffer.append("-P");
        for(int j= 0;j< tspPlugin.plugin.length;j++) {
            for (int z = 0; z < tspPlugin.plugin[j].length; z++) {
                buffer.append("\uFFFF");
                buffer.append(tspPlugin.plugin[j][z]);
            }
        }

        // OUTPUT
        buffer.append("\uFFFF");
        buffer.append("-O");
        for(int u= 0;u< tspPlugin.output.length;u++) {
            buffer.append("\uFFFF");
            buffer.append(tspPlugin.output[u]);
        }
        // Build UTF-16 buffer with application names and plugins.
        return buffer.toString().getBytes("UTF-16LE");
    }
```

Now we could catch your main functions in java JNI project correctly.

we have prepared below JNI API for communication with your tsduck.dll 

```
interface TspLibrary extends Library {

	TspLibrary INSTANCE = (TspLibrary) Native.loadLibrary(Tsp.getMainLib(), TspLibrary.class, singletonMap(OPTION_FUNCTION_MAPPER, (FunctionMapper) (lib, method) -> {
		return method.getName();
	}));

	int tspyVersionInteger();

    // void* tspyNewAsyncReport(int severity, bool sync_log, bool timed_log, size_t log_msg_count);
    long tspyNewAsyncReport(int severity, boolean sync_log, boolean timed_log, int log_msg_count);

    // void* tspyNewTSProcessor(void* report);
    long tspyNewTSProcessor(long report_pointer);

    // bool tspyStartTSProcessor(void* tsp, const tspyTSProcessorArgs* args, const uint8_t* plugins, size_t plugins_size);
    boolean tspyStartTSProcessor(long tsp_address,long args, byte[] plugins,int size_of_plugins);

    // void tspyWaitTSProcessor(void* tsp);
    void tspyWaitTSProcessor(long tsp_address);

    // void tspyTerminateAsyncReport(void* report);
    void tspyTerminateAsyncReport(long report_pointer);

}
```
I hope that we could developed as you mentioned.
Let me know if you like to have your main classes

Thank you so much for your great project

Taha

lelegard, 2021-01-10:
Java bindings are now available, starting with version 3.25-2163. See [an overview here](https://tsduck.io/doxy/javabindings.html). They are functionally identical to the Python bindings but use a clean JNI interface, not the dirty hack of Python bindings. To use them, either rebuild from the repo or fetch the latest [nightly builds](https://tsduck.io/download/prerelease/).

tahaarian, 2021-01-11:
such a great work 
we are so thanksful

closed 2022-09-23 by lelegard