tsduck/tsduck #708: [Q] Can I use a Technisat skystar S2 board? TSdvb does not see it in wes7.

vk-hca, 2021-01-06:
I am running win7 embedded and trying to extract Himawarri cast HRIT data stream. on Jsat 2b at 154E.
The downlink carries multicast UDP packets which are wrapped in DVB-MPE and need to be unwrapped. As I understand it not all demod chips will give this base level data.

The card is a Technisat skystar S2 board it appears to work with the Technisat software for it and the current DVBdeam version on normal tv dvbs2 signals from another sat.

DVBdream  reports  the deice as B2C2 - direct on driver v Device #1
The board uses an flexcop2b. 

I am trying to execute this line: but it returns no device.

tsp -I dvb --adapter 0 --delivery-system "DVB-S2" --lnb "5150000000" --frequency 4148000000 --modulation QPSK --symbol-rate 2586148 --fec-inner "3/5" --roll-off 0.2 --polarity "horizontal" -P mpe --pid 0x03E9 --udp-forward --log -O drop

tslsdvb -e outputs as follows:

=== Device category CAPTURE

device "VIA HD Audio Input"
  Some supported interfaces:
    interface IAMDeviceRemoval
    interface IAMFilterMiscFlags
    interface IAMovieSetup
    interface IBaseFilter
    interface IKsControl
    interface IKsPropertySet
    interface IKsTopologyInfo
    interface IMediaFilter
    interface IMediaSeeking
  IKsPropertySet properties support:
  IKsControl properties support:
  IKsTopologyInfo:
    Categories: AM_KSCATEGORY_AUDIO AM_KSCATEGORY_CAPTURE {EB115FFC-10C8-4964-831D-6DCB02E6F23F}
    Node 0, type {4D837FE0-C555-11D0-8A2B-00A0C9255AC1}

  - Pin "Recording Control", direction: output
    Some supported interfaces:
      interface IAMBufferNegotiation
      interface IAMStreamConfig
      interface IAMStreamControl
      interface IKsControl
      interface IKsPin
      interface IKsPropertySet
      interface IMediaSeeking
      interface IPin
      interface IQualityControl
    IKsPropertySet properties support:
    IKsControl properties support:

  - Pin "Input1", direction: input
    Some supported interfaces:
      interface IKsControl
      interface IKsPin
      interface IKsPropertySet
      interface IMemInputPin
      interface IPin
      interface IQualityControl
      interface IStreamBuilder
    IKsPropertySet properties support:
    IKsControl properties support:

=== Device category SPLITTER

=== Device category TVTUNER

=== Device category BDA_NETWORK_PROVIDER

=== Device category BDA_TRANSPORT_INFORMATION

=== Device category BDA_RECEIVER_COMPONENT

=== Device category BDA_NETWORK_TUNER

=== Tuning spaces

tslsdvb: CoCreateInstance : class not registered

It seems these cards are a better proposition in linux, Unfortunately I don't have one to hand with a pci slot, so I am stuck with the windows box and I am in no way familiar with DVB or or much else inside MS. So I assume its running through some kind of windows construct called BDA. Windows device manager shows the board as a network card.

The question is there a way to get to this stream from this card, and is it possible in a windows box?
Alternatively I can change the OS in the machine to Debian if it will work there.

lelegard, 2021-01-06:
> The downlink carries multicast UDP packets which are wrapped in DVB-MPE and need to be unwrapped. As I understand it not all demod chips will give this base level data.

A demod gives you a transport stream. The UDP stream is MPE-encapsulated into one PID of this transport stream. If you can receive this TS using TSDuck, then its `mpe` plugin will let you extract the UDP stream from the TS.

> The card is a Technisat skystar S2 board it appears to work with the Technisat software for it and the current DVBdeam version on normal tv dvbs2 signals from another sat.
>
> It seems these cards are a better proposition in linux, Unfortunately I don't have one to hand with a pci slot, so I am stuck with the windows box and I am in no way familiar with DVB or or much else inside MS. So I assume its running through some kind of windows construct called BDA. Windows device manager shows the board as a network card.

You got the point. Each OS, Windows or Linux, provides a more (Linux) or less (Windows) common API to access tuners, provided that some appropriate driver software for this tuner is available and presents the right API. On Windows, there are two levels of such software: BDA drivers and DirectShow filters. There is no common userland API to BDA drivers. Each vendor implements its own undocumented API. On top of the BDA driver, the tuner vendor provides a DirectShow filter. At this level, the user API is more or less standard and tuner-agnostic applications such as TSDuck can interact with the driver using DirectShow.

Most tuner vendors provide both a BDA driver and a DirectShow filter and their tuners can be used by TSDuck. The [TSDuck user's guide](https://tsduck.io/download/docs/tsduck.pdf) provides a list of tested tuners in section 7.1.4, based on the feedback of various users. And you can see that, unfortunately, TechniSat is the only tuner vendor which does not provide a DirectShow filter and which, consequently, cannot be used with tuner-agnostic applications such as TSDuck on Windows.

The tested tuner which is mentioned in the user's guide is a USB one, not a PCIe one. But usually each tuner vendor has a consistent software architecture across all its devices. I tested this device myself, from the large tuner collection of one of may colleagues. So, I can confirm there is no way to use it on Windows. I assume that the TechniSat proprietary software to watch TV uses the proprietary interface of its tuner driver. But this interface is specific to TechniSat and not documented.

> The question is there a way to get to this stream from this card, and is it possible in a windows box?
> Alternatively I can change the OS in the machine to Debian if it will work there.

As you may realize, reading some discussions in this forum, including recent ones, using tuners on Windows is a pain. It works (except with TechniSat devices) but the DirectShow framework is no longer actively maintained by Microsoft. Recent modulations such as ISDB are poorly supported. The performances are no very good. DirectShow is overly and uselessly complicated. Etc, etc.

As a rule of thumb, if you need a stable and reliable reception, I always recommend Linux. The LinuxTV API is simple, fast and reliable. It is maintained by the community and new modulations are smoothly integrated. Then you need to check the quality of the driver for each tuner. I cannot tell for the TechniSat one.

Currently, the only DVB-S2 tuner I own is a DVBsky S960 which is pretty good and reliable. But this is a USB one. I also have a PCTV DVB-S2 461e (USB as well) but it does not work well and I do not recommend it. In the past, I used very reliable Hauppauge PCI tuners but I no longer have them.

Concerning the distro, well, Linux is Linux and you choose the one you like. However, to get stock drivers for recent devices, you need a recent kernel. So, in the Debian family, I would recommend Ubuntu over Debian. The Debian kernels are old and do not integrate recent drivers. If you are familiar with the Linux kernel, you can always rebuild recent drivers from sources on older kernels. But if you do not want to struggle with driver compilation, better use Ubuntu (I would similarly recommend Fedora over RedHat/CentOS for the same reasons).

vk-hca, 2021-01-06:
OK Lelegrad, thank very much for the response, you have confirmed what I have figured from the last couple of days messing about with it.

I have seen mentions of a Linux Driver for it working this card with Centos for processing GEOS cast streams in South America and it being in more recent Kernels. So I will try Debian first as I have the installer ready on a stick.  I keep it near the MS box, threat being "next time you foul yourself your of here forever" Being embedded saves its skin till now. I used Ubuntu a lot about 15 years ago but  felt later revisions lost their way and bloated where as Debian just works with everything I have thrown at it so I will try it first. 

SO I will  change OS and report back the results.   If this card cant provide the stream, I would go to a USB TBS5520SE which would avoid the PCI slot requirement.  I will leave the issue open till I have a result on the card with TSDuck either way in the next few days.

vk-hca, 2021-01-08:
Now I have written over windows with linux which always feels good.  On Debian buster, all installs ok without issue, and it has /dev/dvb/adapter0.

The machine is an Intel not AMD and Debian was not happy about the AMD requirement in the deb but TSDuck built fine from source without any warnings or error with the only extra requirement was curl.

hca@wx:~$ tslsdvb
0: /dev/dvb/adapter0 ("Conexant CX24120/CX24118", DVB-S, DVB-S2) 

and further 
hca@wx:~$ tslsdvb --verbose

0: /dev/dvb/adapter0 ("Conexant CX24120/CX24118", DVB-S, DVB-S2)
   Device: /dev/dvb/adapter0/frontend0

  Delivery system ............................ DVB-S
  Frequencies:
    Min ................................ 950,000,000 Hz
    Max .............................. 2,150,000,000 Hz
    Step ................................. 1,011,000 Hz
    Tolerance ............................ 5,000,000 Hz
  Symbol rates:
    Min .................................. 1,000,000 sym/s
    Max ................................. 45,000,000 sym/s
    Tolerance .................................... 0 sym/s
  Spectral inversion ........................... off
  FEC(inner) .................................. auto
  Modulation .................................. QPSK

  Capabilities: inversion auto, FEC 1/2, FEC 2/3, FEC 3/4, FEC 4/5, FEC 5/6,
    FEC 6/7, FEC 7/8, FEC auto, QPSK, 2nd generation, recover

However the command I need to run is failing as a wrong input option. I think it was intended to be used in a win box but the linux format  tsp -I dvb --/dev/dvb/adapter0......  device gives the same result.

I tried -d for debug output in the command but I cant get any change in the output response? just tsp -d shows as below

hca@wx:~$ tsp -d
ID   State      Output               E-Level  Times(r/u/s)   Command [run=0/1]
0    finished   /tmp/ts-out.nLrT9T   -1       0.00/0.00/0.00 H
1    finished   /tmp/ts-out.a7gKgI   -1       0.00/0.00/0.00 H
2    finished   /tmp/ts-out.dER38R   -1       0.00/0.00/0.00 H
3    skipped    (no output)                                  [2]&& 4
4    skipped    (no output)                                  [3]&& 2

hca@wx:~$ tsp -I dvb --adapter 0 --delivery-system "DVB-S2" --lnb "5150000000" --frequency 4148000000 --modulation QPSK --symbol-rate 2586148 --fec-inner "3/5" --roll-off 0.2 --polarity "horizontal" -P mpe --pid 0x03E9 --udp-forward --log -O drop
Wrong option I.

Looking at the documentation I think the command is ok, but how to i determine what is wrong with option I,

lelegard, 2021-01-08:
> The machine is an Intel not AMD and Debian was not happy about the AMD requirement in the deb but TSDuck built fine from source without any warnings or error

"amd64" is the generic Debian name for x64 architecture, for both AMD and Intel CPU. It also works on 64-bit Intel CPU. Maybe you use a 32-bit distro. What does `name -a` say?

> with the only extra requirement was curl.

The script `build/install-prerequisites.sh` installs all required dependencies on Linux and macOS. See [the build instructions here](build/install-prerequisites.sh).

> However the command I need to run is failing as a wrong input option. 

Your post is unclear about the command and option. Could you please copy/paste the exact command and its output?

vk-hca, 2021-01-09:
Yes I also thought "amd64" is the generic Debian name and was not sure why the issue with the deb file, but having used both Debian and Centos over they years I know both are are bit pedantic but both are rock solid.  In the early days were never had nice things like RPM and DEB installers, so I immediately went for the building with source option which had no apparent issues at all. 

Here is step by step what I have done so far.

- I have this install of Debian on couple of machines here but they are nuc's with no pci slots, so its a fresh install .
- I got the gziped sourced from here   wget https://github.com/tsduck/tsduck/archive/v3.24-2109.tar.gz and decompressed it.
- I followed the instruction on the build page here   https://tsduck.io/doxy/building.html
- I chose the "Building without specialized dependencies" option from that page with 
"make NOPCSC=1 NOCURL=1 NODTAPI=1"   because I have no CA, need for html or a deteck device.
however there were warnings about missing curl.

- so I installed curl and curl dev, issued "make clean", then "make NOPCSC=1 NOCURL=1 NODTAPI=1" again.
this time I watched the compile run with no error or warnings. so I ran " make install". I did not edit the make files.

- Perfect, commands all appear to work, as I posted above TSDuck finds and reports the card details with tslsdvb --verbose
- I did not install TSDuck dev as I did not expect to need it for this task. Also I did not run build/install-prerequisites.sh.

Since your response I ran "build/install-prerequisites.sh" and it completed installing the requirements without warning or error. Since then I have not rebuilt TSDuck. Maybe I should? There is no apparent change after the install of these additional items.

Here are answers to your questions in last post.

Here is uname output:

uname -a
Linux wx 4.19.0-13-686-pae #1 SMP Debian 4.19.160-2 (2020-11-28) i686 GNU/Linux

Here is the command and output supposed to be used to process the stream.  I think this was as used in a windows machine. 

tsp -I dvb --adapter 0 --delivery-system "DVB-S2" --lnb "5150000000" --frequency 4148000000 --modulation QPSK --symbol-rate 2586148 --fec-inner "3/5" --roll-off 0.2 --polarity "horizontal" -P mpe --pid 0x03E9 --udp-forward --log -O drop
Wrong option I.

And here is the same command and output were I changed it for linux device.  eg /dev/dvb/adapter0

tsp -I dvb --/dev/dvb/adapter0 --delivery-system "DVB-S2" --lnb "5150000000" --frequency 4148000000 --modulation QPSK --symbol-rate 2586148 --fec-inner "3/5" --roll-off 0.2 --polarity "horizontal" -P mpe --pid 0x03E9 --udp-forward --log -O drop
Wrong option I.

I am not sure if there is some thing wrong in the command, or is it trying to set an unsupported option in the card? I tried to add options for debug and verbose with no success the output remains "Wrong option I". 

The card is running with signal, and as I said before worked with normal DVB-s2 tv in windows.

Out of interest this is the weather satellite project using TSDuck and the source of the command line I am using https://github.com/sam210723/himawari-rx

Thank again for your help on this.

lelegard, 2021-01-09:
> `uname -a`
>
> `Linux wx 4.19.0-13-686-pae #1 SMP Debian 4.19.160-2 (2020-11-28) i686 GNU/Linux`

This is a 32-bit distro. This is why the 64-bit (amd64 in Debian parlance) package could not be installed. Installing a 32-bit distro in 2021 is a bit a pity. You should use 64-bit distros. Some distros (not Debian) not even longer provide 32-bit binaries. I don't provide 32-bit prebuilt binaries for TSDuck on Linux.

> `tsp -I dvb --adapter 0 --delivery-system "DVB-S2" --lnb "5150000000" --frequency 4148000000 --modulation QPSK --symbol-rate 2586148 --fec-inner "3/5" --roll-off 0.2 --polarity "horizontal" -P mpe --pid 0x03E9 --udp-forward --log -O drop`
>
> `Wrong option I.`

"Wrong option I" is not a TSDuck message. There must be some other command in your path which is named tsp but has nothing to do with TSDuck. Try `which tsp` and `tsp --version` to help characterise which kind of executable it is.

Your command is otherwise syntactically correct. Just done a copy/paste on my machine and the options are accepted.

vk-hca, 2021-01-09:
Ok thanks for that, that is a bit strange, I don't remember ever having a 32bit Debian install, I thought it was the same thumb drive I used on the nuc. which reports as below.

hca@nuk2:~$ uname -a
Linux nuk2 4.19.0-6-amd64 #1 SMP Debian 4.19.67-2+deb10u1 (2019-09-20) x86_64 GNU/Linux
hca@nuk2:~$

Bear with me, there's about 6 install sticks of various things from over the years, maybe I used the wrong one or its related to the processor in the MB, although its supposed to be 64 bit spec.
Intel® Atom™ Processor D410 (1.6 GHz) 
Data width	64 bit
The number of CPU cores	1
The number of threads	2
- Microsoft® Windows® 7 / 7 64-bit / Vista™ / Vista™ 64-bit / XP / XP 64-bit compliant

Thanks for the confirmation on the input command, thats found the first issue.
Aside from that tsp --version returned a number that increased every time, correct its not TSDuck! 

tsp -V
Task Spooler v1.0 - a task queue system for the unix user.
So I removed it and now I get the following
hca@wx:~/tsduck-3.24-2109$ tsp --version
tsp: TSDuck - The MPEG Transport Stream Toolkit - version 3.24-2109

Now the command line out puts :
hca@wx:~/tsduck-3.24-2109$ tsp --verbose -I dvb  -d /dev/dvb/adapter0 --delivery-system "DVB-S2" --lnb "5150000000" --frequency 4148000000 --modulation QPSK --symbol-rate 2586148 --fec-inner "3/5" --roll-off 0.2 --polarity "horizontal" -P mpe --pid 0x03E9 --udp-forward --log -O drop
* dvb: using /dev/dvb/adapter0 (DVB-S, DVB-S2)
* Error: dvb: DVB frontend FE_SET_TONE error: Remote I/O error

which is a kernel ioctl response. Now in this particular case there is I have need for 22khz or for that matter diseq so I could rebuild with out the call, but its unlikely driver dos not support 22khz for KU universal lnbs.

Here is dump of loaded modules, 
hca@wx:~/tsduck-3.24-2109$ lsmod
Module                  Size  Used by
fuse                      90112  3
appletalk              28672  0
psnap                  16384  1 appletalk
llc                        16384  1 psnap
ax25                    49152  0
isl6421                16384  1
cx24120              28672  1
mt312                 20480  0
stv0297               16384  0
bcm3510             20480  0
lgdt330x             20480  0
nxt200x              20480  0
snd_hda_codec_via      20480  1
mt352                  16384  0
snd_hda_codec_generic    65536  1 snd_hda_codec_via
snd_hda_intel          32768  4
stv0299                20480  0
snd_hda_codec          94208  3 snd_hda_intel,snd_hda_codec_generic,snd_hda_codec_via
b2c2_flexcop_pci       16384  0
snd_hda_core           57344  4 snd_hda_intel,snd_hda_codec,snd_hda_codec_generic,snd_hda_codec_via
b2c2_flexcop           28672  1 b2c2_flexcop_pci
snd_hwdep              16384  1 snd_hda_codec
joydev                 20480  0
s5h1420                20480  1 b2c2_flexcop
snd_pcm                81920  3 snd_hda_intel,snd_hda_codec,snd_hda_core
cx24113                20480  1 b2c2_flexcop
cx24123                20480  1 b2c2_flexcop
snd_timer              28672  1 snd_pcm
dvb_core              102400  3 b2c2_flexcop,stv0299,lgdt330x
snd                    61440  15 snd_hda_intel,snd_hwdep,snd_hda_codec,snd_timer,snd_hda_codec_generic,snd_pcm,snd_hda_codec_via
soundcore              16384  1 snd
coretemp               16384  0
sg                     28672  0
pcspkr                 16384  0
iTCO_wdt               16384  0
iTCO_vendor_support    16384  1 iTCO_wdt
serio_raw              16384  0
evdev                  20480  10
parport_pc             28672  1
ppdev                  20480  0
lp                     20480  0
parport                40960  3 lp,parport_pc,ppdev
ip_tables              20480  0
x_tables               24576  1 ip_tables
autofs4                36864  2
ext4                  532480  1
crc16                  16384  1 ext4
mbcache                16384  1 ext4
jbd2                   86016  1 ext4
crc32c_generic         16384  2
fscrypto               24576  1 ext4
ecb                    16384  0
crypto_simd            16384  0
cryptd                 20480  1 crypto_simd
aes_i586               20480  0
hid_generic            16384  0
usbhid                 45056  0
hid                   102400  2 hid_generic,usbhid
sd_mod                 49152  3
ata_generic            16384  0
i915                 1331200  2
ata_piix               32768  2
video                  40960  1 i915
libata                192512  2 ata_piix,ata_generic
i2c_algo_bit           16384  1 i915
psmouse               126976  0
drm_kms_helper        135168  1 i915
scsi_mod              172032  3 sd_mod,libata,sg
ehci_pci               16384  0
uhci_hcd               40960  0
r8169                  73728  0
ehci_hcd               65536  1 ehci_pci
i2c_i801               24576  0
lpc_ich                24576  0
drm                   323584  4 i915,drm_kms_helper
usbcore               188416  4 usbhid,ehci_hcd,uhci_hcd,ehci_pci
realtek                16384  1
mfd_core               16384  1 lpc_ich
libphy                 45056  2 r8169,realtek
usb_common             16384  1 usbcore
button                 16384  0

Firstly I should try and figure how I ended up with a 32 bit install and change it to 64 bit because its a waste but as I have built from source, not the deb, I assume everything should still work for now?

Possibly I should put a screen/kyb etc on the machine and install some thing like linux tv to verify these things first?

lelegard, 2021-01-09:
Prebuilt packages are always built on the latest version of each distro, 64-bit except Raspian which is built on ARM 32. For Debian, it is version 10.x (buster).

To rebuild a Debian package from sources, this is trivial:
~~~
build/install-prerequisites.sh
make deb
~~~
You get the `.deb` packages in the `installers` directory.

Tuner issues on Linux are usually due to driver or firmware error. Since the tuner is recognised as a DVB tuner, the driver should be ok. Maybe it requires a firmware which is not installed. Use the command `dmesg` to inspect kernel message and look for any error when initialising the device on boot and then on first use by `tsp`. Missing firmware messages can be found here.

To install most needed DVB firmare, use:
~~~
sudo build/install-openelec-dvb-firmware.sh
~~~

It it still does not work (this is another firmware from somewhere else), google for it....

vk-hca, 2021-01-10:
I think for now best close this issue with the assessment that the Technisat skystar S2 PCI board will not work in windows WES7.

I have located the correct stick and installing Debian 64bit now. I feel a bit stupid about that blue.

I have had a look about and see there are lots of tools and software for these cards from the dvb TV community. That sort of thing has no appeal to me but I will work through drivers etc with these provided tools to get it up as a tv card first then try for TSDuck like I did with wes7 windows.

Thank you so much for your time and comments till now. Hopefully after that I can come back and report you if the cards works or not.

closed 2021-01-10 by lelegard