tsduck/tsduck #709: Getting every line of output (Async Report) [Q]

tahaarian, 2021-01-11:
Hello @lelegard 

I hope you are doing well

According to your advice, we could developed our `java JNA` module that will detect some kind of events (we have used your History plugin).

Now we need to filter output log that will generate with History plugin. 

Actually we have prepared a module that could handle some command like below script

`tsp -v -m --synchronous-log -I ip -r 239.0.0.1:1234 -P history -c --suspend-packet-threshold 60 --ignore-stream-id-change -O drop`

as you could see, we used` drop (output)` that will write automatically to console. but we need to get each line of output log. 

for example, we have gotten below log from our module and we need to catch each line such as a String param or any object that could be manageable

[image]

but we can not find any related function that could use for that 

we will appreciated if you could help us 

Taha

lelegard, 2021-01-11:
In a C++ application, the typical way would be to supply an instance of your own subclass of `ts::Report`, overriding the method `writeLog()` to get the logged lines. Take care to make it thread-safe since this object is potentially called at the same from multiple threads (each plugin executes in its own thread).

In Java, there is no direct way. The `ts::Processor` instance needs an instance of `ts::Report` but there is no way to create a Java class which inherits from a C++ class. I had a similar question from Python developers and I made the same answer.

As a workaround, you can use a named pipe. Use the file name of the named pipe with option `--output-file` in plugin `history`. You can read the other end of the named pipe from the Java code and get the output lines of the plugin.

tahaarian, 2021-01-12:
Dear @lelegard 

Thank you for your quick respond.

We will appreciated if you could describe the java solution 

Actually we have wrote bellow script but we can not drop logs in output file and `result.log` was empy

`tsp -v -m --synchronous-log -I ip -r 239.0.0.1:1234 -P history -c --suspend-packet-threshold 60 --ignore-stream-id-change -O drop > F:\tsduck\result.log`

Thank you in advance

lelegard, 2021-01-12:
This is a shell issue, not a TSDuck issue. As with all shell command lines, the logs are written to standard error to avoid messing with standard output. This is required when tsp is used as a pipe with the standard output carrying a transport stream. In your command, you redirect standard output, not standard error. In a shell, you would write `tsp ... 2>log`. You are apparently using the old DOS CMD. I do not know the syntax for this one, I never use it. On WIndows, I use bash or PowerShell, never the obsolete DOS CMD.

lelegard, 2021-01-12:
> We will appreciated if you could describe the java solution

What do you mean by this? The java binding are quite simple, they only give access to the high level TSProcessor class. They are described [here](https://tsduck.io/doxy/javabindings.html) and [here](https://tsduck.io/doxy/group__java.html) with some examples [here](https://github.com/tsduck/tsduck/tree/master/sample/sample-java).

tahaarian, 2021-01-12:
> This is a shell issue, not a TSDuck issue. As with all shell command lines, the logs are written to standard error to avoid messing with standard output. This is required when tsp is used as a pipe [...]

May i ask you to prepare a script with pipe structure (related to our script requirement) ? our output log could be same as below :

```
* 2021/01/11 09:31:31 - history: 2745923: PID 302 (0x012E), PES stream_id is 0xE0 (Video 0)
* 2021/01/11 09:31:31 - history: 2745922: PID 301 (0x012D) suspended, clear, service 0x0001
* 2021/01/11 09:31:31 - history: 2746230: PID 301 (0x012D) restarted, clear, service 0x0001
* 2021/01/11 09:31:31 - history: 2746230: PID 301 (0x012D) suspended, clear, service 0x0001
* 2021/01/11 09:31:31 - history: 2746414: PID 301 (0x012D) restarted, clear, service 0x0001
```

I think that we are doing somethings wrong

Thanks

tahaarian, 2021-01-12:
> What do you mean by this? The java binding are quite simple, they only give access to the high level TSProcessor class. They are described [here](https://tsduck.io/doxy/javabindings.html) and [...]

I mean that we need to write this kind of log in file (according to your solution) then filter the output file (string based) same as your pipe solution 

For example we need to detect PIDs that will suspended or restarted, so we should write some string parser (filter included) 

Thank you in advance.

lelegard, 2021-01-12:
You have to read the TSDuck user's guide. The `history` plugin has a `-o` option to specify an output file name. You have to create a named pipe, a pipe with a file name, and specify the file name of that pipe with that `-o` option. On Unix systems (Linux, macOS), named pipes are created using `mkfifo` system call or shell command (same name). On Windows, I have no idea. But this is a system question, not a TSDuck question.

tahaarian, 2021-01-13:
Hello again @lelegard 

I am sorry for continuously disturb. 
I have developed named pipe according to your advice and pass to `-O` plugin same as below screenshot. So as you could see, it catch `drop error`

Our named pipe is `\\.\pipe\LOCAL\test.txt` and would be validated but we have got the error.

[image]

Our final command is :

`tsp -v -m --synchronous-log -I file F:\java projects\mpeg-ts-analyzer\src\test\resources\test.ts -P history -c --ignore-stream-id-change --suspend-packet-threshold 60 -O drop \\.\pipe\LOCAL\test.txt`

We have got below error :

`* 2021/01/13 09:18:30 - Error: drop: no parameter allowed, use options only`

we will appreciated if you could help us 

Thank you in advance

lelegard, 2021-01-13:
You have placed the pipe name as a parameter of the `drop` plugin while this plugin takes no parameter at all (cf. the error message). The pipe should be used with option `-o` in the `history` plugin: `tsp ... -P history -o \\.\pipe\... ... -O drop`.

In addition to that, I may have good news for you. I tried to figure out if it could be possible to enhance the Java bindings in a way that is currently not possible with the Python bindings and have a Java Report class which could handle the logged messages in some user Java class. This is in fact possible, thanks to the power of the JNI spec.

So, in the TSDuck Java bindings, there is now an [AbstractAsyncReport](https://github.com/tsduck/tsduck/blob/master/src/libtsduck/java/tsduck/src/io/tsduck/AbstractAsyncReport.java) Java class which can be derived in the application to handle the TSDuck log messages. See [an example here](https://github.com/tsduck/tsduck/blob/master/sample/sample-java/SampleMessageHandling.java). You need to install the latest TSDuck version, available from [the nuightly builds](https://tsduck.io/download/prerelease/).

This way you may avoid the burden of named pipes (easy on Unix but a kind of pain on Windows).

tahaarian, 2021-01-13:
wow ... Thank you so much Mr @lelegard 

Such a great job. Thank you in advance

Its a great `pre-release` that could be handle our requirement. 
According to your way, we have installed the` nuightly build ( version 3.25-2168)` and then make an override class for  `AbstractAsyncReport`, but we have got below error 

[image]

Actually we have added `tsduck.jar` as a library. On the other hand when we have built your example, had gotten same error 

I think that is is related to class file version. 

Taha

lelegard, 2021-01-13:
The binaries were compiled with the latest LTS JDK from AdoptOpenJDK:
~~~
PS C:\WINDOWS\system32> java --version
openjdk 11.0.9.1 2020-11-04
OpenJDK Runtime Environment AdoptOpenJDK (build 11.0.9.1+1)
OpenJDK 64-Bit Server VM AdoptOpenJDK (build 11.0.9.1+1, mixed mode)
~~~
I wonder how I could enforce a class format which compatible with Java 8.

In the meantime, you may rebuild the Java bindings from sources. There are only 9 `.java` files in the TSDuck Java bindings.

tahaarian, 2021-01-13:
Thank you so much. 
I have built it for java 8

Great job

lelegard, 2021-01-13:
I modified the compilation options for the Java bindings. It should be compatible with Java 8 now (but not older than 8). This will be available by tomorrow, after the automatic creation of the [nightly builds](https://tsduck.io/download/prerelease/). Please wait for version 3.25-2171 or higher, download, test the Java bindings with Java 8 and please report here if this is ok.

tahaarian, 2021-01-19:
Hello @lelegard 

Thank you so much for your support. we could use it for `java 8`. such a great modification.

Just a point. Actually now we should install your release build and catch native library from` OS PATH `(according to your below code)

```java
    public static synchronized void loadLibrary() {
        if (!loaded) {
            String path = null;
            // Try in same directory as plugins (typically in a development environment).
            path = searchLibraryInPath(System.getenv("TSPLUGINS_PATH"), libName);
            // Try various typical Unix directories
            if (path == null && !isWindows) {
                path = searchLibraryInPath("/usr/local/lib:/usr/lib:/usr/lib64", libName);
            }
            // Load the native library by explicit file name or using system search rules.
            if (path != null) {
                System.load(path.toString());
            }
            else {
                System.loadLibrary(libName);
            }
            initialize();
            loaded = true;
        }
```

But most of teams need to use your `TSDUCK `as a dynamic library that could attach to project (for example, gettting just `tsduck.dll` as library) same as below : 

[image]

As you could see, we have added your `dll `to our resources path in project, and need to do all functionality (included `AbstractAsyncReport -> logMessageHandler`) ... but we have got below errors 

[image]

We have modified your native loader that could load from `dll `but would be generated below error :

[image]

Briefly, i think it could be useful if we have dynamic loader for each `dll` (because you have implemented each part on different dll) that does not need to install your build 

We will appreciated if you could help us.

Thank you in advance 

Taha

lelegard, 2021-01-19:
> Briefly, i think it could be useful if we have dynamic loader for each dll (because you have implemented each part on different dll) that does not need to install your build

Your implementation is uselessly complicated. You should not load any TSDuck DLL from your application, let TSDuck do the job. Plugins are loaded by the TSProcessor when needed. All TSDuck common features are in tsduck.dll / libtsduck.so.

If you have a standard installation, then there is nothing to do.

If you install TSDuck DLL / shared libraries in some non-standard locations, simply put them all in the same directory and make environment variable `TSPLUGINS_PATH` point to this directory (can be more than one directory, just like in `$PATH`). See the user's guide, chapter 3, `tsp` section, "Plugin activation options" sub-section. See also section 6.1.

closed 2021-12-20 by tahaarian

udhayNuk, 2022-11-28:
@lelegard 
how to Get pcr sysnchronous log each line of output using tsp
which command was used how to display only the audio and video sync in output log
please reply me

thank you in advance

lelegard, 2022-11-28:
@udhayNuk, please do not hijack older and non-related issues. If you have a new question, please open a new issue and describe your question precisely and thoroughly (your question is unclear).