tsduck/tsduck #710: [Q] Python examples

lars18th, 2021-01-12:
Hi @lelegard ,

I'm exploring how to create python scripts with TSDuck to go forward from the simple shell scripts. And the current examples are very good. However, I feel that we need almost one more example. It seems that the two current examples only mimics a shell command encapsulated over the Python runtime. But, how we need to do for extracting data?

One example:

- A Python script that reads a network stream and populate the PAT and PMT tables inside some Python's dictionaries.

Please, can you provide _one_ example of such type of functionality?
I ask only if you have time to do it, because with this example it will be come more easy to learn about how to use the TSDuck as a library with the Python binding.

Thank you!

lelegard, 2021-01-12:
You cannot do that. TSDuck remains a C++ library with heavy usage of C++ features. A Python class instance remains a Python object. Python and Java bindings implement just a few selected high-level classes, typically TSProcessor and AsyncReport which is needed to use TSProcessor. Each Python or Java class is fully different from the corresponding C++ class. Each Python or Java class instance simply encapsulates a pointer to a corresponding C++ class instance.

Most C++ classes such as demuxes are implemented using TSDuck-defined interfaces to user-specified object. This works only with C++ objects.

For each arbitrary TSDuck class you would like to use from Python or Java, you need to write the corresponding class in Python and Java plus the more or less hacky glue code to interface it (very hacky in Python, ok in Java). Since TSDuck contains 600+ classes, the amount of work is not manageable.

Add to this that Python and Java are garbage-collected languages for which implementing finalization methods is _discouraged_ (!!!), then you get a barely manageable pool of objects. Consequently, by experience, a Python+C++ or Java+C++ application will never be as reliable as a pure Python-only, Java-only or C++-only application. If you limit the interactions to a few high-level objects such as TSProcessor, it remains manageable. If you implement hundredths of low-level mixed-language classes, forget about it, better rewrite a shadow version of TSDuck in Python or Java.

lars18th, 2021-01-12:
Hi @lelegard ,

Thank you for your clarification. And after reading your response then I have one more question: It will be (in the future) some new Python binding enhancements? Let me to explain why I ask for this...

- In the current Python examples we can do (more or less) the same as with the `tsp` tool. In fact, the advantage is that now it's quite easy to _control_ the processing from the Python code. However, if we want to _extract_ some data, the only solution is to "parse" the output. This output can be text content (from terminal and pipes), XML files or in some cases JSON files too. Then this is not very different to execute the processing from a shell script and pass the outcomes to the Python script.

In this scenario... and please do not understand that I am criticizing your work, perhaps it will be desirable to provide _some_ Python object that can be used to "read" basic content of the TS. For example, a good proposal can be the `tsanalyze` tool/plugin. With a Python binding of it we can obtain directly useful data from the TS in realtime directly in the Python code (like, "tsid", "list of services", etc.). In fact, a JSON output is already included, and perhaps it can be extended to populate a Python object with the obtained data.

What you think?
Regards.

lelegard, 2021-01-12:
I will probably not enhance the Python binding much. For two reasons: The first reason is subjective: writing more and more ugly glue code would be a quite long, tedious and honestly uninteresting job without real functional added value to the project. The second reason is more technical: Python has an integration mode than is quite unsuitable to software delivery that is briefly [explained here](https://tsduck.io/doxy/pythonbindings.html#pyimplem). Using `ctypes` to interact with Python significantly limits your options with a library such as TSDuck which heavily uses object oriented features. This is very different from Java where JNI  provides the full power of using Java classes and VM from C++ code with zero integration burden. This is why I may enhance the Java bindings a bit further that the Python ones.

lars18th, 2021-01-12:
OK. But what about a "simple" mode to "transfer" the JSON output to the Python code in realtime? Any idea to achieve this?

lelegard, 2021-01-12:
AKAIK, Python `ctypes` can only be used when Python calls C/C++ and returns synchronously. Here, you are referring to a C++ thread which would push some C++ data to some Python class, outside any synchronous call from Python. This is possible with Java and JNI, not with Python and `ctypes`. I may be wrong but, in that case, someone has to explain how to do.

lars18th, 2021-01-13:
Then, I feel there is not solution for the Python binding.
However, one last change to try: It will be possible to provide a Python binding for the JSON output? The idea is something like this:

```
import ts

# Create an asynchronous report to log multi-threaded messages.
# Initial level is verbose, using time-stamped messages.
rep = ts.AsyncReport(severity = ts.Report.Verbose, timed_log = True)
rep.info("TSDuck version: %s" % ts.__version__)

############################### NEW ###############################
# Create an asynchronous JSON data representation.
json_report = ts.AsyncJSON(rep, callback)

def callback(json_data):
    # User code to catch the change of the JSON output
    pass
############################### END ###############################

# Create a TS processor using the report.
tsp = ts.TSProcessor(rep)

# Set some global TS processing options.
...
# Set plugin chain.
...
tsp.plugins = [
    [' ... '],
    ['analyze', '--json-python', ... ],
    [' ... '],
]
...

# Run the TS processing and wait until completion.
tsp.start()
tsp.waitForTermination()
rep.terminate()
```

Then instead of populate the JSON file, the plugin `analyze` with the `--json-python` parameter, will construct an in memory representation of the JSON data (so, not really needed a JSON parser) and it calls to the Async. Callback function when it changes. So the user has the entry point to catch and process the changes.

What you think about this approach?

lelegard, 2021-01-13:
How would you call a Python function from C++ using `ctypes`?

lars18th, 2021-01-13:
> How would you call a Python function from C++ using `ctypes`?

You can read the section on callbacks in the cyptes documentation: https://docs.python.org/3/library/ctypes.html#callback-functions

Remember please, that I'm not requesting for a complete or huge Python binding. In order to not waste your time with non-useful tools, I ask only if you can, without much effort, provide a _simple_ way to extract data from the TS stream directly from the Python code _without_ using intermediate files.

lars18th, 2021-01-15:
Hi @lelegard ,

I suspect that you're working on this idea of adding callbacks to the python (and Java) bindings. Right? Then yeah!
Just to know: You'll implement something like the _JSON-native-object_ in the binding? If you can, what I want to obtain is a direct communication with the library data without the need of parsing some String data (like JSON or XML). You want to achieve this in some way?

Futhermore, please note this simple "typo" bug (in the Python code it references a Java class!):
https://github.com/tsduck/tsduck/blob/4497e40badffe25667eaad1eaf7f7a4a1c530ee6/sample/sample-python/sample-message-handling.py#L25

And thank you for this impresive/amazing project! :+1:

lelegard, 2021-01-15:
> what I want to obtain is a direct communication with the library data without the need of parsing some String data (like JSON or XML).

I won't do that.

The software architecture is not done for that. Plugins are command line oriented. They execute in distinct threads. They just produce files and logs. 

Additionally, passing complex hierarchical binary data between C++ and Python is the kind of long, boring and uninteresting work I already said I won't do.

The simplest and most reliable way is to serialise those complex hierarchical binary structures. And serialising JSON data means ... generating JSON text (same for XML). Deserialising JSON text is trivial in Python, I think.

So, the idea is that a new option `--json-line` has been added everywhere `--json` was used. In this case, the JSON text is on one line (just like in REST API's) and produced as a log line. In addition to that, you just noticed that it was possible to provide a `Report` class in Python or Java which intercepts log messages: instead of logging to the console, the `TSDProcessor` instance and its plugins push their messages to an application-defined object. This object will then receive the one-line JSON data just like any message. The `--json-line` option has one optional value, a text prefix. For instance, using `--json-line='@FOO_JSON@'`, the log line containing the JSON text contains `@FOO_JSON@` just before the JSON text. It is consequently easy for the application object to filter the right line. The JSON text is everything that follows this prefix.

Thanks for reporting the typo, this was of course the remaining of a copy/paste from the equivalent sample in Java.

lars18th, 2021-01-15:
Hi @lelegard ,

Thank you for the upgrade! Your vision about maintining the KISS model is without doubt a good idea. I feel the new `Report` class binding will be suficient. Anyway can you improve the `tsscan` tool including the JSON output? At time, it's the only tool that I think will be improved much with such output.

lelegard, 2021-01-15:
`tsscan` has an XML output through the option `--save-channels`.

lelegard, 2021-01-15:
See https://github.com/tsduck/tsduck/blob/master/sample/sample-python/sample-analyze-ts.py

lars18th, 2021-01-16:
> `tsscan` has an XML output through the option `--save-channels`.

True, you're right. That's the XML format that can be redirected to the STDOUT. So, you think that has sense include a similar `--xml-line` output in TSDuck? Please only consider it if it makes _real_ sense.

 > See https://github.com/tsduck/tsduck/blob/master/sample/sample-python/sample-analyze-ts.py

Good example! Thank you! :+1:

lelegard, 2021-01-16:
`tsscan` is a separate process, not something that can execute inside the Python process using a `TSProcessor` instance. Don't make things more complicated than needed. On a programming standpoint, it is probably easier to let `tsscan` create a real XML file on disk and then load it in Python. Just run the process without the burden of pipes, redirections and reading the pipe line by line. Just load an XML file.

lars18th, 2021-01-16:
OK, then if you agree we can close this issue?

Thank you @lelegard for all updates.

closed 2021-01-17 by lelegard

lars18th, 2021-01-18:
Thank you @lelegard for adding too the `--log-xml-lin`. I feel these two output modes (XML and JSON) will be sufficient to develop complex Python (and Java) scripts using the TSDuck toolbox.

Regards.

lars18th, 2021-01-22:
Hi @lelegard ,

Thank you too for the XML-to-JSON conversion. :+1: