tsduck/tsduck #711: [FEAT] Fork plugin with unconditional subprocess kill

lars18th, 2021-01-13:
#### Is your feature request related to a problem?
I want to start a child process when running the `tsp` tool. And the subprocess don't use PIPE communication, only NETWORK streams. So I use this:
```
tsp -I ip 239.0.0.1:1234 -P until -s 30 -P ... -P fork -i 'my-tool 239.0.0.1:1234' -O drop
```
As you can see the processing of TSDuck is done _before_ the fork, so the packets are not required to flow over the subprocess (and they are dropped at end). But the problem is that the child process _is not killed_ when the `until` plugin expires. Please, note that this is the _only_ problem with such command... all is working without troubles, and the only problem is the stop of the subprocess.

#### Proposed solution:

My suggestion is to add a new parameter `--force-kill` to the `fork` plugin. With this new parameter the subprocess will be enforced to kill when the TSP process terminates.
#### Alternatives:

I don't have any alternative to this.

#### Additional information:

Perhaps it will useful too, to add the parameter `--no-flow` to not waste efforts writing to the PIPE of the subprocess. This will be valid when the child process doesn't reads from the PIPE.

I hope you want to implement this.
Thank you!

lelegard, 2021-01-13:
We have already discussed this in #690 and my answer was that TSDuck is not a shell. What you want to do has nothing to do with transport stream packet processing. What you want to do is perfectly achievable using a shell script (some Unix shell or Windows PowerShell). You mentioned in #690 that you did not want a shell. But you cannot, at the same time, want something and refuse the tools which naturally solve your request.

lars18th, 2021-01-13:
Hi @lelegard ,

> We have already discussed this in #690 and my answer was that TSDuck is not a shell. What you want to do has nothing to do with transport stream packet processing. What you want to do is perfectly [...]

Sorry! I don't want to disturb... It seems you're right. I'll try to write a shell script to run the subprocess and catch from the script the fork signal to end. Perhaps this will be the more robust and simple solution to the problem.

Excuse me, please!

Just one related request: For the plugin `fork` perhaps a parameter `--timeout` will be desirable. This will be different from the `--seconds` as the last doesn't start to count if no data is coming. So the objective of the `--timeout` is to start the count when the process start. You agree with that? This will solve the problem when the input is `-I ip` and no packets are comming.

lelegard, 2021-01-13:
>  For the plugin fork perhaps a parameter --timeout will be desirable. This will be different from the --seconds as the last doesn't start to count if no data is coming

There is no option `--seconds` in the plugin `fork`.

lars18th, 2021-01-13:
> > For the plugin fork perhaps a parameter --timeout will be desirable. This will be different from the --seconds as the last doesn't start to count if no data is coming
> 
> There is no option `--seconds` in the plugin `fork`.

Sorry, sorry! I'm mixing some points... my related request is for the plugin `until` (and not the `fork`). With `until` the counter starts with the first packet, so a `--timeout` option will be desirable to stop the tsp processing if no data is comming.

Please, excuse me for the confusion!

lelegard, 2021-01-13:
This is not possible. There is no asynchronous feature in packet processing plugins. A plugin has the opportunity to execute code 1) on start, 2) on each packet, 3) on close.  If nothing happens during one hour between start and the first packet, no plugin will be called, no termination action can be triggered. That is the tsp architecture. It could have been different but this is the way it works.

Your concern could be addressed using the `--receive-timeout` in the input plugin `ip`. In the absence of packet during the timeout, the input terminates. That's all we have.

lars18th, 2021-01-14:
> Your concern could be addressed using the `--receive-timeout` in the input plugin `ip`. In the absence of packet during the timeout, the input terminates. That's all we have.

Thank you! That's the solution. The TSDuck documentation is huge (and very well documented), so it's difficult to learn all. :wink: 

Then I close this issue.
Thank you @lelegard !

closed 2021-01-14 by lars18th

lars18th, 2021-01-15:
Hi @lelegard ,

Sorry for commenting about this _closed_ issue. But I want to share my solution with other users. Perhaps you want to include it too as an example in the documention. It works like a charm without scripts and it's sufficient robust (in Linux):
```
tsp -I ip 239.0.0.1:1234 --receive-timeout 5000 \
    -P until -s 30 \
    ... \
    -P filter -n -p 0-8191 \
    -P fork 'my-tool 239.0.0.1:1234 & PID=$!; cat - > /dev/null; kill -9 $PID' \
    -O drop
```
In the lines ` ... ` you can add all the processing you want to execute inside the TSDuck chain. The source data will come from the `my-tool` using only the network, and not the PIPEs. This will simplifies the work when the target source is an external process that you can only call to it to start/stop.

Regards.