tsduck/tsduck #712: [Question] How to monitor PAT timings?

bertothunder, 2021-01-16:
Hi @lelegard, I'm still working on a minimal ETR290 monitoring plugin over TSDuck. This plugin implements all the logic to handle PAT and PMT; it derives from TableHandler, registers a SectionDemux, which is fed via feedPacket on processPacket() method, I've a handleTable() method that receives the PAT, gets all PMT pids and prepares it for the PMT that will arrive.  The next calls to HandleTable are working, I can decode the PMTs etc. 

So far so good, everything works.

However, the handleTable() method is never called again after the first time for both PAT and PMT. According to the programmer's guide, seems the table is constructed once and never constructed again, which I imply it makes my method not be called anymore.

If this is the case, how could I know when the PAT and PMTs are arriving in order to monitor the timings between each PAT and PMT? Shall I force a destroy of the table in the demux? Seems like "too much" to me, maybe there's a better way here?

Thanks!

lelegard, 2021-01-16:
The method `handleTable()` is called for each _new_ complete table. When the same PAT is broadcast again and again, it is sent once only. If a new PAT, typically with an incremented version number, is received then `handleTable()` will be called again with the new table. This is a typical usage for a receiver which does not want to run code again and again when nothing changes.

In your case, you want to be notified of every section, even if this is the same one all the time. For this, you need to implement `SectionHandlerInterface` which provides the method `handleSection()`. This method is called for each section.

The `SectionDemux` accepts a table handler and a section handler. Both are optional. You can specify one or the other or both, your choice:
~~~
SectionDemux(DuckContext& duck,
             TableHandlerInterface* table_handler = nullptr,
             SectionHandlerInterface* section_handler = nullptr,
             const PIDSet& pid_filter = NoPID);
~~~

To mesure the time interval between PAT's, you should probably count the number of TS packets between two occurrences of table id 0, section number 0 (in the unlikely case a PAT had more than one section). Because of latencies due to the buffering, do not use timers in your plugins, count packets and derive the time interval using bitrates and packet count (see functions such as `PacketInterval()`). Additionally, your plugin will also work on captured files.

bertothunder, 2021-01-18:
Hi @lelegard, I understand this sample principle would apply for PMTs? I guess the section won't affect in the case of the PMTs, or should I only check for section 0?

lelegard, 2021-01-18:
It is true that ISO 13818-1 says that a PMT should have only one section. So, all PMT sections should have number zero. But if you implement a tool which checks the correctness of the stream, you must not assume that the stream is correct, right? So I would recommend to keep the same checking for all kind of tables. And you may even add a test that all sections have number zero for tables such as PMT which should have only one section by the standard.