tsduck/tsduck #713: [Q] Trying to modify SDT table (name and provider) of SPTS stream

Litrona, 2021-01-17:
I am trying to change the SDT table, just the name of the program channel and the provider. However is not changed.
This is my cmdline:

`
tsp -d1 -v --realtime -I ip 238.0.0.1:1234 -P sdt -p Provider01 -n Channel01 -O ip 192.168.1.12:1234
`

And this is the debug output:

`
tsp: Debug: debug level set to 1
tsp: Debug: ====> tsp -d1 -v --realtime
* Debug: debug level set to 1
* Debug: registering processor plugin "sdt", status: ok
* Debug: ip: debug level set to 1
* Debug: ip: ====> tsp -I ip 238.0.0.1:1234
* Debug: ip: debug level set to 1
* Debug: ip: ====> tsp -O ip 192.168.1.12:1234
* Debug: trying to load "/usr/bin/tsplugin_sdt.so"
* Debug: /usr/bin/tsplugin_sdt.so: cannot open shared object file: No such file or directory
* Debug: trying to load "/usr/bin/sdt.so"
* Debug: /usr/bin/sdt.so: cannot open shared object file: No such file or directory
* Debug: trying to load "/usr/lib64/tsduck/tsplugin_sdt.so"
* Debug: /usr/lib64/tsduck/tsplugin_sdt.so: cannot open shared object file: No such file or directory
* Debug: trying to load "/usr/lib64/tsduck/sdt.so"
* Debug: /usr/lib64/tsduck/sdt.so: cannot open shared object file: No such file or directory
* Debug: trying to load "/usr/lib64/tsplugin_sdt.so"
* Debug: /usr/lib64/tsplugin_sdt.so: cannot open shared object file: No such file or directory
* Debug: trying to load "/usr/lib64/sdt.so"
* Debug: /usr/lib64/sdt.so: cannot open shared object file: No such file or directory
* Debug: trying to load "/usr/local/lib64/tsduck/tsplugin_sdt.so"
* Debug: /usr/local/lib64/tsduck/tsplugin_sdt.so: cannot open shared object file: No such file or directory
* Debug: trying to load "/usr/local/lib64/tsduck/sdt.so"
* Debug: /usr/local/lib64/tsduck/sdt.so: cannot open shared object file: No such file or directory
* Debug: trying to load "/usr/lib/tsduck/tsplugin_sdt.so"
* Debug: sdt: debug level set to 1
* Debug: sdt: ====> tsp -P sdt -p Provider01 -n Channel01
* Debug: tsp: buffer size: 89,240 TS packets, 16,777,120 bytes
* Debug: ip: setting socket reuse address to 1
* Debug: ip: binding socket to 238.0.0.1:1234
* ip: joining multicast group 238.0.0.1:1234 from local address 192.168.1.25
* ip: joining multicast group 238.0.0.1:1234 from local address 10.243.55.195
* Debug: ip: initial buffer load: 7 packets, 1,316 bytes
* ip: unknown initial input bitrate
* Debug: control commands: found 1 packet processor plugins
* Debug: ip: output thread started
* Debug: sdt: packet processing thread started
* Debug: ip: input thread started
* Debug: sdt: adding standards DVB to none
* sdt: SDT version 0 modified
* Debug: ip: input: got bitrate 4,000,000 b/s
* Debug: ip: input: got bitrate 4,000,000 b/s
`

The analysis report is:

`
===============================================================================
|  TABLES & SECTIONS ANALYSIS REPORT                                          |
|=============================================================================|
|  PID: 0x0000 (0)                                                       PAT  |
|-----------------------------------------------------------------------------|
|  0x00 (0, PAT), TID ext: 0x0001 (1)                                         |
|      Repetition: ..... 100  ms  Section cnt: ...... 100                     |
|      Min repet.: ...... 89  ms  Table cnt: ........ 100                     |
|      Max repet.: ..... 101  ms  Version: ............ 0                     |
|=============================================================================|
|  PID: 0x0011 (17)                                                  SDT/BAT  |
|-----------------------------------------------------------------------------|
|  0x42 (66, SDT Actual), TID ext: 0x0001 (1)                                 |
|      Repetition: ..... 504  ms  Section cnt: ....... 20                     |
|      Min repet.: ..... 503  ms  Table cnt: ......... 20                     |
|      Max repet.: ..... 504  ms  Version: ............ 0                     |
|=============================================================================|
|  PID: 0x1000 (4096)                                                    PMT  |
|  Service: 0x0001 (1) Live                                                   |
|-----------------------------------------------------------------------------|
|  0x02 (2, PMT), TID ext: 0x0001 (1)                                         |
|      Repetition: ..... 100  ms  Section cnt: ...... 100                     |
|      Min repet.: ...... 89  ms  Table cnt: ........ 100                     |
|      Max repet.: ..... 101  ms  Version: ............ 0                     |
===============================================================================
`

Am I doing anything wrong ?
Thx in advance.

Litrona, 2021-01-17:
OK, I found a way to do it.

`
-P svrename -p Provider01 -n Channel01
`
Is it ok ?

lelegard, 2021-01-17:
Hi,

An SDT potentially contains many services. If you want to rename one, you must specify which one is renamed:
~~~
$ tsp -P sdt --help
...
  -n value
  --name value
      Specify a new service name for the added or modified service. For new
      services, the default is an empty string.
...
  -p value
  --provider value
      Specify a new provider name for the added or modified service. For new
      services, the default is an empty string.
...
  -s id
  --service-id id
      Add a new service or modify the existing service with the specified
      service-id.
...
~~~
With the plugin `sdt`, if you want to modify one specific service, you must specify it using option `--service-id` or `-s`.

The plugin `svrename`, on the other hand, renames the first service in the PAT if none is specified:
~~~
$ tsp -P svrename --help
...
Usage: tsp -P svrename [options] [service]

Parameter:

  Specifies the service to rename. If the argument is an integer value (either
  decimal or hexadecimal), it is interpreted as a service id. Otherwise, it is
  interpreted as a service name, as specified in the SDT. The name is not case
  sensitive and blanks are ignored. If no service is specified, the first
  service in the PAT is used.
...
~~~
If what you want to do is just renaming a service without other hack on the SDT, the plugin `svrename` is more appropriate.

Litrona, 2021-01-18:
Ok, thanks.

closed 2021-01-18 by Litrona