tsduck/tsduck #714: Can tsp mux used as a Symulcrypt capable TS MUX

nikpakar, 2021-01-17:
Dear all, Sorry if its a dumb question. Any constructive feedback would be highly appreciate.

I want to build a MPEG TS Multiplexer capable of Symulcrypt interface with inputs from SRT sources and output on a Dektek ASI card.

Can I use tsp with input plugin SRT, output plugin Dektek and packet processor plugins MUX and Scrambler to achieve this ?

If its technically capable, would it be good and reliable enough to run in a production environment.

Any feedback would be highly appreciated.

Thanks

lelegard, 2021-01-17:
Hi,

What you want to do is quite feasible and demonstrated in section 5.2.19 in the user's guide. Just use `srt` as input plugin instead of `dvb` and forget about the TS and service renaming (unless you need this as well).

TSDuck was developed to address lab, test and demo purposes. Your case is the typical integration test bed for a conditional access system. It integrates with a real ECMG, a real EMMG and emulates everything else. I have used that kind of configuration many times over years.

I have never recommended TSDuck as a production system. Although the code is relatively robust by self-discipline and strict application of good software engineering practices, it is not qualified for production. If you use some CAS and encryption system, I assume that you provide some Pay-TV service. If your subscribers pay for the service, they probably expect some professional quality of service as provided by professional equipment.

Note that the random generation of control words (content encryption keys) has not gone through security certification. This is normally a very strict security requirement for production systems. But the code being open source, you can still review it.

That being said, after 20 years in the Digital TV business, most so-called professional software code I have seen is usually much worse than TSDuck, but this is just a personal opinion. TSDuck is provided "as is", for free, without guarantee and will never hold any responsibility in case of production issue. In a word, use at your own risks...

nikpakar, 2021-01-18:
Thank you so much for the detailed response. Every thing clear and understood.

One last clarification, i want to input many live sources over SRT and read some where tsp would not to time sync hence its not advisable to mux live sources.

Can you please advise about this.

Thanks again for your help.

lelegard, 2021-01-18:
>  tsp would not to time sync hence its not advisable to mux live sources.

I do not understand your precise problem. If you want to transmit a live source (SRT or whatever) to an ASI card, you must have a live source with a sustained rate. As already mentioned 10 or 20 times here, `tsp` is not a MUX. If you have one source, your output will be as reliable as the source. But I hope you do want to create a production MPTS by stacking many `merge` plugins with SPTS inputs. In that case, the various SPTS will be _merged_ but certainly not _muxed_.

tbertschi, 2021-01-18:
Hello,

I actually have a side question:

Can the "scrambler" plug-in accept a configuration for several ECMGs, in order to perform a real simulcrypt (e.g. scramble once, but protect the CW with multiple CAS ECMs), or can it only handle one ECMG? If not, would there be some sideways to somehow still achieve that?

lelegard, 2021-01-18:
No, there is no way to achieve that. This is where the limitation of TSDuck as a lab tool is obvious. It can be used with one CAS (a typical integration test task) but not several CAS (a typical production task).

I do not see any way to achieve this with TSDuck. In the DVB SimulCrypt head-end architecture, the SCS (SimulCrypt Synchronizer) really deserves its name: the _synchronizer_. It is a complex equipment which synchronizes equipments from various vendors:

- Gather all requirements for all ECMG, globally resolve some of them (minimum crypto period for instance), apply individual conflicting ones (delay start, delay stop for instance).
- Manage connections and backup connections to various SuperCAS ids.
- Generate all ECM's in advance, with respect to all CAS requirements.
- Handle redundancies of ECMG's, automatically enlarge crypo-periods for all when one fails.
- Handle redundancy of the SCS itself (share CW and future ECM's which are generated in advance)
- Etc...

Implementing one single stream with one CAS is relatively easy, although the TSDuck plugin `scrambler` is one of the most complicated ones, have a look at the source code. But implementing a production-grade DVB SimulCrypt SCS supporting multiple CAS and all redundancy requirements for production is quite complicated. This is why this is a matter of professional equipment. Nothing is free...