tsduck/tsduck #715: Adding support for ISDB-T Content ID desriptor (0x54)

trollcop, 2021-01-19:
#### Is your feature request related to a problem?
When running tstables --japan file.ts, the content descriptor is printed as follows:

```
      - Descriptor 2: Content (0x54, 84), 2 bytes
        Content: 0x00 (unknown) / User: 0xFF
```

What should print out:

```
Content: 0x00 (news regular/general) / User: 0xFF
```

#### Proposed solution:
Here's a list in the tsduck.names format for the ISDB-specific content genres, generated manually from the list in ARIB TR-B14v6.5 part 4 Appendix A:

https://gist.github.com/trollcop/749d8108245001cc03c24d419e468049

You will also need to add a hack similar to:
```
if ((duck.standards() & Standards::JAPAN) == Standards::JAPAN) {
    // Japan / ISDB uses a completely different mapping.
    return NamesMain::Instance()->nameFromSection(u"ContentIdJapan", ....
```

inside `tsNames.cpp` where it deals with `ts::names::Content`.

#### Alternatives:
Ignore this request

#### Additional information:
No other information to be reported. But it would be nice to see this implemented.

lelegard, 2021-01-19:
Very good analysis, thank you.

Now integrated. I also checked that ABNT uses yet another mapping and integrated it as well.

In summary: ETSI defined the descriptor and its semantics for DVB. ARIB reused the DVB descriptor but redefined the semantics for ISDB in Japan. ABNT reused ISDB but modified again the semantics for Brazil. What a mess...

trollcop, 2021-01-19:
Cool, the changes look what I expected them to be. I'll try it out whenever the daily builds finish.

trollcop, 2021-01-21:
I can confirm its working as expected in windows nightly build 2182:

```
      - Descriptor 4: Content (0x54, 84), 6 bytes
        Content: 0xA9 (junior/senior high school students) / User: 0xFF
        Content: 0x5F (other show) / User: 0xFF
        Content: 0xAB (lifelong education/qulification) / User: 0xFF
```

Which matches with the program title:
` Event name: "NHK高校講座　ベーシック数学「平方根を知る」🈑"`

Also it seems to be common to combine several categories together, not sure if it's a thing in DVB as well:

```
        Event name: "NHK俳句　題「氷る（こおる）」"
      - Descriptor 1: Content (0x54, 84), 4 bytes
        Content: 0x85 (literature/liberal arts) / User: 0xFF
        Content: 0xAB (lifelong education/qulification) / User: 0xFF
```

And, there's a minor typo there (must have been that in arib docs and I missed it: `qulification` -> `qualification`.

lelegard, 2021-01-21:
Typo fixed, thanks for reporting.

closed 2021-04-16 by lelegard