tsduck/tsduck #716: [FEAT] tstabcomp extensions

lars18th, 2021-01-21:
#### Is your feature request related to a problem?
I want to use the `tstabcomp` tool from Python scripts; and I've some ideas to extend the funcionatlity and improve the usability.

#### Proposed solution:
1. The input and output of the tool doesn't support PIPES. For example, you can't use `-o -` to write to the stdout. And also the PIPE as input is not supported. I feel this lack is because the tool is oriented to an off-line processing. However, when using scripts the use of pipes has sense.
2. It will be desirable to have a Python binding of this tool too. In the same way as the `tsp` binding, with this tool integrated in the language will be possible to compile _any_ type of binary represenation of PSI/SI tables. This will open the use of the binding to more scenarios, as we can generate inside the script the XML representation of the data, and obtain the binary representation with one call to the library.

#### Alternatives:
The only alternative at time is to use intermediate XML/BIN files in a temporal directory. Is not a bad solution, but it can be improved to be more robust and productive.

#### Additional information:
I'm not sure about the effort to implement these two suggestions. But I'm sure they can improve a lot the productivity using the TSDuck from Python code.

lars18th, 2021-01-25:
Hi @lelegard ,

I see the huge improvement around the JSON ouput in last commits. Thank you! I feel they will improve a lot the use of the TSDuck library with third party languages. :+1: 

However, regarding specifically the `tstabcomp` tool. What you think? It has sense the PIPE support? And it has sense a Python binding? I know that more than one way exists to implement whatever functionality. But usually I prefer every time the most _clean_ solution. So prefer to ask you about your point of view.

Sorry if this request sounds insistent. I just want to hear from you once you have time to reply.
Thank you very much!

lelegard, 2021-01-25:
The XML-to-JSON conversion is documented in section 2.6 of the user's guide.

Using stdin/stdout (and consequently pipes) on `tstabcomp` has been added.

The natural programming interface for `tstabcomp` is the C++ class `SectionFile`. The command `tstabcomp` is just a small wrapper around it. I will look at a way to use it from Python and Java. Reading/writing binary, XML and JSON (write only) is not a problem. However, there are a lot of small command line options which are naturally encapsulated in `DuckContext` and `xml:Tweaks` which can be a pain to interface in other languages. Not really difficult, just long and boring. I may probably just leave the default values from Python and Java and simply don't interface them.

lars18th, 2021-01-26:
Hi @lelegard ,

Thanks for the comment. Any improvements you can make by wrapping some bindings (and I really want to emphasize this, as the idea is to provide simple and easy to implement improvements) will be welcome.

Regards.

lelegard, 2021-01-26:
The C++ class [`SectionFile`](https://github.com/tsduck/tsduck/blob/master/src/libtsduck/dtv/tsSectionFile.h) now has [Java bindings](https://github.com/tsduck/tsduck/blob/master/src/libtsduck/java/tsduck/src/io/tsduck/SectionFile.java) and [Python bindings](https://github.com/tsduck/tsduck/blob/master/src/libtsduck/python/ts/section.py).

Examples are provided in [Java](https://github.com/tsduck/tsduck/blob/master/sample/sample-java/SampleTables.java) and [Python](https://github.com/tsduck/tsduck/blob/master/sample/sample-python/sample-tables.py).

lars18th, 2021-01-26:
Hi @lelegard ,

This project is awesome, and your improvements are incredible. :+1: 
Please continue to improve it as best you can with the same rigorousness.

closed 2021-01-26 by lars18th