tsduck/tsduck #717: [FEAT] pkg-config file in dev deb package

gizahNL, 2021-01-21:
#### Is your feature request related to a problem?
I want to use meson to configure a project in which I will be using tsduck.so, meson "automatically" finds dependencies via pkg-config files (or CMAKE files)

#### Proposed solution:
Generate pkg-config .pc files and install them to an appropriate location (i.e.: /usr/lib/pkgconfig)
This would make setting up projects significantly easier

gizahNL, 2021-01-21:
```
prefix= /usr
exec_prefix=${prefix}
includedir=${prefix}/include/tsduck
libdir=${prefix}/lib

Name: tsduck
URL: https://tsduck.io
Description: tsduck
Version: 3.24.2109
Cflags: -I${includedir} -I${prefix}/include/PCSC
Libs: -L${libdir} -ltsduck -lsrt -lcurl  -lpcsclite -lpthread -lrt -ldl -lm -lstdc++
Libs.private: -Wl,-Bstatic -lcurl -Wl,-Bdynamic -Wl,-Bsymbolic-functions -Wl,-z,relro -Wl,-z,now -Wl,--as-needed -lnghttp2 -lidn2 -lrtmp -lssh -lssh -lpsl -lssl -lcrypto -lssl -lcrypto -llber -lldap -llber -lbrotlidec -lz  -lpcsclite -lpthread -lrt -ldl -lm -lstdc++
```
example I quickly handcrafted from the output of tsconfig

lelegard, 2021-01-21:
This could be done but I am not familiar with this. I have never used meson. I have always associated pkgconfig files with projects which use libtool and configure scripts. But I may be wrong.

Note that I have always hated projects with configure scripts because I don't like the idea of distributing incomplete sources (headers to be generated by configure). I don't like the idea of hard-coded paths in binaries (the infamous "prefix"). And those projects are rarely Windows-friendly (I don't like Windows much but it is used by many).

Anyway, I can do this. But I need some help about the pkgconfig file.

- What name should this file have? `tsduck.pc`?
- What is the meaning of `Libs.private`?
  - Why do you have this alternance of static and dynamic options?
  - Why do you reference exotic libraries such as `nghttp2`, `ssh` or `ldap` which are not relevant to TSDuck?

lelegard, 2021-01-22:
> I want to use meson to configure a project in which I will be using tsduck.so, meson "automatically" finds dependencies via pkg-config files (or CMAKE files)
> ....
> This would make setting up projects significantly easier

I had a look at pkconfig in general. I don't see how you could resolve dependencies from that. It can definitely extract compilation or link options. It also identifies the product name and version. But it does not seem to include _clear_ dependency information as implemented by traditional package managers such `rpm` and `dpkg`.

So, I see no added value:

- Compilation and linking options: Already provided by the command `tsconfig` (examples below). This command is equivalent to `curl-config` and similar `*config` commands from other projects.
- Dependencies: Already provided in `rpm` or `dpkg` databases, `brew` on macOS, user-contribution on AUR for Arch's `pacman`.
- Identification of name and version: Already done in package managers' databases. The command `tsversion` also provides version information, including dependent libraries.
- In fact, `pkg-config` does not seem much used for libraries. Taking the example of a standard installation of a recent Ubuntu 20.10, fully loaded with lots of development libraries, used to develop TSDuck among others (954 "lib*" packages are installed):
~~~
$ dpkg -l 'lib*' | grep '^ii' | wc -l
954
$ ls /usr/share/pkgconfig
adwaita-icon-theme.pc  dri3proto.pc   mobile-broadband-provider-info.pc  systemd.pc	      xf86vidmodeproto.pc
applewmproto.pc        fixesproto.pc  poppler-data.pc			 udev.pc	      xineramaproto.pc
bash-completion.pc     fontsproto.pc  presentproto.pc			 videoproto.pc	      xkbcomp.pc
bigreqsproto.pc        fontutil.pc    randrproto.pc			 xbitmaps.pc	      xkeyboard-config.pc
compositeproto.pc      glproto.pc     recordproto.pc			 xcmiscproto.pc       xorg-sgml-doctools.pc
damageproto.pc	       inputproto.pc  renderproto.pc			 xextproto.pc	      xproto.pc
dmxproto.pc	       iso-codes.pc   resourceproto.pc			 xf86bigfontproto.pc  xtrans.pc
dpmsproto.pc	       kbproto.pc     scrnsaverproto.pc			 xf86dgaproto.pc      yelp-xsl.pc
dri2proto.pc	       m17n-db.pc     shared-mime-info.pc		 xf86driproto.pc
~~~
I see only 44 `.pc` files, mostly from exotic packages, the `.pc` files or which contain only version identification. None of them come from library packages. I don't see how we can consider this as a standard way of managing dependencies for development libraries.

Potential issues:

- The `.pc` file contains at least one hard-coded root: `prefix=/usr` for instance. So, we need to customize it for each installation since many package managers provide ways to install in alternate roots. On the contrary, `tsconfig` uses its own location to identify the installation root and is transparent in case of alternate installations.
- The compilation options are hard-coded in `.pc` files, even when they come from a third party library. In your sample proposed `.pc` for TSDuck, there is a complex list of options in  `Libs.private`. I just realized that they come from `tsconfig --static-libs`, which includes `curl-config --static-libs`. But if we upgrade curl, this list may change. The command `tsconfig` invokes `curl-config` and gets the updated options. Nothing is hard-coded.

So, to summarize, I see no added value in `pkg-config`, I just see potential problems.

Sample usage of `tsconfig`;
~~~
$ tsconfig --help

Syntax: tsconfig [options]

  --cflags      pre-processor and compiler flags
  --libs        library linking flags
  --static-libs static library linking flags

  --prefix      installation prefix
  --bin         directory for TSDuck executables
  --lib         directory for TSDuck dynamic libraries (except plugins)
  --plugin      directory for TSDuck plugins
  --config      directory for TSDuck configuration files
  --include     include directory

  --help        display this help and exit
  --version     output version information
  --vernum      output the version information as a number

$
$ tsconfig 
version: 3.25.2180
prefix: /usr
bin: /usr/bin
lib: /usr/lib
plugin: /usr/lib/tsduck
config: /usr/share/tsduck
include: /usr/include/tsduck
cflags: -I/usr/include/tsduck  -I/usr/include/PCSC --std=c++11
libs:  -ltsduck -lsrt -lcurl  -lpcsclite -lpthread -lrt -ldl -lm -lstdc++
static-libs: /usr/lib/libtsduck.a  -Wl,-Bstatic -lcurl -Wl,-Bdynamic -Wl,-Bsymbolic-functions -Wl,-z,relro -Wl,-z,now -Wl,--as-needed -lnghttp2 -lidn2 -lrtmp -lssh -lssh -lpsl -lssl -lcrypto -lssl -lcrypto -llber -lldap -llber -lbrotlidec -lz  -lpcsclite -lpthread -lrt -ldl -lm -lstdc++
$
$ tsversion --version=all
tsversion: TSDuck - The MPEG Transport Stream Toolkit - version 3.25-2180
Built Jan 19 2021 - 16:00:19
Using GCC 10.2.0, C++ std 2011.03
Web library: libcurl: 7.68.0, ssl: OpenSSL/1.1.1f, libz: 1.2.11
SRT library: libsrt version 1.4.1
Dektec: DTAPI: 5.43.0.147
$
~~~

eli-schwartz, 2021-12-05:
Hmm, looks like this was implemented in commit bfe860f7693872f13361489d1f116dd3cbf5d19b?

For the record:

> I have always associated pkgconfig files with projects which use libtool and configure scripts. But I may be wrong.

It's the standard Unix mechanism, and is commonly created also by projects which build using simple Makefiles, and used by other projects that build using simple Makefiles. It is touted as a generic and trivially-parseable way to export compiler definitions, that does NOT (unlike custom foo-config style scripts such as curl-config) rely on nonstandard options or require executing a shell script.

Also, meson uses and recommends pkg-config, and has a robust pkg-config file generator (and when given a build description with a library, it can detect that the library was built using a dependency on tsduck.pc and correctly add that as a `Requires`, which @gizahNL perhaps finds valuable). ;)

> I see only 44 `.pc` files, mostly from exotic packages, the `.pc` files of which contain only version identification. None of them come from library packages. I don't see how we can consider this as a standard way of managing dependencies for development libraries.

This is the pkg-config "data-only" directory used by packages that want to export non-library information. These are not directly usable via `pkg-config <name> --cflags --libs`, but are usable via `pkg-config <name> --variable=<customvar>`. You have 44 files in that directory -- I have 64.

Conversely I have 1100 such files in /usr/lib/pkgconfig/ which is the directory for library information. Every one of these should correspond to a development library.

This should probably be fixed here! Currently https://github.com/tsduck/tsduck/blob/ab8f9c61b82771e88f3b607455d547d6dd8c0baf/src/libtsduck/config/Makefile installs a pkg-config file to the "data-only" location. 

> * The `.pc` file contains at least one hard-coded root: `prefix=/usr` for instance. So, we need to customize it for each installation since many package managers provide ways to install in alternate [...]

It is probably worth distinguishing between binary package managers and source package managers. Traditionally, the `/usr` is generated by a configure script or `meson setup` or a CMakeLists.txt, or as this project does, by sed'ing out the file template while installing files. This works well with source package managers.

Binary package managers typically do not allow installing in different locations. Some do, I suppose, but in this case it is possible to run `pkg-config --define-variable=prefix=/path/to/different/location`.

> * The compilation options are hard-coded in `.pc` files, even when they come from a third party library. In your sample proposed `.pc` for TSDuck, there is a complex list of options in [...]

The canonical method of handling this is to add the `Requires: libcurl` field for shared libs, or `Requires.private: libcurl` for static libs. Then the pkg-config program will include `libcurl.pc` when drawing its compiler dependency graph, and emit the curl project's libs (or static libs) alongside your project's libs.

So nothing is (or should be) hardcoded.

Seems like the final implementation in this repo handles that correctly, anyway. :)

lelegard, 2021-12-05:
Hi @eli-schwartz,

The `Requires` and `Requires.private` directives are essential indeed. That is what convinced me to add a `tsduck.pc` file, without hard-coded path. I still see many libraries without pkgconfig support. I am still not convinced that this is "universal" (anyone using some convention tends to consider it as universal :)). Anyway, if this can be useful, this is fine for me.

That being said, I do not use `pkgconfig` myself and I am not sure that `tsduck.pc` follows the right conventions. Any feedback is welcome.

> Conversely I have 1100 such files in /usr/lib/pkgconfig/ which is the directory for library information. Every one of these should correspond to a development library.

I am a bit confused about what you call "library" and "data-only" pkgconfig files.

Searching for pkgconfig files, I run this command on a few VM's which run on my Mac:
~~~
for d in $(find /etc /usr -name pkgconfig -type d 2>/dev/null); do echo $d : $(ls -1 $d | wc -l); done
~~~

On Fedora 35:
~~~
/etc/pkgconfig : 1
/usr/lib64/pkgconfig : 150
/usr/lib/pkgconfig : 4
/usr/share/qbs/modules/pkgconfig : 1
/usr/share/qbs/modules/Exporter/pkgconfig : 2
/usr/share/pkgconfig : 37
~~~

On Ubuntu 21.10:
~~~
/usr/lib/x86_64-linux-gnu/pkgconfig : 95
/usr/lib/pkgconfig : 1
/usr/share/qbs/modules/Exporter/pkgconfig : 2
/usr/share/qbs/modules/pkgconfig : 1
/usr/share/pkgconfig : 45
~~~

On Debian 11.1:
~~~
/usr/lib/pkgconfig : 1
/usr/lib/x86_64-linux-gnu/pkgconfig : 27
/usr/lib/x86_64-linux-gnu/org.gnome.Music/pkgconfig : 1
/usr/share/pkgconfig : 44
~~~

On Linux Mint 20.2:
~~~
/usr/lib/pkgconfig : 3
/usr/lib/x86_64-linux-gnu/pkgconfig : 44
/usr/share/pkgconfig : 44
/usr/share/qbs/modules/pkgconfig : 1
/usr/share/qbs/modules/Exporter/pkgconfig : 2
~~~

On Alpine Linux 3.14.3:
~~~
/usr/lib/pkgconfig : 22
/usr/share/pkgconfig : 1
~~~

On Raspbian 11.1 (Raspberry Pi):
~~~
/usr/lib/pkgconfig : 1
/usr/lib/arm-linux-gnueabihf/pkgconfig : 47
/usr/share/pkgconfig : 43
~~~

So, the "right" location is far from obvious...  `/usr/share/pkgconfig` looks like the best common directory.

Could you explain or post a pointer which precisely explains what kind of pkgconfig file should be installed where?

closed 2022-09-23 by lelegard