tsduck/tsduck #719: [FEAT] get lcn digital terrestrial in Italy

cristianpuddu, 2021-01-26:
I have a question about LCN.
When I use Tsscan (or tsanalyze)  don't show lcn (country: Italy).
So, I had the idea of using "psi  --default-pds 0x00000028" and this fix the problem. But in this way,I have the lcn separately from the rest, with more checks, many files xml and analysis to do
I understand that the problem is not Tsduck but my question is:  Can I get the lcn by Tscan/tsanalyze, or implement -default-pds 0x000028 in the tsscan? 
Thanks

lelegard, 2021-01-26:
Hi @cristianpuddu 

The LCN is always a pain. The concept of LCN does not even exist in MPEG or DVB. So, it is usually defined in private descriptors. But there are almost as many different (private) ways of signalling the LCN as there are countries or operators. I see distinct private "LCN-defining descriptors" from DTG / OFCOM, EACEM / EICTA, Eutelsat, NorDig, BSkyB. And this is for DVB only, add ATSC and ISDB to complete the landscape (with slightly different ways of numbering channels). And also add to this mess that some operators create invalid signalisation by omitting the mandatory private descriptor specifier (which is your case if you had to specify a default PDS). You can see that getting LCN is very different from the rest of the technical PSI/SI. 

In the general case, getting the LCN is a sort of black magic. When you know _your_ configuration, which private descriptor, in which table, and how far the signalisation is incorrect, then it becomes easy. But it can never be a generic method for other operators.

Could you at least post here a binary capture of the NIT (`-P tables -t 0x40 -b nit.bin`) ?

Which network is this? Italy terrestrial TV? Sat or cable? If you have a USB or PCI tuner to capture terrestrial TV, it would be interesting to get a snapshot of all TS in the Italian terrestrial TV for the [TSDuck stream repository](https://tsduck.io/streams/). This is the way we collectively gain some knowledge on the various existing network practices.

lelegard, 2021-01-26:
But, yes, having `--default-pds` in `tsscan` and `tsanalyze` would help. I should have done this before for the sake of consistency, not only for the LCN.

lelegard, 2021-01-26:
> But, yes, having --default-pds in tsscan and tsanalyze would help.

Done it commit 6ac9497c8336bcb813fba254a3768ee0fc0df6e2. It was just one line of code to add in each tool (and the same line in each of them).

cristianpuddu, 2021-01-26:
> Hi @cristianpuddu
> 
> 
> Could you at least post here a binary capture of the NIT (`-P tables -t 0x40 -b nit.bin`) ?

Thanks Lelegard,  ok in the next  days I'll do this

> Which network is this? Italy terrestrial TV? Sat or cable? If you have a USB or PCI tuner to >capture terrestrial TV, it would be interesting to get a snapshot of all TS in the Italian terrestrial [...]

Terrestrial tv.
I'm making some test with Raspberry and module tv Hat. It like a prope should do automatic scan at least every 6 hours  in my region (island of Sardinia).

You can see the results of ts analysis: you can  download them, from this page https://www.sardegnahertz.it/listing/ They include all the main Italian national mux

lelegard, 2021-01-27:
> You can see the results of ts analysis: you can download them, from this page https://www.sardegnahertz.it/listing/ They include all the main Italian national mux

Looking at your data, it is clear that the signalization is invalid according to DVB standards. The private data specifier descriptor is missing in the NIT and there is no standard way to interpret the 0x83 descriptor. EACEM, DTG, Eutelsat, NorDig, all use 0x83 for some form of LCN descriptor but each of them with a slightly different binary layout. Only the private data specifier descriptor indicates which layout should be used.

I have already seen missing private data specifier descriptors in closed satellite networks, where all receivers are provided by the commercial operators and their specifications may be tweaked to some proprietary specificities. But having a public national broadcaster doing the same mistake on an open public network is a shame! If I were an Italian citizen, I would fill an official complaint.

cristianpuddu, 2021-01-27:
ok I understand, thanks for the information...but why  TVs and receivers store  Italian lcn perfectly? I set the country in the TV as Italy, start the search and they are stored.

lelegard, 2021-01-27:
> ok I understand, thanks for the information...but why TVs and receivers store Italian lcn perfectly? I set the country in the TV as Italy, start the search and they are stored.

Maybe because the Italian regulator acted as a private operator and specified "Italian proprietary" conventions mentioning that there is no PDS descriptor. There is probably a public document which specifies this on some official Italian government or TV regulator site. Do you know where?

Or the TV manufacturers have a conservative / defensive approach on the Italian or European market and interpret tag 0x83 as an EACEM descriptor in the absence of prior PDS descriptor. Incorrect signalization happens and TV / STB vendors have learnt how to be resilient on the various markets.

But, I am surprised to see that on a public network. In France, the national regulator organized regular meetings with experts from all actors in the country (operators, broadcasters, CAS vendors, equipment manufacturers, etc). 10-15 years ago, I used to attend those meetings during a couple of years, representing the company I worked for at the time. All technical decisions were heavily discussed and no blatant non-standard feature would have been accepted if a standard alternative (such as a PDS descriptor) was possible.

lelegard, 2021-01-27:
I found that the AGCOM is the Italian regulator for TV.

All their documents are in Italian and I cannot find an official document regarding the signalization rules on the DTTV. The document [Allegato 2alla delibera n. 685/15/CONS](https://www.agcom.it/documents/10179/3485599/Allegato+23-12-2015+1450874273437/6e6e83b0-f276-4111-a3c1-e0792e0f3ad4?version=1.1) says:

> e.1.Network Information Table (NIT)
> 
> network_name_descriptor
> service_list_descriptor
> cable_delivery_system_descriptor
> satellite_delivery_system_descriptor
> terrestrial_delivery_system_descriptor
> T2_delivery_system_descriptor
> logical_channel_descriptor

This is pretty vague and unclear. The document is not very technical (as far as I can see, not speaking Italian) and focuses on features. But mentioning logical_channel_descriptor technically requires private_data specifier descriptor.

The document [HD Book 3.0](https://dgtvi.tivu.tv/pdf/HD-BOOK-DTT-3.0.pdf) from "HD Forum Italia" quite expectedly says in section 7.3. (Private Data): _"When private descriptors are present in a broadcast, a private data specifier descriptor SHOULD be used (cf. EN 300 468) to identify the definer of the private descriptor."_ And the logical_channel_descriptor is then explicitly listed as one of these private descriptors. The [UHD Book 1.0.1](https://www.hdforumitalia.it/wp-content/uploads/2019/06/UHD-BOOK-Final-1.0.1-with-cover.pdf) is similarly specific in "7.2.2.3.2 Terrestrial delivery".

So, yes, your broadcaster is clearly violating the standards, including the Italian documents. TV sets work correctly only thanks to the built-in resiliency of manufacturers which are used to violations of standards !

lars18th, 2021-01-29:
Hi @lelegard ,

One question: And is not possible to add a parameter to "enforce" (or assume) that the LCN descriptor is using one of the known formats?

lelegard, 2021-01-29:
> One question: And is not possible to add a parameter to "enforce" (or assume) that the LCN descriptor is using one of the known formats?

Yes, that is precisely the role of the option `--default-pds`. It enforces the interpretation of a specific "private context" which enforces the interpretation of otherwise unknown descriptors when the signalisation is "invalid" in the sense of DVB.

This option was missing in `tspsi` and `tsanalyze` but I fixed this following this issue.

lelegard, 2021-01-29:
And, by the way, @cristianpuddu kindly provided Italian DVB-T snapshots, [available here](https://tsduck.io/streams/?name=italy-sardinia-dttv).

cristianpuddu, 2021-01-29:
> > But, yes, having --default-pds in tsscan and tsanalyze would help.
> 
> Done it commit [6ac9497](https://github.com/tsduck/tsduck/commit/6ac9497c8336bcb813fba254a3768ee0fc0df6e2). It was just one line of code to add in each tool (and the same line in each of them).

Hi @lelegard,
I have tested default-pds in tsscan. I expected in output also the column with the lcn but instead this is not displayed. am I asking too much?  :-)

lelegard, 2021-01-29:
> I have tested default-pds in tsscan. I expected in output also the column with the lcn but instead this is not displayed. am I asking too much? :-)

If LCN are defined in this TS, you should find them. Here is the output of `tsscan` I just did (on one single frequency to speed up the process):
~~~
$ tsscan --uhf-band --first-channel 35 --last-channel 35 --service-list
* UHF channel 35 (586 MHz), strength: 51%
  Transport stream id: 1, 0x0001

  LCN Name                   Provider ServId TSId   ONetId Type PMTPID
  --- ---------------------- -------- ------ ------ ------ ---- ------
    2 France 2               GR1 A    0x0101 0x0001 0x20FA 0x01 0x006E 
    3 F3 Paris Ile-de-France GR1      0x0111 0x0001 0x20FA 0x01 0x00D2 
   14 France 4               GR1 A    0x0104 0x0001 0x20FA 0x01 0x0136 
   27 franceinfo:            GR1 A    0x0106 0x0001 0x20FA 0x01 0x0262 
   30 BFM Paris              R1 TFL   0x0170 0x0001 0x20FA 0x01 0x02C6 
~~~

lelegard, 2021-01-30:
@cristianpuddu, can you see LCN on some TS from the scan or none at all? From the capture files you provided, the NIT in the file mux2canaleitalia.ts does not contain LCN descriptor. All other files have one. So, `tsscan` should give LCN for all muxes except mux2.

cristianpuddu, 2021-01-30:
@lelegard 
For example this is the Mux1 RAI. The most important mux and the only one available on VHF
Here is the output:
<pre><code>
tsscan --vhf-band --first-channel 9 --last-channel 9 --bandwidth 7 --default-pds 0x00000028 --service-list
* VHF channel 9 (205.500 MHz), strength: 61%
  Transport stream id: 1, 0x0001

  Name               Provider ServId TSId   ONetId Type PMTPID
  ------------------ -------- ------ ------ ------ ---- ------
  Rai 1              RAI      0x0D49 0x0001 0x013E 0x01 0x0102 
  Rai 2              RAI      0x0D4A 0x0001 0x013E 0x01 0x0101 
  Rai 3 TGR Sardegna RAI      0x0D4B 0x0001 0x013E 0x01 0x0100 
  Rai Radio1         RAI      0x0D4C 0x0001 0x013E 0x02 0x0103 
  Rai Radio2         RAI      0x0D4D 0x0001 0x013E 0x02 0x0104 
  Rai Radio3         RAI      0x0D4E 0x0001 0x013E 0x02 0x0105 
  Test HEVC main10   Rai      0x0D52 0x0001 0x013E 0x1F 0x012C 
  Rai News 24        RAI      0x0D53 0x0001 0x013E 0x01 0x0118 
</pre></code>

Version is 3.25-2199

lelegard, 2021-01-30:
Weird. I cannot reproduce this here because the local DTTV network has a correct signalization. To reproduce that kind of problem, I usually inject a modified TS in a modulator and loop back to a tuner. However, because of the pandemic, we all work at home and I have no modulator at home, only tuners.

I will try to simulate this somehow.

lelegard, 2021-02-03:
Hi @cristianpuddu,

The issue is fixed. I implemented a more generic and reusable way of locating LCN's which includes all known methods (EICTA/EACEM, Sky, Nordig, DTG, etc). This generic method correctly uses the generic options such as `--default-pds`. You may use the latest repo and rebuild or use the [latest nightly builds](https://tsduck.io/download/prerelease/).

Using your capture files and a tuner emulator, `tsscan` gives this. Don't pay attention to the frequencies, they are fake in the tuner emulator.

~~~
$ tsscan -d etuner.xml --uhf-band --service-list --last-channel 28 --default-pds eacem
* UHF channel 21 (474 MHz), strength: 100%, quality: 100%
  Transport stream id: 1, 0x0001

  LCN Name               Provider ServId TSId   ONetId Type PMTPID
  --- ------------------ -------- ------ ------ ------ ---- ------
    1 Rai 1              RAI      0x0D49 0x0001 0x013E 0x01 0x0102
    2 Rai 2              RAI      0x0D4A 0x0001 0x013E 0x01 0x0101
    3 Rai 3 TGR Sardegna RAI      0x0D4B 0x0001 0x013E 0x01 0x0100
   48 Rai News 24        RAI      0x0D53 0x0001 0x013E 0x01 0x0118
  100 Test HEVC main10   Rai      0x0D52 0x0001 0x013E 0x1F 0x012C
      Rai Radio1         RAI      0x0D4C 0x0001 0x013E 0x02 0x0103
      Rai Radio2         RAI      0x0D4D 0x0001 0x013E 0x02 0x0104
      Rai Radio3         RAI      0x0D4E 0x0001 0x013E 0x02 0x0105

* UHF channel 22 (482 MHz), strength: 100%, quality: 100%
  Transport stream id: 18385, 0x47D1

  ServId TSId   ONetId PMTPID
  ------ ------ ------ ------
  0x0001 0x47D1 0x217C 0x0064
  0x0002 0x47D1 0x217C 0x0020
  0x0003 0x47D1 0x217C 0x0021
  0x0004 0x47D1 0x217C 0x0022
  0x0005 0x47D1 0x217C 0x0023
  0x0006 0x47D1 0x217C 0x0024
  0x0007 0x47D1 0x217C 0x0025
  0x0008 0x47D1 0x217C 0x0026
  0x0009 0x47D1 0x217C 0x0027
  0x000A 0x47D1 0x217C 0x0028
  0x000B 0x47D1 0x217C 0x0029
  0x000C 0x47D1 0x217C 0x002A
  0x000D 0x47D1 0x217C 0x002B
  0x000E 0x47D1 0x217C 0x002C
  0x000F 0x47D1 0x217C 0x002D
  0x0010 0x47D1 0x217C 0x002E
  0x0011 0x47D1 0x217C 0x002F
  0x0012 0x47D1 0x217C 0x0030
  0x0013 0x47D1 0x217C 0x0031
  0x0014 0x47D1 0x217C 0x0032
  0x0015 0x47D1 0x217C 0x0033
  0x0016 0x47D1 0x217C 0x0034
  0x0017 0x47D1 0x217C 0x0035
  0x0018 0x47D1 0x217C 0x0036
  0x0019 0x47D1 0x217C 0x0038
  0x001A 0x47D1 0x217C 0x0039
  0x001B 0x47D1 0x217C 0x003A
  0x001C 0x47D1 0x217C 0x003C
  0x001D 0x47D1 0x217C 0x003B
  0x001E 0x47D1 0x217C 0x003E
  0x001F 0x47D1 0x217C 0x003D
  0x0020 0x47D1 0x217C 0x0037

* UHF channel 23 (490 MHz), strength: 100%, quality: 100%
  Transport stream id: 930, 0x03A2

  LCN Name                ServId TSId   ONetId Type PMTPID
  --- ------------------- ------ ------ ------ ---- ------
    4 Rete4 HD            0x0BBC 0x03A2 0x0110 0x19 0x00D6
    5 Canale5 HD          0x0BBD 0x03A2 0x0110 0x19 0x00D7
    6 Italia1 HD          0x0BBE 0x03A2 0x0110 0x19 0x0041
   55 Mediaset Extra      0x0BEF 0x03A2 0x0110 0x01 0x00FF
   66 Mediaset Italia Due 0x0BFA 0x03A2 0x0110 0x01 0x010A
[... 46 lines cut ...]
* UHF channel 27 (522 MHz), strength: 100%, quality: 100%
  Transport stream id: 43008, 0xA800

  LCN Name             Provider         ServId TSId   ONetId Type PMTPID
  --- ---------------- ---------------- ------ ------ ------ ---- ------
   14 Telesardegna     Telesardegna     0x0001 0xA800 0x3001 0x01 0x0820
  114 Telesardegna 1   Telesardegna     0x0002 0xA800 0x3001 0x01 0x0821
  214 Telesardegna 2   Telesardegna     0x0003 0xA800 0x3001 0x01 0x0822
  609 Telesardegna 3   Telesardegna     0x0004 0xA800 0x3001 0x01 0x0823

* UHF channel 28 (530 MHz), strength: 100%, quality: 100%
  Transport stream id: 297, 0x0129

  LCN Name              Provider  ServId TSId   ONetId Type PMTPID
  --- ----------------- --------- ------ ------ ------ ---- ------
   10 Videolina         Videolina 0x0001 0x0129 0x217C 0x01 0x0405
   13 TeleCostaSmeralda Videolina 0x0005 0x0129 0x217C 0x01 0x0407
  110 Videolina 2       Videolina 0x0002 0x0129 0x217C 0x01 0x0402
  210 Videolina  3      Videolina 0x0003 0x0129 0x217C 0x01 0x0403
  510 Videolina HD Test Videolina 0x0006 0x0129 0x217C 0x01 0x0406
  601 Videolina 4       Videolina 0x0004 0x0129 0x217C 0x01 0x0404
      Radiolina         Videolina 0x000A 0x0129 0x217C 0x02 0x0210
      Rad1              Videolina 0x000B 0x0129 0x217C 0x0C 0x006F
      Radio Bonaria     Videolina 0x000C 0x0129 0x217C 0x02 0x0070
~~~

cristianpuddu, 2021-02-03:
Hi @lelegard, 
I've tested last build and I can confirm that it works perfectly! 
Thanks!

lelegard, 2021-02-03:
You’re welcome. 
And thanks a lot for your donation, it is appreciated. 
Cheers 
-Thierry

closed 2021-08-24 by cristianpuddu