tsduck/tsduck #720: [Q] EIT short event descriptor weirdness

trollcop, 2021-01-30:
[eit.zip](https://github.com/tsduck/tsduck/files/5897654/eit.zip)

So I'm using tstables --japan on the 2 attached files.
Both contain 0x4D descriptor with text and description fields.
However, in case of eit.ts, only the 'event name' is displayed, but 'description' is not printed, as if its length is not included.

by manually parsing both tables I can see that the data is there and to the best of my knowledge, looks complete:

eit2.ts:
```
4e	table id
f
0 54 	length

5c 70 	service id
c3 	resv version current next
00 	section number
01 	last section number
7e 8e 	tsid
7e 8e 	onid
01 	segment_last_section_number
4e 	last_table_id

27 cf	event_id
e7 6c 15 00 00	start_time
03 00 00 	duration
00 39 		running | free_ca | desc_loop_length
	c1 	copy_control
	01 	len
	04 	data

	c4 	audio component
	09	len
	f2 03 10 0f
	ff 5f 6a 70
	6e

	54 	content
	02 
	00 ff 

	50 	component
	06 f1 b3 00 6a 
	70 6e 

	4d 	short_event
	1d 	len (29)

	6a 70 6e 	jpn
	0a 		event_name_length (10)
	3c 2b 3c 67
	4a 7c 41 77 
	23 31

	0e 		text_length (14)
	3c 2b 3c 67
	4a 7c 41 77
	46 62 4d 46
	23 31 

40 5c c7 4d	crc32

```

eit.ts:
```
47 40 12 10 00 
4e 
f
0 9c 

dc 70 
c1 
00 
01 
7c 8e
7c 8e 
01 
4e 

00 02 
e7 6c 15 00 00 
03 00 00 
00 81 

4d 	short_event

65	length(101)
6a 70 6e	jpn
12 	event_name_length(18)

46 7c 4b 5c
36 26 3b 3a
45 5e 3c 2b
3c 67 4a 7c
41 77 

4e 	event_text_lenth(78)

3f 27 24 4f 46 77 24 58 24 49 
3b 36 24 6a 24 4c 24 6b 24 72
32 66 24 2c 40 24 43 2f 24 3e 
3e 6f 24 4a 24 69 24 73 4d 2d
30 59 24 4e 31 7c 3b 33 3a 23
46 7c 31 5b 24 28 24 46 40 75
24 2d 4c 34 38 2b 24 38 3f 6c
24 52 24 62 24 3b 24 3a 

50 06 f1 b3 00 6a 70 6e 54 02 00 ff c1 01 84 c4 09 f2 03 10 0f ff 6f 6a 70 6e bf fe cf 29 ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff ff
```

however when running the above tstables thing, I get:
```
    - Descriptor 4: Short Event (0x4D, 77), 29 bytes
        Language: jpn
        Event name: "自主放送２"
        Description: "自主放送内容２"
``` for eit2.ts and only

```
      - Descriptor 0: Short Event (0x4D, 77), 101 bytes
        Language: jpn
        Event name: "日本共産党自主放送"
``` for eit.ts (but notice how length is correct at 101 bytes.

Is this a problem in the ts, the descriptors, some weirdness in EIT parsing or... ?

lelegard, 2021-01-30:
Hi @trollcop,

Trying a manual decode, copy/paste from your report:
[image]
Does it look good?

Could you please extract the binary EIT section and post it here?

I recommend the following scenario. First locate first and last packet of the incriminated section:
~~~
$ tsp -I file foo.ts -P tables -p18 -a --packet-index -O drop
...
* First TS packet: 163, last: 291
* EIT p/f Other, TID 79 (0x4F), PID 18 (0x0012)
  Section: 0 (last: 1), version: 4, size: 170 bytes
...
~~~
Then, extract it:
~~~
$ tsp -I file foo.ts -P until --packet 292 -P skip 163 -P tables -p18 -a -b eit.bin -O drop
~~~
Verification of the binary file:
~~~
$ tstabdump eit.bin 

* EIT p/f Other, TID 79 (0x4F)
  Section: 0 (last: 1), version: 4, size: 170 bytes
...
~~~

trollcop, 2021-01-30:
Hi,

Yes, the decode looks correct. What do you mean about finding the EIT packets, the 2 .ts files I posted was just that, a ts packet with p/f EIT table consisting of 2 sections (present/following), would further binary extracting it help something?

btw, running tsp I get the following (interestingly enough, the 'description' *IS* there:

```

* First TS packet: 0, last: 1
* EIT p/f Actual, TID 78 (0x4E), PID 18 (0x0012)
  Version: 0, sections: 2, total size: 318 bytes
  - Section 0:
    Service Id: 56432 (0xDC70)
    Original Network Id: 31886 (0x7C8E)
    Segment last section: 1 (0x01)
    Last Table Id: 78 (0x4E), EIT p/f Actual
    - Event Id: 2 (0x0002)
      Start UTC: 2021/01/30 15:00:00
      Duration: 03:00:00
      Running status: undefined
      CA mode: free
      - Descriptor 0: Short Event (0x4D, 77), 101 bytes
        Language: jpn
        Event name: "F|K\6&;:E^<+<gJ|Aw"
        Description: "?'$OFw$X$I;6$j$L$k$r2f$,@$C/$>>o$J$i$sM-0Y$N1|;3:#F|1[$($F@u$-L48+$8?l$R$b$;$:"
      - Descriptor 1: Component (0x50, 80), 6 bytes
        Content/type: 0x01B3 (MPEG-2 video)
        Component tag: 0 (0x00)
        Language: jpn
      - Descriptor 2: Content (0x54, 84), 2 bytes
        Content: 0x00 (unknown) / User: 0xFF
      - Descriptor 3: unknown (0xC1, 193), 1 bytes
        0000:  84                                         .
      - Descriptor 4: unknown (0xC4, 196), 9 bytes
        0000:  F2 03 10 0F FF 6F 6A 70 6E                 .....ojpn
  - Section 1:
    Service Id: 56432 (0xDC70)
    TS Id: 31886 (0x7C8E)
    Original Network Id: 31886 (0x7C8E)
    Segment last section: 1 (0x01)
    Last Table Id: 78 (0x4E), EIT p/f Actual
    - Event Id: 3 (0x0003)
      Start UTC: 2021/01/30 18:00:00
      Duration: 03:00:00
      Running status: undefined
      CA mode: free
      - Descriptor 0: Short Event (0x4D, 77), 101 bytes
        Language: jpn
        Event name: "F|K\6&;:E^<+<gJ|Aw"
        Description: "?'$OFw$X$I;6$j$L$k$r2f$,@$C/$>>o$J$i$sM-0Y$N1|;3:#F|1[$($F@u$-L48+$8?l$R$b$;$:"
      - Descriptor 1: Component (0x50, 80), 6 bytes
        Content/type: 0x01B3 (MPEG-2 video)
        Component tag: 0 (0x00)
        Language: jpn
      - Descriptor 2: Content (0x54, 84), 2 bytes
        Content: 0x00 (unknown) / User: 0xFF
      - Descriptor 3: unknown (0xC1, 193), 1 bytes
        0000:  84                                         .
      - Descriptor 4: unknown (0xC4, 196), 9 bytes
        0000:  F2 03 10 0F FF 6F 6A 70 6E                 .....ojpn
```

OK.. and actually... running tstables *without* `--japan` also prints the description (albeit as junk ascii, of course).

So this must be something related to ISDB decoding?

I did the extraction as you recommended, and got the following from tstabdump:

```

* EIT p/f Actual, TID 78 (0x4E)
  Section: 0 (last: 1), version: 0, size: 159 bytes
  Service Id: 56432 (0xDC70)
  TS Id: 31886 (0x7C8E)
  Original Network Id: 31886 (0x7C8E)
  Segment last section: 1 (0x01)
  Last Table Id: 78 (0x4E), EIT p/f Actual
  - Event Id: 2 (0x0002)
    Start UTC: 2021/01/30 15:00:00
    Duration: 03:00:00
    Running status: undefined
    CA mode: free
    - Descriptor 0: Short Event (0x4D, 77), 101 bytes
      Language: jpn
      Event name: "F|K\6&;:E^<+<gJ|Aw"
      Description: "?'$OFw$X$I;6$j$L$k$r2f$,@$C/$>>o$J$i$sM-0Y$N1|;3:#F|1[$($F@u$-L48+$8?l$R$b$;$:"
    - Descriptor 1: Component (0x50, 80), 6 bytes
      Content/type: 0x01B3 (MPEG-2 video)
      Component tag: 0 (0x00)
      Language: jpn
    - Descriptor 2: Content (0x54, 84), 2 bytes
      Content: 0x00 (unknown) / User: 0xFF
    - Descriptor 3: unknown (0xC1, 193), 1 bytes
      0000:  84                                         .
    - Descriptor 4: unknown (0xC4, 196), 9 bytes
      0000:  F2 03 10 0F FF 6F 6A 70 6E                 .....ojpn
```

attached is eit.bin

[eit-bin.zip](https://github.com/tsduck/tsduck/files/5897885/eit-bin.zip)

trollcop, 2021-01-30:
Sorry, i missed one packet. Both sections here.
[eit-new.zip](https://github.com/tsduck/tsduck/files/5897896/eit-new.zip)

lelegard, 2021-01-30:
> What do you mean about finding the EIT packets, the 2 .ts files I posted was just that,

Sorry, I did not notice that you already posted the TS files at the beginning of your initial post, my fault :(

I do not see the same thing:
~~~
$ tstabdump --japan eit-new.bin

* EIT p/f Actual, TID 78 (0x4E)
  Section: 0 (last: 1), version: 0, size: 159 bytes
  Service Id: 56432 (0xDC70)
  TS Id: 31886 (0x7C8E)
  Original Network Id: 31886 (0x7C8E)
  Segment last section: 1 (0x01)
  Last Table Id: 78 (0x4E), EIT p/f Actual
  - Event Id: 2 (0x0002)
    Start JST: 2021/01/30 15:00:00
    Duration: 03:00:00
    Running status: undefined
    CA mode: free
    - Descriptor 0: Short Event (0x4D, 77), 101 bytes
      Language: jpn
      Event name: "日本共産党自主放送"
      Description: "色は匂へど散りぬるを我が世誰ぞ常ならん有為の奥山今日越えて浅き夢見じ酔ひもせず"
    - Descriptor 1: Component (0x50, 80), 6 bytes
      Content/type: 0x01B3 (Video 1080i(1125i), 16:9 aspect ratio, without pan vectors)
      Component tag: 0 (0x00)
      Language: jpn
    - Descriptor 2: Content (0x54, 84), 2 bytes
      Content: 0x00 (news regular/general) / User: 0xFF
    - Descriptor 3: ISDB Digital Copy Control (0xC1, 193), 1 bytes
      Recording control: 2 (Copy can be made for only one generation)
      User-defined: 0x4 (4)
    - Descriptor 4: ISDB Audio Component (0xC4, 196), 9 bytes
      Content type: 0x02 (2)
      Component type: 0x03 (2/0 mode (stereo))
      Component tag: 0x10 (16)
      Stream type: 0x0F (MPEG-2 AAC Audio)
      Simulcast group: 0xFF (255, none)
      Main component: true
      Quality indicator: 0x2 (Mode 2)
      Sampling rate: 0x7 (48 kHz)
      Language code: "jpn"
      Text: ""

* EIT p/f Actual, TID 78 (0x4E)
  Section: 1 (last: 1), version: 0, size: 159 bytes
  Service Id: 56432 (0xDC70)
  TS Id: 31886 (0x7C8E)
  Original Network Id: 31886 (0x7C8E)
  Segment last section: 1 (0x01)
  Last Table Id: 78 (0x4E), EIT p/f Actual
  - Event Id: 3 (0x0003)
    Start JST: 2021/01/30 18:00:00
    Duration: 03:00:00
    Running status: undefined
    CA mode: free
    - Descriptor 0: Short Event (0x4D, 77), 101 bytes
      Language: jpn
      Event name: "日本共産党自主放送"
      Description: "色は匂へど散りぬるを我が世誰ぞ常ならん有為の奥山今日越えて浅き夢見じ酔ひもせず"
    - Descriptor 1: Component (0x50, 80), 6 bytes
      Content/type: 0x01B3 (Video 1080i(1125i), 16:9 aspect ratio, without pan vectors)
      Component tag: 0 (0x00)
      Language: jpn
    - Descriptor 2: Content (0x54, 84), 2 bytes
      Content: 0x00 (news regular/general) / User: 0xFF
    - Descriptor 3: ISDB Digital Copy Control (0xC1, 193), 1 bytes
      Recording control: 2 (Copy can be made for only one generation)
      User-defined: 0x4 (4)
    - Descriptor 4: ISDB Audio Component (0xC4, 196), 9 bytes
      Content type: 0x02 (2)
      Component type: 0x03 (2/0 mode (stereo))
      Component tag: 0x10 (16)
      Stream type: 0x0F (MPEG-2 AAC Audio)
      Simulcast group: 0xFF (255, none)
      Main component: true
      Quality indicator: 0x2 (Mode 2)
      Sampling rate: 0x7 (48 kHz)
      Language code: "jpn"
      Text: ""

$ 
~~~

Using the current repo (3.25-2205) and displayed on a macOS terminal window.

Does it look correct ?

trollcop, 2021-01-30:
Yes, its correct there.
I'm using win64 prerelease build
```
tstabdump: TSDuck - The MPEG Transport Stream Toolkit - version 3.25-2182
```
trying to grab 2205 now...

```
      - Descriptor 0: Short Event (0x4D, 77), 101 bytes
        Language: jpn
        Event name: "日本共産党自主放送"
      - Descriptor 1: Component (0x50, 80), 6 bytes

tstables.exe --version
tstables: TSDuck - The MPEG Transport Stream Toolkit - version 3.25-2205
```

Same with 2205

lelegard, 2021-01-30:
It is probably another Windows console issue.

I tried the same command on a Win10 VM, in three different types of console: old command console, powershell console and mintty ("git bash" window):

[image]

You can see that the mintty console displays correctly (or at least something that looks Japanese to me) while the two native Windows consoles display garbage. My Windows consoles are probably not programmed correctly for Japanese display but I notice that the "description" line is not displayed at all, just like you described. Since that line appears on the mintty window, we can say that the UTF-8 characters are correctly emitted by the application (this is the same executable with the same command line parameters in all three cases). But this specific sequence of characters is probably a problem for the native Windows consoles.

I have probably repeated this a thousand times here and elsewhere, but if you want to avoid problems when running technical tools (ie. non-Office desktop applications), avoid Windows. Or at least avoid Windows weird consoles, use reliable consoles with a reliable shell (Cygwin or MSys).

trollcop, 2021-01-31:
Welp, you can add Windows Terminal (Preview) to the list of consoles that don't display it. I guess I'm going to go ahead and create  a bug there, referencing this one, and hopefully someone at msft can figure it out.
It's so fucking weird to eat the entire line tho?! Consistently, at that?

trollcop, 2021-02-02:
Hmm, according to a comment in the Terminal bug, this is because you are piping output through (your own?) edition of "more" for paging, and that does something with the output. But yeah, anyway, this is probably (?) not a tsduck issue.

lelegard, 2021-02-02:
The pager which is used in all screen displays is the system one, typically `less` on Linux and macOS and `more` on Windows. So, if something is corrupted through this pager, this is the system's fault :) In all paged-display commands, you can try with option `--no-pager` to output directly to the console.

trollcop, 2021-02-02:
> you can try with option `--no-pager` to output directly to the console.

OK, so that actually works.
The output is correct.

`tstables --japan --no-pager eit.ts | more` results in garbage characters tho (i guess some console charset difference?)

Could there be something related to the way the data is piped to `more` or whatever it is on windows?

lelegard, 2021-02-02:
The output pager is managed by class `OutputPager` and the process and its pipe are created by class `ForkPipe`. The output bytes are simply written by the Win32 `WriteFile` function. See [here](https://github.com/tsduck/tsduck/blob/master/src/libtsduck/base/tsForkPipe.cpp#L570).

closed 2022-09-23 by lelegard