tsduck/tsduck #721: [Q] Python bindings in (Windows) PORTABLE package

lars18th, 2021-01-31:
Hi @lelegard ,

Any reason to not include the Python bindings inside the Portable Windows package?

I'm including this inside my python scripts to use the TSDuck _without_ install:
```
import sys
sys.path.append('C:/Portable/TSDuck/python')

import ts
...
```

I suggest to include this small subdirectory in the package. And perhaps the Java binding too.
And please, document (if you want) this simple method to use the binding with the portable version.

I hope this helps others as well.
Regards.

lelegard, 2021-01-31:
The Windows portable package only includes the tools and plugins, not the development environment. I can modify it to include the development environment for C++, Java and Python, but this will make it much bigger.

I understand that you are interested in Python only. But a C++ or Java developer will legitimately ask "why him and not me?"

lars18th, 2021-02-01:
Hi @lelegard ,

I have this "simple" solution: Create a new package named "dev-bindings.zip" that represents the portable version of the Python/Java/C++ environments.

At time, in Windows to "extract" the python binding I need to install, copy, deinstall the Windows version. Or do a git clone, or ZIP download... Several different solutions exist. However, a simple package will be welcome.

So, please consider it.
Thank you!

lars18th, 2021-02-02:
Hi @lelegard ,

I close this request as you've already implemented it https://github.com/tsduck/tsduck/commit/35c7f804aaa2f60aa04adf95377e7c9658f1de35

Thank you!

closed 2021-02-02 by lars18th