tsduck/tsduck #722: [Q] PIPE and Network inputs for Tuner Emulator?

lars18th, 2021-02-02:
Hi @lelegard ,

I see last changes incorporating the Tuner Emulator class. And I feel the idea is to provide some TS files as muxes to the entire toolbox.  So a pyshical pair of one modulator and one tuner will be not required to check the tools that use tuners, right?

That's a good idea. But I have a "simple" (not a full integration request) suggestion of an enhancement. Usually I work only with external tuners, in fact with _network tuners_, like SAT>IP servers and HDHomeRuns. I prefer these devices over internal or USB tuners as they are more flexible and robust. So my suggestion is to add the functionality to support not only local files as feeds of the Tuner Emulator, but also a remote pipes.

For example, with some SAT>IP servers you can receive the stream over HTTP (the standard ditactes a RTSP dialect). And with the HDHomeRuns with simple CLI commands you can tune/receive/stop the tuner. Then, assuming that the remote tuner provides the full TS, my suggestion can be summarized as:

- Add a PIPE input support for the Tuner Emulator. It will work starting a shell command (the one indicated by the user) and receiving the TS over a PIPE. More or less like does the fork-input plugin.
- Add Multicast/Unicast input support too. The concept is the same but using a UDP/RTP stream as input.

What you think?
I don't recommend a complex integration, just a more inputs. In fact, to use "external" sources we have at time sufficient plugins and sources to inject the stream. However, perhaps this enhancement will open the door to support other types of tuners.

Regards.

lelegard, 2021-02-02:
The tuner emulator is not a real feature of TSDuck. This is more an internal debug tool.

Having different technical types of sources is addressed from the beginning by the `tsp` or `tsswitch` input plugins.

Technically, the tuner emulator is completely useless with `tsp`, just use another input plugin, `file` or `ip`. What you call a network tuner is not a tuner as far as `tsp` is concerned, this is just an IP source which can be read by the `ip` plugin. Similarly, anything that is sent through a pipe is read by the default `file` input plugin (or `fork` input plugin in the case of multiple inputs with `tsswitch`).

The only use case for the tuner emulator, something that cannot be achieved by an input plugin, is when an application explicitly manipulates a tuner and, more importantly, decides to switch on different frequencies at arbitrary moments. Within the TSDuck toolset, only `tsscan` is concerned. I just quickly developed the tuner emulator to troubleshoot issue #719 and other future similar issues.

Reproducing such a problem requires either:
- The physical presence of a tuner in a specific geographical region. Generally not possible. Expensive on time and plane tickets :)
- In a lab with captured files, using as many modulators as there are muxes. Stream each file in a modulator using `tsp` and mix everything to the input of a tuner. Expensive on equipment.
- In a lab with captured files, quickly develop a tuner emulator which can be transparently used by `tsscan` to perform an emulated but realistic scan.

I chose the last option and found root cause for issue #719. And because I believe that all work should be an investment, I integrated the tuner emulator in TSDuck for faster reuse in future issues involving scanning a network in some remote country.

lars18th, 2021-02-02:
Hi @lelegard ,

Thank you for the response! Anyway, it will be interesting that at some point the TSDuck will support _real_ network tuners (SAT>IP server and HDHomeRuns). The `tsscan` tool is useful, and _shared_ tuners over the network are be far the best option to tune DTV signals in a development/research environment.

Regards.

lelegard, 2021-02-03:
These "network tuners" seem to have a proprietary software interface (just quickly looked at HDHomeRuns site). On the client side, there is some software which sends tuning requests over the network and then receives the stream. If TSDuck would support that kind of devices, there are three cases:

- The client side emulates the behaviour of a native tuner, using Linux TV or DirectShow API. In that case, those tuners are already supported by TSDuck, like any other tuner.
- The client side uses a _documented_ API or network protocol. It would be possible to develop a dedicated input plugin for TSDuck (or a variant of tuner emulator).
- The client side is a proprietary TV watching application without information in the communication protocol with the server with the actual tuners. In that case, forget about it...

So, if you have that kind of tuner, you need to check which one of the above cases applies. Then, we will see what can be done.

lars18th, 2021-02-03:
Hi @lelegard ,

In response to your request, both type of devices (SAT>IP servers, based on a CENELEC Standard; and HDHomeRuns that's a vendor device with an opensource API) are in the _second_ case: they work with a _documented_ API:

- SAT>IP: https://www.satip.info/resources/specification/
- HDHomeRuns: https://github.com/Silicondust/libhdhomerun

Notes:
- SAT>IP: Apart from service discovery (using SSDP), for the control plane and the transport plane it uses RTSP+RTP, with a small modifications mainly for retuning and control the pid-filtering. So it's easy to implement if you send the correct RTSP commands and listen for the RTP stream.
- HDHR: Two protocols exist. The one to support is the older, that is the targeted by the libhdhomerun. It's a binary protocol, but anyone can link with the library for Windows, linux, osx and bsd. The interface is common. Futhermore, a simple tool called `hdhomerun_config` exists for all platforms (including Windows). Try it installing in Ubuntu linux with `apt-get install hdhomerun-config`. Using this tool the interface can be similar to the RTSP protocol in the sense that you can send commands in the control plane and receive the stream over UDP/RTP unicast/multicast.

So, both protocols use the same architecture: A control plane based on commands and a transport plane based on RTP/UDP.

However, for both protocols at time simple tools exist to encapsulate all in one command (almost in a linux environment) and receive the stream over a PIPE. So my suggestion **at this point** is to support for the Tuner Emulator, not only FILES but PIPES. You have the code of the fork input plugin, and it will be easy (I think) to extend it for supporting this example:
```
<?xml version="1.0" encoding="UTF-8"?>
<tsduck>
  <defaults delivery="DVB-T" bandwidth="4,000,000" directory="/home/user/script-tuner"/>
  <channel frequency="474,000,000" file="mux1.ts"/>
  <channel frequency="482,000,000" pipe="satip_script --tune 482 --stdout"/>
  <channel frequency="490,000,000" pipe="satip_script --tune 490 --stdout"/>
  <channel frequency="498,000,000" file="mux4.ts"/>
  <channel frequency="506,000,000" pipe="hdhomerun_config_script --tune 506 --stdout"/>
  <channel frequency="514,000,000" pipe="hdhomerun_config_script --tune 514 --stdout"/>
  <channel frequency="522,000,000" file="mux7.ts"/>
  <channel frequency="530,000,000" pipe="cat mux8.ts" delivery="DVB-T2" bandwidth="6,000,000"/>
</tsduck>
```

That's just an example of what I request **now**. In order to follow the KISS model, this can be the first iteration. And note the `cat file.ts` example. This implementation has then the advantage that can be extended for other devices/protocols. However, if you want after this step to implement direct SAT>IP and/or HDHR support, then you can use software servers that implement these protrocols (different opensource projects exist of both). So you don't need any physical device like the ones I use.

I hope you want to consider almost this first iteration.
Thank you!

lars18th, 2021-02-03:
Hi @lelegard ,

Just to mention it: I'm using HDHomeRuns (DVB-T) for more than 10 years. And SAT>IP servers for around 6 years. And I will never use internal or USB tuners again. With the only exception of small USB tuners for use them with a notebook when testing terrestrial or cable signals in mobility. The reason is the pain of drivers. I want to use only open standards or known protocols. In addition, using devices over the network is more flexible than using them over a direct connection.

Regads.

lelegard, 2021-02-03:
Pipes have been added to the tuner emulator.

lars18th, 2021-02-03:
> Pipes have been added to the tuner emulator.

Thank you @lelegard ! I'll check it.

closed 2022-09-23 by lelegard