tsduck/tsduck #723: [FEAT]SCTE-35 preroll timing analysis

BartmanEH, 2021-02-02:
#### Is your feature request related to a problem?
No. Not a problem with tsduck per se, rather a problem analyzing an important parameter of SCTE-35 based systems--preroll.

#### Proposed solution:
Currently tsduck extracts SCTE-35 information into a concise table including details such as the PTS time that a SCTE-35 splice insert message defines. The additional information I'm looking for is to derive the associated preroll timing. SCTE-35 splice insert messages are originated as SCTE-104 messages from an automation system to an SCTE-104/35 compliant video encoder. The SCTE-104 message contains a pre_roll_time parameter which informs the video encoder how long in advance of the SCTE-35 execution time it should send the SCTE-35 command. The video encoder will take its liberties with this timing request but ultimate generate a key frame (I-frame) in the video at precisely the requested splice insert PTS time and SCTE-35 messages in advance of this PTS time roughly in accordance with the pre-roll time request. So what is needed is to analyses the resulting transport stream to ensure compliance of SCTE-104 directives and what is missing today in tsduck is the analysis of SCTE-35 preroll timing. The preroll time would be the difference in PTS value of the splice insert command's PTS value (the future time that the splice is to be executed) and the relative time when the command was sent. Since SCTE-35 packets do not have their own timing information (no PTS), this would need to be correlated to the corresponding Service the SCTE-35 command relates to, such as a video service, from which timing could be derived from PTS/PCR.

What I propose is to add an additional parameter to the SCTE-35 output table. Currently an example output of an example command line such as `tsp -P tables --tid 0xFC --packet-index -O drop -I file [redacted].ts`:
```
* SCTE 35 Splice Information, TID 252 (0xFC), PID 60 (0x003C)
  Short section, total size: 40 bytes
  - Section 0:
    Protocol version: 0
    Encryption: none
    PTS adjustment: 0x000000000 (0)
    CW index: 0x00 (0), tier: 0xFFF (4095)
    Command type: 0x05 (SpliceInsert), size: 20 bytes
    Splice event id: 0x01409DC6, cancel: 0
    Out of network: yes, program splice: yes, duration set: yes, immediate: no
    Time PTS: 0x0647FA2E3
    Duration PTS: 0x0007BA0CE (8102094), auto return: yes
    Unique program id: 0x0000 (0), avail: 0x00 (0), avails expected: 0
    CRC32: 0x7DFE6365 (OK)
```
I would proposed to add a line something like:
`preroll time: 5.810 secs`

Additional proposed changes:
- Add Splice event id as decimal equivalent in brackets after hex value
- Add Time PTS as seconds equivalent in brackets after hex value

#### Alternatives:
Work around for deriving preroll timing would likely include verbose output that includes PTS/PCR values of packets and manually doing the math to compute the delta but I have been unable to find the appropriate data to do this, partly because SCTE-35 packets are not timed with PTS.

#### Additional information:
Two other tools currently provide this information--the paid for professional [Compressed Media Analyzer from Sencore](https://www.sencore.com/product/cma-1820-compressed-media-analyzer/) and the free [DVBinspector](https://github.com/EricBerendsen/dvbinspector) the latest release now including preroll timing information at my request.

lelegard, 2021-02-03:
Hi @BartmanEH 

The architecture of TSDuck is made of a lot of small utilities which perform individual specialized tasks. Plugins and tools such as `tstables` or `tstabdump` only extract sections and display their content in various formats. The pre-roll information you mention is not part of the SCTE-35 sections. It is the result of a dynamic analysis of the stream: 1) associate video PID's and SCTE-35 PID's, 2) monitor time stamps on video PID, 3) extract splice commands and finally extrapolate the pre-roll time.

So, as such, `tstables` and the like will never report the pre-roll time, only the actual content of the various tables and sections.

It is possible, however, to develop a dedicated SCTE-35 analysis tool which does this.

> Additional proposed changes:
> - Add Splice event id as decimal equivalent in brackets after hex value
> - Add Time PTS as seconds equivalent in brackets after hex value

This is already done.

BartmanEH, 2021-02-04:
I wouldn't presume to ask you to develop a new SCTE-35 analysis tool--I can get by with DVBinspector for now.

How do I pull down the new version of tsduck using Homebrew? I did `brew update` then `brew outdated` but it doesn't show an update being available for tsduck homebrew tap.

lelegard, 2021-02-04:
Hi @BartmanEH,

Homebrew only installs official versions. Version 3.25 is not yet out. But you may simply pull the repo and recompile. Since Hombrew recompiles TSDuck on installation, it does not make much difference in time and installation prerequisites.

See [details here](https://tsduck.io/doxy/building.html). It should be simply:
~~~
git clone https://github.com/tsduck/tsduck.git
cd tsduck
make -j10
make install
~~~

And by the way, I pushed a small plugin named `splicemonitor` which may meet your needs. The magic of TSDuck is that you can implement most new features in virtually no time.

BartmanEH, 2021-02-04:
thank you for the compiling instructions--I'm a noob at compiling. It did work and I'm now using `version 3.25-2223`. I'm sure how the time in Sec works yet: `Time PTS: 0x0647FA2E3 (1,686,086,371, 18,734,293 ms)`

lelegard, 2021-02-04:
Activate the plugin (-P) and provide option --help to that plugin:
~~~
tsp -P splicemonitor --help
~~~
Since you downloaded the source code, have a look at tsduck.pdf in the doc subdirectory.

BartmanEH, 2021-02-04:
Amazing! You created a new plugin and provided the updated documentation for it as well. This post is simply to acknowledge your hard work, your responsiveness and your thoroughness. Thanks so much. I will explore the new documentation and play with the new `splicemonitor` plugin and provide feedback here shortly.
[image]

BartmanEH, 2021-03-02:
I'm comparing the preroll calculation results of the new `splicemonitor` plugin against the other two analyzers identified in my OP. `splicemonitor` plugin is reporting the prerolls of the two spliceinsert messages as:
```
* splicemonitor: splice PID 0x003C (60), event 0x01409DC6 (21011910) out, occurrence #1, time to event: 5,286 ms
* splicemonitor: splice PID 0x003C (60), event 0x01409DC6 (21011910) out, occurrence #2, time to event: 5,104 ms
```
The other two analyzers are reporting prerolls of: 5.810/5.809 and 5.272/5.271. The other two analyzers are within 1 mSec of eachother but `splicemonitor` is quite different.

Analyzing the `splicemonitor` source code, I see preroll time calculation is dependent on the computed bitrate of the input. Could this account for the discrepancy? I tried forcing bitrates using `--bitrate` but it didn't seem to make the preroll time any different.

lelegard, 2021-03-02:
To understand the difference, we must understand how the other tools compute their preroll time. I cannot tell.

Let's try to guess anyway. I assume that all tools use the event PTS as end time. So, the difference may come from the evaluation of the start time (the reception time of the splice command). If TSDuck consistently reports 520 ms and 160 ms _less_ than other tools, it means that TSDuck evaluates the start time _later_ in the future. What TSDuck uses as start time is the latest PTS value that was found in the service plus the time between this latest PTS and the splice command. To evaluate this additional short time, TSDuck counts the number of TS packets between the latest PTS and the splice command. Using the bitrate, we translate this number of packets in milliseconds.

Maybe the other tools only use the latest PTS as start time, meaning that their start time is before TSDuck start time, giving a slightly longer preroll time than TSDuck. Or maybe this is too clever, in a desperate attempt to make TSDuck more clever than others 😄 and there is simply big bug in TSDuck computation 😞. Solving this is left as an exercise to the reader, as all lazy authors commonly say...

BartmanEH, 2021-03-03:
I analyzed a different sample and the the results were
TSDuck:
```
* splicemonitor: packet 5,225,972, splice PID 0x003C (60), event 0x0132BC47 (20102215) out, occurrence #1, time to event: 5,212 ms
* splicemonitor: packet 5,281,438, splice PID 0x003C (60), event 0x0132BC47 (20102215) out, occurrence #2, time to event: 4,182 ms
```
DVBInspector:
```
splice_time (preroll time = 5.915 secs)
splice_time (preroll time = 4.669 secs)
```
The delta is 5,915-5,212=703ms and 4,669-4,182=487ms so not 'consistent' per se.

I've asked the author of DVBInspector to clarify how pre-roll time is calculated for its purposes. As for the other analyzer, it is an expensive piece of software from a rather large company from which it is likely impractical to extract such details.

I would be interested to see what the time value is between the splice command packet's event PTS time and the last PTS value in the associated service to see how this compares to see if this is the difference as you suggest. I will explore the documentation to see if theres a way to dump PTS values from the samples I'm analyzing.

EricBerendsen, 2021-03-03:
Let my try to explain what I did in DVB Inspector;

I take the pts_time in the SCTE-event, and compensate it with the pts_adjustment in same event (all examples I have seen pts_adjustment ==0, but it might be different). That is my end time. 

For start time I take the (calculated) PCR value for that service at the time of the SCTE-event arriving. Because DVB Inspector first reads the entire stream before doing any calculations, that makes life a bit easier for me; I can calculate bitrate based on entire stream ((lastPCR - firstPCR)/packetsdifference). So it is easy to calculate PCR value at moment of SCTE-event. 

So I think the main difference is I base the calculation on PCR, not PTS.

In a picture, I calculate the height of the geen arrow; (it's should match the orange arrow, as that is the actual pre-roll time)

[image]

Not sure why you would want to use PTS values (blue dots) for calculation. They would lead to smaller values for pre-roll.

Just my 2 cents.

lelegard, 2021-03-03:
Thanks @EricBerendsen for the explanation, this is very clear.

The main difference with TSDuck is that it works on potentially live streams. So, events, packets, sections are processed at the time they appear on the stream, without the full information of the entire stream. A splice information table is displayed at the time the section appears, including its preroll time. This means that we need to estimate the pre-roll time before having seen the event itself (the PTS value in a packet). So, although the PCR is the only real TS clock, we have no idea of what the PCR will be when the event occur. We only know its PTS (the value in the splice information table, after PTS adjustment as you rightly mentioned).

So, we assume here that the offset between PTS and PCR is more or less constant. This assumption is of course incorrect but a better approximation than mixing PTS and PCR in the same formula. We have the following sequence of events:

- (A) got latest PTS in the service
- (B) got a splice information table with event PTS = X
- (C) will get a packet with PTS >= X (aka "the" event)

The pre-roll time we seek is time(C) - time(B). The real value is PCR(C) - PCR(B). Unlike DVBinspector, TSDuck does not know PCR(C) since this is in the future. We only know PTS(C) from the splice information table.

If we assume that PTS(x) - PCR(x) is more or less constant (more or less), we can say that

- pre-roll time = PCR(C) - PCR(B) ~= PTS(C) - PTS(B) = PTS(C) - PTS(A) - (PTS(B) - PTS(A))
- PTS(C) is in the splice table
- PTS(A) was fetched from a packet in (A)
- PTS(B) - PTS(A) ~= PCR(B) - PCR(A) is computed from the bitrate and the packet distance between (A) and (B).

There are several approximations here which may explain the difference. But since DVBinspector has read the entire stream first and has a relatively precise evaluation of the PCR of each event, its evaluation is better.

EricBerendsen, 2021-03-03:
@lelegard I think the difference in interpretation we have is when the actual splice should be done. You say it should be done when a packet (start of frame) is received with PTS >= X (aka "the" event). 

So the black arrow in this picture;

[image]

My interpretation is that "the" event is done when that frame should be displayed (when PCR == PTS from splice information table == PTS from packet).

I don't think there is an official exact definition of pre-roll time, so there is no right or wrong answer...

lelegard, 2021-03-04:
> My interpretation is that "the" event is done when that frame should be displayed (when PCR == PTS from splice information table == PTS from packet).

"The" event is the point where some equipment has to do something. The pre-roll time is the duration during which the equipment has the time to prepare for the action to do. Thus, the definition of "the" event depends on the context and the target equipment.

In the original SCTE 35, the target was video splicing in an intermediate re-transmitter (national satellite feed with national ads => local terrestrial retransmitter replacing national ads with local ads => terrestrial broadcast => TV set). In this context, the equipment which processes the SCTE cue is the retransmitter and "the" event is the reception of a packet containing the event PTS and the PCR of the TS at this packet is different.

Now, on IPTV or OTT, SCTE 35 is a trigger to switch program (no longer video splicing in the original stream). In this context, the equipment is the STB or TV set and "the" event is the PCR / wall-clock time the value of which is the event PTS.

So, depending on the context, the relationship between the event PTS (as signaled in the splice information table) and the real wall-clock time is different.

> I don't think there is an official exact definition of pre-roll time, so there is no right or wrong answer...

Yes, different tools work differently, they have different available input data and produce similar but slightly different results. This is the same with experimental scientists: they have different instruments, none of them is perfect but the scientist knows exactly what they are measuring and combining the results of all instruments relatively to their own context, the scientist can reach a conclusion. This is why this discussion can be useful to the users of our tools so that they can understand how to interpret the various results.

BartmanEH, 2021-03-04:
I do not follow the difference defined for the two use cases: 1) ad insertion for terrestrial re-broadcast and 2) IPTV/OTT stream switching. Both could use a splice_insert() message to effect an action at splice_time() and both use cases may have dependencies on pre-roll time to facilitate the action--whether it's spooling up a local file or transcoding a stream to prepare for the splice action.

Notwithstanding any differences in interpretation or usage of pre-roll time, this is defined in the [SCTE-104 standard](https://www.scte.org/pdf-redirect/?url=https://scte-cms-resource-storage.s3.amazonaws.com/ANSI_SCTE-104-2019a-1582645426999.pdf) as having a minimum value of 4 seconds:
```
12.3. Minimum Pre-roll Time Supported by this Interface
In compliance with the requirements of SCTE 35 [1], and in keeping with the advice of SCTE 67 [13], the
minimum non-zero value of pre-roll time shall be 4000 milliseconds for a splice_request.
```
It is with this requirement that I am analyzing streams for compliance so interpretive deltas of 500+mSec can make the difference between complying or not. I need to know if the splice_insert() message is being issued 4+ seconds in advance of the defined splice_time() in the splice_insert() message.

Granted I am not an expert on PTS/PCR nor their relationship to wall-clock time and I cannot follow the complexities of the calculations at hand. I am pleased that I have two analysis tools that use the benefit of non-realtime analysis to produce consistent accurate results for pre-roll time. I was endeavouring to see if realtime analysis with tsduck was feasible so that pre-roll time compliance warnings/errors could be generated on-the-fly. It seems that this may not be possible although I hold out some hope that if a consistent use case is defined along with consistent interpretation of parameters, that consistent results could yet be possible by defining a new parameter for the splicemonitor plugin to tune is calculations accordingly.

lelegard, 2021-03-04:
The definition of "pre-roll time" is pretty vague in SCTE 104:

> pre_roll_time–An 16-bit field giving the time to the insertion point in milliseconds.

>The timestamp() indicates the time to process the splice_info_section. The pre-roll field indicates the amount of time, in milliseconds, after being processed that the action will occur. 

I interpret this as a real time ("wall-clock time") duration between the reception and the _video splicing_ point (SCTE 35 and 104 only mention _video splicing_, not program switch as in IPTV and OTT). The video splicing point is the time where the video splicer equipment starts to replace audio and video elementary stream inside the unmodified structure of the service. In the case of OTT program switch, the insertion point occurs at the presentation time, not at the packet processing time. So, you may add up to 200 ms (or more?) between the two. If your objective is to monitor defects from an expected behavior, then you need to start by defining what you expect. And it seems that precisely defining "pre-roll time" is far from obvious.

More pragmatically, in the absence of precise case-by-case definition, you may simply integrate some "expected inaccuracy", probably around 500 ms to be on the safe side.

> I was endeavouring to see if realtime analysis with tsduck was feasible so that pre-roll time compliance warnings/errors could be generated on-the-fly.

It is possible to compute the exact time between the splice command and the first packet of "the" event when the event occurs, not when the splice command is detected. The display of the splice command and the display of the pre-roll time will then be distinct, the time between the two being the pre-roll time. This can be done.

BartmanEH, 2021-03-04:
>If your objective is to monitor defects from an expected behavior, then you need to start by defining what you expect. 

I am trying to relate what I expect in the clearest terms I know how as a systems engineer (but not a PTS/PCR expert) which is why I stated that "I need to know if the splice_insert() message is being issued 4+ seconds in advance of the defined splice_time() in the splice_insert() message." The splice_insert messages I'm analyzing comprise a splice_time with a pts_time parameter. This is the 'time' at which we expect the splice_insert will occur so the pre-roll time I'm looking for is this future pts_time (defined in the splice_insert message) less/minus the current time of the splice_insert message (derived in relation to its associated service).

lelegard, 2021-03-05:
To make sure of the interpretation, I added the computation of the actual duration between the reception of the first splice command for a given event and the corresponding event (the packet with a PTS for which PTS >= event PTS as announced in the splice command). This duration is computed from packet count and bitrate.

Sample test using a TS fragment from a real Russian stream containing SCTE 35 cue from the TSDuck stream repository:
~~~
$ tsp -I file tsduck-test/input/test-084.ts -P splicemonitor -O drop 
* splicemonitor: splice PID 0x041F (1055), event 0x00000012 (18) out, occurrence #1, time to event: 9,294 ms
* splicemonitor: splice PID 0x041F (1055), event 0x00000012 (18) out, occurrence #2, time to event: 7,358 ms
* splicemonitor: splice PID 0x041F (1055), event 0x00000012 (18) out, occurrence #3, time to event: 5,415 ms
* splicemonitor: splice PID 0x041F (1055), event 0x00000012 (18) out, occurred, actual pre-roll time: 9,556 ms
* splicemonitor: splice PID 0x041F (1055), event 0x00000012 (18) in, occurrence #1, time to event: 9,530 ms
* splicemonitor: splice PID 0x041F (1055), event 0x00000012 (18) in, occurrence #2, time to event: 7,624 ms
* splicemonitor: splice PID 0x041F (1055), event 0x00000012 (18) in, occurrence #3, time to event: 6,372 ms
* splicemonitor: splice PID 0x041F (1055), event 0x00000012 (18) in, occurred, actual pre-roll time: 9,701 ms
~~~

The actual time, measured when the event actually occurs, is slightly larger than the anticipated time when receiving the first splice command.

You should try on your streams and compare with DVBinspector and the other tools. Based on @EricBerendsen's explanations, I assume that the results will be similar to DVBinspector.

I see two possible explanations, depending on the stream and context.
- The anticipated time computation can be based on PTS only and the PTS-PCR difference is not constant (we already know that) and the variation of this PTS-PCR difference is responsible for the inaccuracy.
- The PTS in the actual event is not exactly the announced PTS but slightly larger.

I would personnally bet that the first reason is more significant.

My conclusion is that a valid evaluation of a pre-roll time is not possible _before_ the event actually occurs. It is possible only after the event occurs, either using a complete off-line file as DVBinspector does or at the time of the event as TSDuck. A monitoring system can detect incorrect pre-roll time after the error but cannot anticipate it.

Just as a test, I added an option `--no-adjustment` to remove the previous-PTS to splice-command adjustment based on packet count. It increases the anticipated time but does not change much. I would recommand to not use it.

~~~
$ tsp -I file tsduck-test/input/test-084.ts -P splicemonitor --no-adjustment -O drop 
* splicemonitor: splice PID 0x041F (1055), event 0x00000012 (18) out, occurrence #1, time to event: 9,360 ms
* splicemonitor: splice PID 0x041F (1055), event 0x00000012 (18) out, occurrence #2, time to event: 7,440 ms
* splicemonitor: splice PID 0x041F (1055), event 0x00000012 (18) out, occurrence #3, time to event: 5,440 ms
* splicemonitor: splice PID 0x041F (1055), event 0x00000012 (18) out, occurred, actual pre-roll time: 9,556 ms
* splicemonitor: splice PID 0x041F (1055), event 0x00000012 (18) in, occurrence #1, time to event: 9,560 ms
* splicemonitor: splice PID 0x041F (1055), event 0x00000012 (18) in, occurrence #2, time to event: 7,640 ms
* splicemonitor: splice PID 0x041F (1055), event 0x00000012 (18) in, occurrence #3, time to event: 6,420 ms
* splicemonitor: splice PID 0x041F (1055), event 0x00000012 (18) in, occurred, actual pre-roll time: 9,701 ms
~~~

EricBerendsen, 2021-03-05:
At the risk of increasing the confusion, let me try to clarify something;

(Quotes re-ordered for readability)

> My conclusion is that a valid evaluation of a pre-roll time is not possible _before_ the event actually occurs. It is possible only after the event occurs, either using a complete off-line file as [...]

DVBInspector does not read the entire stream until **the** event (whatever that may mean) occurs. It takes the pts_time in splice_info_section, and calculates the difference with the PCR at the moment the section is received. Only to calculate the PCR it uses PCRs from beginning and end of whole file (But that has another problem as we will soon see...) Calculating the exact PCR could be done using only info from the past, when you already know the bitrate and have had some previous PCRs.

> Sample test using a TS fragment from a real Russian stream containing SCTE 35 cue from the TSDuck stream repository:
>
> ```
> $ tsp -I file tsduck-test/input/test-084.ts -P splicemonitor -O drop
> * [...]
 
Of course I tried, and this is what I got:
splice_time (preroll time = 6,635 secs)
splice_time (preroll time = 0,629 secs)
splice_time (preroll time = **-5,376** secs)

splice_time (preroll time = 36,108 secs)
splice_time (preroll time = 30,100 secs)
splice_time (preroll time = 24,092 secs) 

Starting to question my own sanity, negative pre-roll..

Looks like the sample file is concatenated from two files, as there is a discontinuity in the PCR;

[image]

So this is a problem when calculating PCR from entire file.....

Split file into two, try again;

splice_time (preroll time = 10,459 secs)
splice_time (preroll time = 8,458 secs)
splice_time (preroll time = 6,458 secs)

splice_time (preroll time = 10,460 secs)
splice_time (preroll time = 8,459 secs)
splice_time (preroll time = 6,458 secs)

I agree with @lelegard 's interpretation, looking for the first PTS == time in splice_info_section probably makes more sense, as this is when something has to be done to change the stream and replace video/audio in the TS. That will result in lower values then in my interpretation, that waits until the PCR has been reached. 

What is the user needs is depending on the application.

BartmanEH, 2021-03-05:
I'm eager to test the new results but I suppose I need to compile from source to try this? I'm not very adept at compiling but I note that there is no release for this new version and, hence, homebrew doesn't have an upgrade available for macOS. For me, cloning github repos, compiling and then figuring out how to run the new version without corrupting the homebrew release installation is a nightmare of terminal commands and folder logistics. I do not mean to belittle the considerable effort you have put into your build guide documentation, which I reviewed, however, it is not for the faint of heart and a software developer I am not. I shall endeavour to do so if we feel that there may be several iterations yet to test before we agree on the final desired functionality.

lelegard, 2021-03-05:
> Looks like the sample file is concatenated from two files, as there is a discontinuity in the PCR;

@EricBerendsen, yes your are right. The original real-life stream was a multi giga bytes file in the TSDuck stream repository. This storage is on an OVH cloud hosting service for tsduck.io.

The automated test suite for TSDuck is archived in a git repo on GitHub and I limit the input test samples to 20 MB max. To get the splice out and splice in of a few minutes ads sequence in the same file for the sake of non-regression testing, I had to cut and join two parts with the complete sequences (all splice commands and the event itself).

Anyway, a PCR discontinuity may still happen in any real stream, at least every 26 hours when the 42-bit PCR value wraps up (unless you catch this specific case individually of course). Other discontinuity can always happen. In TSDuck, since there is not always a "beginning" and an "end", I have a class which permanently computes the TS bitrate based on each PID with PCR. I mean that each PCR PID produces a specific TS bitrate (they should be identical on a well-formed TS) and I keep the average of them. Each time a packet discontinuity is encountered, the computation is restarted from scratch for all PID's since the packet count becomes invalid. The resulting TS bitrate is never completely but reasonably accurate and always robust to all kinds of input discrepancy.

lelegard, 2021-03-05:
> I'm eager to test the new results but I suppose I need to compile from source to try this? I'm not very adept at compiling but I note that there is no release for this new version and, hence, homebrew doesn't have an upgrade available for macOS.

@BartmanEH, yes you need to rebuild it. I am also a Mac user. My iMac is my main development platform, all others are just VM's running the Mac... So, I rebuild all the time on the Mac.

The main build instructions are in the TSDuck [programmers guide](https://tsduck.io/doxy/building.html). Once you have cloned (or pulled again) the repo from GitHub, just do this:
~~~
make distclean  # only to make sure your restart from a clean state after a major pull
make -j10       # -j10 or more to speed up the parallel build
make install
~~~
That's all.

Yes, you are right, this overwrites the Homebrew installation or TSDuck. This is not really a problem, you can use TSDuck normally. At the next official upgrade of TSDuck, HomeBrew will first install it normally in `/usr/local/Cellar/tsduck/3.26-xxxx` and will then complain that the previous installation was somewhat overwritten. But it also gives you the solution, it says that you should run some `brew link` command to fix it. I don't remember the exact syntax. Just copy/paste the recommended command and Homebrew will complete the installation of the new version.

To give you more incentive to install the new version, I also added in the meantime an `--alarm-command` option which executes some user-defined command (typically a shell script of yours) when some criteria are not met, typically min/max pre-roll time and min/max number of repetitions of the splice command. This is the same principle as the `bitrate_monitor` plugin. If your goal is is to do monitoring, you may be interested in this.

BartmanEH, 2021-03-05:
for the record, here's what I had to do to build from scratch:
```
rm -rf tsduck                                       # I must have had an old cloned git?
git clone https://github.com/tsduck/tsduck.git      # 'git clone tsduck' didn't work because tsduck repo location unknown?
cd tsduck                                           # obvious to the initiated but otherwise the following steps won't work
build/install-prerequisites.sh
make distclean
make -j10
make install
```
After which I confirmed successful completion with:
```
tsp --version
tsp: TSDuck - The MPEG Transport Stream Toolkit - version 3.26-2266
```
When I installed from homebrew last time I had to run the following command as prompted to get the symlinks working:
`brew link --overwrite tsduck`

BartmanEH, 2021-03-05:
very interesting. the predicted pre-roll time is the same and now there is added information about actual pre-roll time when the event occurred (or is suppose to occur).
```
* splicemonitor: packet 5,225,972, splice PID 0x003C (60), event 0x0132BC47 (20102215) out, occurrence #1, time to event: 5,212 ms
* splicemonitor: packet 5,281,438, splice PID 0x003C (60), event 0x0132BC47 (20102215) out, occurrence #2, time to event: 4,182 ms
* splicemonitor: packet 5,441,896, splice PID 0x003C (60), event 0x0132BC47 (20102215) out, occurred, actual pre-roll time: 3,936 ms
```
However the predicted pre-roll time still varies greatly from DVB Inspector and the gold standard (because of it's hefty price) Compressed Media Analyzer.

EricBerendsen, 2021-03-05:
Maybe going off-topic here, but interesting to compare approaches here;

> Anyway, a PCR discontinuity may still happen in any real stream, at least every 26 hours when the 42-bit PCR value wraps up (unless you catch this specific case individually of course). 

In DVB Inspector PCRs are used for sample rate calculation as long as they are higher than the previous valid PCR. If PCR wraps around new values will be lower and not used for sample rate. For this PID the sample rate will be based on only the initial part, up to the lower values. So wrap around is not a problem. If PCR values suddenly jump to a (much) higher value it will just process them. What should be a jump that is considered too big?

There is no reset after wrong continuity_counter values on TSPackets, or discontinuity_indicators in adaptation field. 

The bit rate for the whole file is the average of the bit rates for all PCRs. And when a file has no PCRs at all, It tries to use TDTs to give an estimate of the bit rate (way less accurate).

DVB Inspector is used mostly for short (at most a few minutes) files, so the chances of discontinuities are not very large. But your test file made me realise there is room for improvement. 

Merely wanted to use this example to indicate never to blindly rely on the results of a program. There can be (and will be) many reasons why a program can (and will) give wrong or strange results. Always keep checking, and when possible, use another program to verify results.

lelegard, 2021-03-06:
> Merely wanted to use this example to indicate never to blindly rely on the results of a program. There can be (and will be) many reasons why a program can (and will) give wrong or strange results. Always keep checking, and when possible, use another program to verify results.

Definitely. Always understand how tools work, what they actually measure, how to interpret the result and cross-check with others. We learn this at high school in science class.

futzu, 2021-05-01:
I use[ threefive](https://github.com/futzu/threefive) to do this. 
threefive returns the closest PTS  to the SCTE-35  along with the SCTE-35 data.  

```
{
    "info_section": {
        "table_id": "0xfc",
        "section_syntax_indicator": false,
        "private": false,
        "sap_type": "0x3",
        "sap_details": "No Sap Type",
        "section_length": 47,
        "protocol_version": 0,
        "encrypted_packet": false,
        "encryption_algorithm": 0,
        "pts_adjustment": 0.0,
        "cw_index": "0xff",
        "tier": "0xfff",
        "splice_command_length": 20,
        "splice_command_type": 5,
        "descriptor_loop_length": 10
    },
    "command": {
        "command_length": 20,
        "command_type": 5,
        "name": "Splice Insert",
        "time_specified_flag": true,
        "pts_time": 21951.133267,      # <-- PTS for the splice insert  
        "break_auto_return": false,
        "break_duration": 75.0,
        "splice_event_id": 18,
        "splice_event_cancel_indicator": false,
        "out_of_network_indicator": true,
        "program_splice_flag": true,
        "duration_flag": true,
        "splice_immediate_flag": false,
        "unique_program_id": 1,
        "avail_num": 18,
        "avail_expected": 255
    },
    "descriptors": [
        {
            "tag": 0,
            "descriptor_length": 8,
            "identifier": "CUEI",
            "name": "Avail Descriptor",
            "provider_avail_id": 18
        }
    ],
    "crc": "0xeaa8473f",
    "pid": "0x41f",
    "program": 1050,
    "pts": 21940.656778     # < -- closest PTS to the SCTE-35 packet in the same program
}
```

BartmanEH, 2021-05-03:
> I use[ threefive](https://github.com/futzu/threefive) to do this.
> threefive returns the closest PTS to the SCTE-35 along with the SCTE-35 data.
> ```
Right but threefive doesn't compute the pre-roll for you, which is the point of this thread. When doing frequent SCTE-35 analysis, one really needs a tool to do the work for you!

futzu, 2021-05-06:
Here,  I just added it. 
```
{
    "info_section": {
        "table_id": "0xfc",
        "section_syntax_indicator": false,
        "private": false,
        "sap_type": "0x3",
        "sap_details": "No Sap Type",
        "section_length": 47,
        "protocol_version": 0,
        "encrypted_packet": false,
        "encryption_algorithm": 0,
        "pts_adjustment": 0.0,
        "cw_index": "0xff",
        "tier": "0xfff",
        "splice_command_length": 20,
        "splice_command_type": 5,
        "descriptor_loop_length": 10
    },
    "command": {
        "command_length": 20,
        "command_type": 5,
        "name": "Splice Insert",
        "time_specified_flag": true,
        "pts_time": 21951.133267,
        "break_auto_return": false,
        "break_duration": 75.0,
        "splice_event_id": 18,
        "splice_event_cancel_indicator": false,
        "out_of_network_indicator": true,
        "program_splice_flag": true,
        "duration_flag": true,
        "splice_immediate_flag": false,
        "unique_program_id": 1,
        "avail_num": 18,
        "avail_expected": 255
    },
    "descriptors": [
        {
            "tag": 0,
            "descriptor_length": 8,
            "identifier": "CUEI",
            "name": "Avail Descriptor",
            "provider_avail_id": 18
        }
    ],
    "crc": "0xeaa8473f",
    "pid": "0x41f",
    "program": 1050,
    "pts": 21940.656778,
    "preroll": 10.476489   # <--- This what you want ?
}
```

lelegard, 2022-09-23:
Automatically closed after 505 days without update and "close pending" label set.

closed 2022-09-23 by lelegard

anshultiwari7, 2023-06-22:
I found this discussion to be very interesting as I was looking for a solution to my problem, which involves monitoring the splicing information. However, the command provided does not yield any output, even after trying different versions of tsduck and transport stream files. Could someone please assist me?

Command used: `tsp -I file testfile.ts -P splicemonitor -O drop`

lelegard, 2023-06-23:
Hi @anshultiwari7 

First, are the SCTE-35 PID's correctly referenced as such in the PMT? The `splicemonitor` plugin scans the various PMT's and locates SCTE-35 PID's. If they are not correctly referenced, it won't find any and you need to specify the PID to monitor (RTFM or use `tsp -P splicemonitor --help` for more details).

Use the verbose option (`tsp -v ...`) to see which splice PID's are monitored:
~~~
* splicemonitor: starting monitoring splice PID 0x040B (1035)
* splicemonitor: starting monitoring splice PID 0x0451 (1105)
* splicemonitor: starting monitoring splice PID 0x03F7 (1015)
* splicemonitor: starting monitoring splice PID 0x041F (1055)
* splicemonitor: starting monitoring splice PID 0x0415 (1045)
~~~

If nothing is displayed, there is no splice PID or they are not correctly referenced in the PMT's.

Second, if splice PID's are found but nothing else is displayed, they there is probably no splice insert event in the file (these are rare events and it is not unusual to find none in short TS files).

You may also try `-P splicemonitor -a` to monitor all splice events, not only insert commands.

Otherwise, 1) carefully read the documentation and 2) please open a new issue, nobody reads old closed issues.