tsduck/tsduck #724: [FEAT] TS  Input switcher: C++ API allow switch control

gizahNL, 2021-02-08:
#### Is your feature request related to a problem?
Currently it is impossible to directly control the input switcher switching via C++ API.
#### Proposed solution:
2 possible solutions could come to mind:
-expose an setInput method on the ts::InputSwitcher plugin.
-allow loading of switching control plugins other than the commandlistener

lelegard, 2021-02-08:
What do you mean by "directly control the input switcher switching" ?

What is the use case exactly ? Do you want to include the `tsswitch` feature inside your application or do you want to send a switch command from your application to an external `tsswitch` command?

gizahNL, 2021-02-08:
I want to control tsswitch directly from my application, included inside it, for now I've hacked something myself that should work, though an upstream supported solution would be nicer. 

I know I could write code that would just sendmsg to the port, but that seems less elegant than simply calling a method on the object, or have a plugin interface to plugin to the tsswitch object.

lelegard, 2021-02-09:
I have added a few methods in the class `ts::InputSwitcher` to control the switch. I also added a simple constructor and start/stop methods. Previously, the code was a direct extract from `tsswitch` and the only constructor handled all the session. You need to pull the repo and rebuild. Please test this and report your feedback here.

lars18th, 2021-02-09:
Hi @lelegard ,

I can see that this improvement will be useful for the programming use of the tool `tsswitch`. So then it would also be good to have the extension for other languages, like Python and Java. What you think? For sure, as commented before, is not useful to expose all of the TSDuck functionalities with bindings. However, some _basic_ functionality will be interesting for fast prototyping. Don't you think so?

gizahNL, 2021-02-09:
> I have added a few methods in the class `ts::InputSwitcher` to control the switch. I also added a simple constructor and start/stop methods. Previously, the code was a direct extract from `tsswitch` [...]

Awesome, very quick indeed! Thanks, I'll let you know as soon as possible.

Would you think a method for querying the current selected output would fit as well?

gizahNL, 2021-02-09:
Not sure if this is an regression, but tsswitch seems to have an abnormally high cpu usage. 

I see about 135% cpu used when running the following:
tsswitch --debug=9 --infinite -I file --infinite ~/Downloads/atsc-cea-708-dtvcc-and-cea-608.ts -I ip 239.168.88.130:14400 -O ip --packet-burst 7 --enforce-burst 127.0.0.1:5000

When I use -H in top to show per thread statistics I see 1 thread locked at ~100%, so I guess something is (accidentally) busy waiting

lelegard, 2021-02-09:
Sorry about that, I will have a look. The problem with `tsswitch` is that it is difficult to automate the tests other than trivial ones (see https://github.com/tsduck/tsduck-test). Tests need some infrastructure or manual operations. This is why I am always a bit reluctant to alter `tsswitch`.

lelegard, 2021-02-09:
Did that command work correctly before upgrading? It seems normal that it runs at full speed. The first (and initial) input is a file and the output is UDP/IP. Nothing is regulating the system. The input file is read at disk speed (and probably as cache memory speed after the first iteration completes) and the UDP datagrams are sent as fast as input data arrive.

gizahNL, 2021-02-09:
> Sorry about that, I will have a look. The problem with `tsswitch` is that it is difficult to automate the tests other than trivial ones (see https://github.com/tsduck/tsduck-test). Tests need some infrastructure or manual operations. This is why I am always a bit reluctant to alter `tsswitch`.

Yes completely understand that ;)

> Did that command work correctly before upgrading? It seems normal that it runs at full speed. The first (and initial) input is a file and the output is UDP/IP. Nothing is regulating the system. The [...]

Before it's the same. Seems logical since your commit did not seem to touch anything related to a loop. 
I've narrowed it down to the file input: if I switch the order of inputs the load seems normal. 

I had assumed the file input would run at the bitrate as calculated from it's PCR automatically, is there a way I can tell it to do this, thereby limiting it's CPU usage?

lelegard, 2021-02-09:
>  if I switch the order of inputs the load seems normal.

This is expected as well. In that case, the initial input is the IP source where packets arrive at the speed they were sent by the sender system. See the [TSDuck user's guide](https://tsduck.io/download/docs/tsduck.pdf), chapter 3, reference section for TSP, section "Rendering speed and transmission speed". I would recommend to read the whole reference section for TSP at least once to understand the concepts.

> I had assumed the file input would run at the bitrate as calculated from it's PCR automatically,

There is no reason to assume this. There are many use cases for TSDuck. Playing with real-time transmission, as you do, is one. Offline file processing is another. If you want to do batch processing of files, there is no reason to slow down the process at rendering speed. Just do it as fast as you can, at disk speed. 

> is there a way I can tell it to do this, thereby limiting it's CPU usage?

Yes, have a look at the `regulate` plugin and browse through the examples in the user's guide, especially section 5.2.4.

Since you are using `tsswitch` and not `tsp`, you must use a regulated `tsp` command as one of the `tsswitch` inputs, something like `tsswitch ... -I fork "tsp -I file xx.ts -P regulate" -I ...` You may want to fine-tune the options, please RTFM.

In fact, the CPU usage is not the actual problem. The question is "what do you want to functionally achieve?" If TSDuck is eating up all CPU's during a few seconds or minutes to do some TS file processing, encryption or whatever, this is fine because this is fast. In your case, the high CPU is just a symptom of something else: the process is running too fast for what you intend to do.

lelegard, 2021-02-09:
> Would you think a method for querying the current selected output would fit as well?

Done

gizahNL, 2021-02-09:
> This is expected as well. In that case, the initial input is the IP source where packets arrive at the speed they were sent by the sender system. See the [TSDuck user's guide](https://tsduck.io/download/docs/tsduck.pdf), chapter 3, reference section for TSP, section "Rendering speed and transmission speed". I would recommend to read the whole reference section for TSP at least once to understand the concepts.
>  <snip>
> Yes, have a look at the `regulate` plugin and browse through the examples in the user's guide, especially section 5.2.4.

Will do! 

> > Would you think a method for querying the current selected output would fit as well?
> 
> Done

Thanks!

lelegard, 2021-02-11:
> I can see that this improvement will be useful for the programming use of the tool tsswitch. So then it would also be good to have the extension for other languages, like Python and Java. 

Done. See a [high-level description of Java & Python bindings](https://tsduck.io/doxy/jpbindings.html)

lars18th, 2021-02-11:
Hi @lelegard ,

> > I can see that this improvement will be useful for the programming use of the tool tsswitch. So then it would also be good to have the extension for other languages, like Python and Java.
> 
> Done. See a [high-level description of Java & Python bindings](https://tsduck.io/doxy/jpbindings.html)

Your work far exceeds my expectations, thank you very much! :+1:

closed 2022-09-23 by lelegard