tsduck/tsduck #725: [Q] Possible to delay scrambling?

garethfenn, 2021-02-10:
Hi there,

Is it possible to delay the scrambler plugin from starting immediately at the beginning of the stream? In other words, generate a short period (~10 seconds) of clear content allowing ECM's to be captured, ready the first key period?

Thanks for writing a great tool!

lelegard, 2021-02-10:
Hi @garethfenn 

Currently, the scrambler plugin works in "paranoid" mode exclusively: it does not let any clear packet pass. As long as the PID's to scramble are not yet known (before the first PMT of the service is received), all packets are nullified. The receiver cannot see any packet until it finishes to process the first ECM.

To do what you want, the plugin should include a new "friendly" mode where packets are passed in the clear until it is reasonable to switch to encrypted mode. Not currently implemented.

The adequate delay is normally the _transition_delay_start_ which is sent by the ECMG. It is usually much shorter that 10 seconds. 2 or 3 seconds are more common. Considering that there are 5 or 10 ECM's per second and the processing time of an ECM is less than one second (even on slow smartcards), 2 seconds _transition_delay_start_ is more than enough is most cases, even if you miss one or two ECM's.

garethfenn, 2021-02-10:
Hi @lelegard,

Thanks for your answer.

> The adequate delay...
Agreed. For live use cases this isn't required. I believe the problem in local file based playback, is that the initial content can be consumed very quickly whilst filling playback buffers... before we've had time to capture an ECM and process it. Content plays back fine from that point on wards with the default TSDuck parameters.

There are a few workarounds at the player level we could employ (or stitch some content together perhaps).... until a "friendly" mode is supported by the scrambler plugin.

closed 2021-02-10 by garethfenn

lelegard, 2021-02-10:
Offline file processing is different indeed. A few years ago, I did exactly this using a shell script for a VoD PoC (in TS mode). The idea was to add null packets at the beginning of the file, enough for the initial ECM acquisition. During the initial period, PAT and PMT are cycled. Quite naturally, ECM's will also be inserted there. When the actual content starts, the player had the time to analyze the ECM's. But the idea was still a paranoid mode : make sure that the receiver is ready to descramble the first content packet but do not let any clear content.

Because the file could be quite large, the file was processed only once. There was a short initial analysis phase of the beginning of the file (get one PAT, one PMT, compute bitrate) and then, restarting at the beginning of the file, everything was done in one pass.

lelegard, 2021-02-11:
Because all previous work should be turned into investment, I added the above mentioned script in the "sample" area of the TSDuck source: https://github.com/tsduck/tsduck/tree/master/sample/packager

You may start from that if your need is similar.

garethfenn, 2021-02-11:
Thanks for adding these example scripts! They worked OK and generated output content that plays backs OK. Unfortunately even with ~10 seconds of delay (first video packet occurs at ~10,000 packet), the same playback behavior is observed.

I believe the issue is that without backpressure from the realtime content presentation, the initial padding is consumed as fast as possible. A small (200ms) delay after first injection, combined with these scripts would probably do the trick.

Thanks again.