tsduck/tsduck #726: [Q]  Segmentation type id in SCTE35 message

Anas92230, 2021-02-17:
Hello,

I use this command for extracting all SCTE35 message : tsp -P tables --tid 0xFC --packet-index -O drop -I file stream.ts
I found all messages :-)
But when I used json-output (tsp -P tables --tid 0xFC --packet-index --json-output message.json -O drop -I file stream.ts) i don't find Provider Placement Opportunity Start. for my information in first mode we have id in hexa but with json output are on decimal.
Thank you for your help
[json_output.txt](https://github.com/tsduck/tsduck/files/5997954/json_output.txt)
[direct_output.txt](https://github.com/tsduck/tsduck/files/5997955/direct_output.txt)

lelegard, 2021-02-17:
Hi @Anas92230 

Thanks for reporting this. The JSON output is just automated translation from XML.

Apparently, the third segmentation descriptor is not correctly deserialized, not converted to C++ object and consequently interpreted as a "generic descriptor" in XML and then in JSON. 

Could you please extract the binary table (option `--binary`) and post it here?

Concerning the difference Hexa vs decimal, the Hexa notation is simply forbidden in JSON. All JSON numbers must be decimal.

Anas92230, 2021-02-17:
Hi @lelegard 
You're welcome Sir :-)
OK it's clear and thanks for remark on JSON.
You can find in attached file the binary export file
Additional request : It's possible to remove splice null? many streams have it about alarming bitrate on pid :-(
[scte35.txt](https://github.com/tsduck/tsduck/files/5998945/scte35.txt)

lelegard, 2021-02-18:
Hi @Anas92230,

Your binary table is incorrect, the _segmentation_descriptor_ which is not interpreted as XML or JSON is truncated: the last mandatory 2 bytes are missing. The text output is essentially a debug tool for human reading. It decodes whatever it can decode. Missing parts, if any, are not reported (admittedly a limitation of the tool). On the other hand, the XML or JSON conversion is a technical tool. It can convert valid sections or descriptors only. If a descriptor is invalid or unknown, it is simply converted as a _generic_descriptor_ with the binary content in hexadecimal.

In your case, the descriptor body, as seen in the JSON output, is:
~~~
43 55 45 49 00 00 04 81 7F FF 00 01 FD EE 30 00 00 34 01 01
~~~

Now, let's manually analyze these bytes using the SCTE 35 specification, section 10.3.3. Segmentation_descriptor():
~~~
identifier: 43554549
event id: 00000481
cancel: 0
reserved: 7F
program_segmentation_flag: 1
segmentation_duration_flag: 1
delivery_not_restricted_flag: 1
reserved: 1F
segmentation_duration: 0001FDEE30
segmentation_upid_type: 00
segmentation_upid_length: 00
segmentation_type_id: 34
segment_num: 01
segments_expected: 01
--- end of descriptor --
~~~
Missing:
~~~
if (segmentation_type_id == ‘0X34’ || segmentation_type_id == ‘0X36’) {
  sub_segment_num : 8 bits
  sub_segments_expected : 8 bits
}
~~~

As a conclusion, the SCTE 35 generator you are using is incorrect and you should report the bug to the vendor.

Anas92230, 2021-06-24:
thanks for all

closed 2021-06-24 by Anas92230