tsduck/tsduck #727: [FEAT] EIT p/f generator plugin

lars18th, 2021-02-18:
#### Is your feature request related to a problem?
When generating homemade EIT tables using the TSDuck tools it's easy to create the schedule events. At time, we can generate the content using the XML TSDuck format manually (aka from custom script) or using some third party tool (opensource). However, this is true only for the schedule tables. But for the Present/Following tables this is not the case. Then, the suggestion is to provide a new plugin that generates these tables.

#### Proposed solution:
The idea is quite simple: Based on the content of the schedule tables, the new plugin `eitgen` can use the readed content of the schedule tables (from the pid 18) and the current time (from the TOT/TDT tables) to generate the P/F tables.

#### Alternatives:
Using the current implementation of the TSDuck, and without other external tool, this is very hard to achieve it. However, you can do it with this complex solution: Using the `tables` plugin you can export the EIT information to XML. Then using a script you can calculate the current and next event and extract them from the XML. After that you can patch the XML to generate the corresponding P/F table. And finally you can compile the binary representation and inject inside the stream. This will work, but is very inefficient and prone to errors using scripts.

#### Additional information:
This new `eitgen` plugin can support some different parameters. For exemple:
- `--service srv1[,srv2,... | all]`: This indicates for which (actual) services the P/F tables are generated. The flag "all" represents all services in the current TS.
- `--service-other srv1:tsid[,srv2:tsid,... | all]`: This indicates for which other services the P/F tables are generated. The format is "ServiceID:transportStreamID". And the flag "all" here represents all services (from the schedule tables) known in other TS.
- `--auto-empty`: If no schedule event is found, then generate default events (based for example on the service name, and filling the gap until the next known event).
- `--from filename[.bin | .xml]`: Read the schedule event data from an external file (BIN or XML), instead of use the data in the pid 18 (that perhaps doesn't exist).
- `--time-offset value]: The minutes of the local time to offset in reference to the event time.
- `--time [tdt | system | date]`: Use as the current time the time inside the TS, the system clock or the "date" value as the starting time.

I feel that with the current implementation of the TSDuck library it will be easy to do a loop inside the plugin to review the current EIT data and extract this info.

What you think @lelegard ? It's this plugin feasible?

lelegard, 2021-02-18:
> It's this plugin feasible?

It is, indeed. It's just work and time. Any volunteer to develop it?

lelegard, 2021-06-18:
Please try the new plugin `eitinject`. Be sure to read its reference section in the user's guide first.