tsduck/tsduck #728: [Q] Auto generate patch-xml files

lars18th, 2021-02-18:
Hi @lelegard ,

I started to use the "patch-xml" option with some plugins and it works very well. However, it's quite difficult to construct such files without errors. Luckily the tool `tsxml` helps a lot to check/debug these files. But, I'm thinking if it's possible to enhance the tool with a simple "comparison" mode. The concept is to compare two XML files and generate the patch version. Then it will be more easy to create the first version (or a draft) of the patch file, and after complete it with more advanced options. For sure, it will be not possible to generate all combinations. But a simple output with small changes will be useful.

What you think? This has sense? It's complext to implement it?
In any case, thanks for listening to us about these crazy ideas.

lelegard, 2021-02-18:
> This has sense? 

Not really, unfortunately. It is technically feasible to build a diff. But a diff is not a patch. An XML patch file contains matching criteria and then updates to apply. Having a diff gives no information on which matching criteria were used.

lars18th, 2021-02-18:
OK. So don't worry about this stupid idea. :smile:

closed 2021-02-18 by lars18th

lenosj, 2021-02-18:
What about an XSLT implementation?
It could have general rules that would build upon and transform the input.
Sorry this is late, but I've done on the fly modification of XML where the
rules are converted to XSLT which is applied to the input.
We might pass the existing tables as XML to the XSLT and have it create the
updated output tables.

lars18th, 2021-02-19:
Hi @lenosj ,

I'm not sure about the advantage of a complex solution. My request was to create a first _draft_ of the patch file. Just that, to after rewrite it manually. It's true that the current patching model of the tables using XML is quite difficult to write. But, this is more or less like programming. I suggest to @lelegard to continue with the KISS model.

Regards.

lelegard, 2021-02-19:
Unlike Java, C++ has unfortunately no XML standard library. C++ developers rely on various heterogeneous XML libraries which bring their portability and integration issues. I refuse to rely on this mess. I initially used TinyXML in TSDuck but it had compilation issues that the maintainers did not really want to address. Additionally, I had to write glue code on top of it that turned almost as big as the tiny library. So, I just dropped TinyXML and implemented basic XML parsing under the existing glue code.

But I do not want to dive into implementations of XML-Schema and XSLT which are powerful but overly complicated. So I use the light-weight builtin "XML Model" and "XML Patch" instead. 

I used XML-Schema and XSLT quite extensively when I worked in Java about a decade ago. They are quite powerful but they require an extremely stable library that C++ does not have. 

And by the way, based on my previous experience, XSLT is much harder to use than the light-weight XML Patch in TSDuck which is in fact really simple: define what you want, define your selection criteria, define what to change. Former Prolog programmers, if there are any left around, will find it quite natural.