tsduck/tsduck #729: TSDuck users references needed

lelegard, 2021-02-23:
Dear TSDuck users,

We need your support to promote the TSDuck project.

Looking through the various issues and questions here, it seems that TSDuck is somewhat used in many companies for various purposes. But there is no real feedback on who is using it for what. In short, TSDuck lacks formal references.

Why does a free open source project need references?

- Any open source project needs users in order to find the motivation to continue.
- A project is better known and easily adopted when other companies already use it and trust it.
- Formal testimony is a good way to show that a product is used.

TSDuck users, we would appreciate your public feedback and testimony on this project and permission to eventually publish it on the [TSDuck site](https://tsduck.io). Recommended information include:

- What kind of company are you: media integrator, TV operator, service provider, equipment manufacturer, other?
- In which environment do you use TSDuck: development, debug, integration, test bed, production.
- How often: from time to time (e.g. debug), frequent (e.g. integration), continuous (e.g. CI test bed, production).
- In which networking context: IP (multicast IP-TV?, HLS?), broadcast (DVB?, ISDB?, ATSC?), media asset processing, other?
- Why do you find it useful? What are the strength of the product? What should be improved?

A valid feedback must include the company name and/or the user name. Please indicate if you allow your company logo to be published with your testimony (yeah, it sounds a bit marketing-ish, but even true engineers need some graphics...)

If possible, a more personalized testimony will make the reference more lively, something like this:

> _"At Foo Incorporated, we use TSDuck in our test environment / production system / integration lab. Some more details here... and we have been happy / angry about it."_
> John Doe, Chief Fantasy Officer at Foo Incorporated

Please provide your feedback here, as a response to this thread. By posting your response here, you implicitly allow the provided information to be published on the [TSDuck site](https://tsduck.io).

If you do not have a GitHub account, you may also [send your reference in an e-mail](mailto:contact@tsduck.io) but you may consider to create one (a GitHub registration is free, it does not spam you and nowadays, if you work in some sort of software-related job, it is much more useful than a Facebook account :-).

And, by the way, if you find TSDuck useful, do not forget to "star" it on GitHub.

Thanks a lot for helping promoting TSDuck.

-Thierry Lelégard, TSDuck author and maintainer

rodneyallan, 2021-03-03:
At International Datacasting Corporation (www.datacast.com) we have been using TSDuck for testing purposes. It has all the features we require and is easy to use.  It is true that you can create a plugin in less than a day! We very much appreciate the responsiveness of Thierry to issues that are brought up. We are also looking at incorporating these libraries into our products going forward. We are very impressed and hope to contribute more to the project in the future.

Feel free to use our logo.

lelegard, 2021-03-04:
@rodneyallan, thanks for the positive feedback

piotr-serafin-epam, 2021-03-30:
TSDuck is used at EPAM Systems (www.epam.com) as a part of streaming/testing infrastructure for one of our clients. Our system is based on shell scripts which are utilising TSDuck to feed [DTE-3114](https://www.dektec.com/products/IP/DTE-3114/) with test streams and distribute DVB-C signal to development CPEs. Entire solution is very stable (with couple of months of uptime) and working 24/7. Thierry was very helpful in the past and when requested to provide support for Source-Specific Multicast ([SSM](https://en.wikipedia.org/wiki/Source-specific_multicast)) it was matter of days before he [did](https://github.com/tsduck/tsduck/commit/315b9b4f65d43a55f363d49ee4496154db58ddf8). Thank you for TSDuck and hoping to contribute in future.

Feel free to use our logo.

lelegard, 2021-03-31:
@piotr-serafin-epam, thanks a lot for your feedback, especially concerning the stability in 24x7 mode.

piotr-serafin-epam, 2021-03-31:
> @piotr-serafin-epam, thanks a lot for your feedback, especially concerning the stability in 24x7 mode.

I've seen you have already updated reference page, please fix header to EPAM Systems (final s is missing currently)

lelegard, 2021-03-31:
> please fix header to EPAM Systems (final s is missing currently)

Sorry about that! Fixed, synchronization in process...

digitaltvguy, 2021-04-03:
We are using TSDuck at one of the major network broadcasters in the United States for ts stream capture and PCAP to TS extraction.  Thank you for a wonderful and unique tool.

lelegard, 2021-04-04:
@digitaltvguy, thanks for the feedback. Would you allow to be quoted on the [TSDuck references page](https://tsduck.io/references/)? In which case, we would need at least the name of the broadcaster.

digitaltvguy, 2021-04-04:
I’ll ask if I can.  I’d personally love to do it!

jdeisenh, 2021-05-18:
Exaring AG is an OTT service provider from germany. Our main product is waipu.tv. 
We use tsduck for development, debugging and for some special use cases in pre-production setups.
Beside the pre-production 24/7 use, we use it quite frequently, mainly for stream analysis of ingest feeds, and scte35 analysis and manipulation.
Main context is the signal ingest, pre transcoding side of our business, where we have MPEG-TS over IP multicast or TCP, and increasingly SRT. 
Tsduck is a really useful and powerfull tool, but not especially easy to grasp and to master. Even describing what it does is a challenge, because, like many useful tools, it has many different uses.
If you know your way around broadcast standards and work with MPEG TS streams, you need it.   
Thank you for this excellent tool.

lelegard, 2021-05-18:
Hi @jdeisenh, thanks a lot for your reference !

jherreromagic, 2021-05-19:
SmartHealth TV Solution is working on several projects related to Telecare, Telemedicine and Education on Smart TV with ISDB-T ATSC , and DVB networks using TSDuck. 

Thanks a lot for your effort, but you could consider also creating some "support or contribute" option, to help you to maintain the project with money.

lelegard, 2021-05-19:
Hi @jherreromagic, thanks a lot.

> you could consider also creating some "support or contribute" option, to help you to maintain the project with money.

The TSDuck site ([tsduck.io](https://tsduck.io/)) has a "[donate](https://tsduck.io/donate/)" link. But the contributions are informal for supporting the project. This is not a commercial support activity for TSDuck.

peter-stanton, 2022-08-18:
Ocean Blue Software is a software development company that specialises in middleware for TV/Set top box systems. We use TSDuck to create and edit test streams for use in debugging and QA of broadcast TV devices. 
Its ability to rewrite SI data from xml is the main feature we use. We maintain a set of python scripts to generate entirely custom TS files for testing and demos, using FFmpeg and TSDuck.
We like it for its wide coverage of DVB standards and its flexibility. Plus the support from TSDuck developers is very quick when we've needed it and it's very well documented.

lelegard, 2022-08-18:
Thanks @peter-stanton, I have added your comment in the references page on tsduck.io.

kalegarner, 2023-07-28:
I am currently using it to inject SCTE-35 on a regular basis successfully to a company called Syncbak/VUit. I have created a program that saves the xml injection on receiving a contact closure from our master control, it then adds up break times from a playlist in real time and then saves the XML file for injection in the multicast stream. It has been running since Feb of this year. Thank you for your commitment to open source brother!

kalegarner, 2023-07-28:
Do you support BAT tokens for support? Would regularly contribute what I have.

lelegard, 2023-07-28:
Hi @kalegarner 

Thanks for your feedback! May I mention Syncbak/VUit in the reference page on tsduck.io?

There is no paid support service for TSDuck. As any FOSS project, support is based on good will from the community of users. 

If you'd like to financially contribute, there is a "donate" page on tsduck.io using PayPal. We do not use BAT token or any other form of cryptocurrency.

pabloko, 2024-08-05:
Hi @lelegard heres my corpo style promo from what im allowed to tell:

In 2019, we successfully incorporated TSDuck into our DVB-T broadcast workflows, enhancing our service offerings to Spanish local television and radio stations through the addition of Electronic Program Guide (EPG), HbbTV (red button) applications, and recently, AI-generated subtitles. Since its integration, TSDuck has operated continuously without any interruption or failures, demonstrating remarkable reliability and efficiency, delivering said services to the living room of tens of thousands of people daily.

Furthermore, TSDuck has emerged as our preferred tool for analysis and debugging of DVB streams. Its comprehensive feature set outshines many commercial professional tools we have evaluated, establishing TSDuck as an indispensable asset in our operational framework.

. Pablo García (if you need to add a company you can list interalmeria.tv)