tsduck/tsduck #731: [FEAT] Compile options to skip building other language bindings

rubenochiavone, 2021-03-02:
#### Is your feature request related to a problem?

No.

#### Proposed solution:

Add make configuration variables so we can toggle whether to build other languages bindings or not.

NOPYTHONBINDINGS : Do not build python bindings
NOJAVABINDINGS : Do not build java bindings

By adding adding those options to the project, people that won't use those languages bindings would have faster builds.

lelegard, 2021-03-02:
> By adding adding those options to the project, people that won't use those languages bindings would have faster builds.

Not really.

Using `cloc`, we can see that TSDuck contains approximately 200,000 lines of pure code (no blank, no comment) or 370,000 lines overall. The C++ code for Java and Python bindings uses 2,000 lines of pure code, thus representing 1% of the code base. So, I do not see any real improvement in compilation speed.

The compilation options such as `NOCURL` or `NOSRT` were only created to remove dependencies to external libraries which could sometimes create problems. They are not made to reduce the compilation time (the corresponding modules are still compiled with `#ifdef` branches compiling code stubs).

Please realize that this kind of option makes the code more complex to read, to maintain and to test all combinations. If you ask this for 1% of code, DVB people will ask for `NOATSC NOISDB`. ATSC and ISDB people will ask for their own, etc. I do not want to enter this nightmare without rational reason. Avoiding external dependencies to third party libraries which may not be present on a given platform is a very valid technical reason. The impatience of the developer is not 😄 

Arguably, the only compilation speed-oriented option is `NOTEST` because unitary tests are very long lists of complex macro assertions which are much slower to compile than any other kind of modules.

The only really efficient way of speeding up the build is to use `make -j10` (or `-j20`, or more, depending on your CPU and memory). The TSDuck makefiles were carefully crafted to support unlimited parallel compilations. My `make -j10` usually takes 2 to 3 minutes on a quad-core i7 iMac from 2015. Not that bad...

Of course, using `make -j2` (because of the small memory) on my Rasberry Pi takes ages. But I don't think we can consider this as a typical development platform. I use it only to generate final ARM releases.

rubenochiavone, 2021-03-02:
Sounds fair. Closing this one. Thanks for your support.

closed 2021-03-02 by rubenochiavone