tsduck/tsduck #732: On new versions of TSDuck (> 3.19-1516) tstables utility generates XML output EIT whit generic_long_tables if some events exceed max length of short_event_descriptor.

JuankGF, 2021-03-02:
Hi, I'm triying to get XML output for an Event Programming Guide (EPG) and detect which events in EPG services has incorrect short_event_descriptors. I tested in recent and old versions of TSDuck and the output of (tstables test.ts --pid 18 --fill-eit --xml-output All_EIT_Out.xml) produces diferent outputs trhroug diferent versions. 
My observation is, like in others DVB TS inspect tools, why not make the maxim effor to show where the problem is? In older version TSDuck extract XML and show generic_descriptors in some events that hasn´t correct structure of short_event_descriptors. Example:
<event event_id="0x0001" start_time="2021-01-17 05:20:03" duration="01:47:35" running_status="not-running" CA_mode="false">
      <short_event_descriptor language_code="eng">
        <event_name>TIEM</event_name>
        <text></text>
      </short_event_descriptor>
      <component_descriptor stream_content="0x00" stream_content_ext="0x02" component_type="0x44" component_tag="0x45" language_code=" CI" text="NE :  MAD MAX 3. MAS ALLA DE LA CUPULA DEL TRUENOS̆inopsis: Tras la catas"/>
      <generic_descriptor tag="0x74">
        6F 66 65 20 6E 75 63 6C 65 61 72 2C 20 4D 61 64
        20 4D 61 78 20 63 72 75 7A 61 20 75 6E 20 64 65
        73 69 65 72 74 6F 20 64 6F 6E 64 65 20 70 69 65
        72 64 65 20 73 75 20 63 61 72 61 76 61 6E 61 20
        64 65 20 63 61 6D 65 6C 6C 6F 73 2E 20 4C 6C 65
        67 61 20 61 20 75 6E 61 20 63 69 75 64 61 64 20
        64 6F 6E 64 65 20 6C 65 20 70 72 6F 70 6F 6E 65
        6E 20
      </generic_descriptor>
    </event>

lelegard, 2021-03-02:
> My observation is, like in others DVB TS inspect tools, why not make the maxim effor to show where the problem is? 

I would really be curious to see how other tools would display this specific binary content.
I would also be curious to see what previous versions of TSDuck displayed for this specific binary content.

I tried to cut/paste your XML, compiled it and reanalysed it. It gives this:

~~~
    - Descriptor 0: Short Event (0x4D, 77), 9 bytes
      Language: eng
      Event name: "TIEM"
      Description: ""
    - Descriptor 1: Component (0x50, 80), 78 bytes
      Content/type: 0x2044 (unknown)
      Component tag: 69 (0x45)
      Language:  CI
      Description: "NE : MAD MAX 3. MAS ALLA DE LA CUPULA DEL TRUENOS̆inopsis: Tras la catas"
    - Descriptor 2: Related Content (0x74, 116), 114 bytes
      Extraneous 114 bytes:
      0000:  6F 66 65 20 6E 75 63 6C 65 61 72 2C 20 4D  ofe nuclear, M
      000E:  61 64 20 4D 61 78 20 63 72 75 7A 61 20 75  ad Max cruza u
      001C:  6E 20 64 65 73 69 65 72 74 6F 20 64 6F 6E  n desierto don
      002A:  64 65 20 70 69 65 72 64 65 20 73 75 20 63  de pierde su c
      0038:  61 72 61 76 61 6E 61 20 64 65 20 63 61 6D  aravana de cam
      0046:  65 6C 6C 6F 73 2E 20 4C 6C 65 67 61 20 61  ellos. Llega a
      0054:  20 75 6E 61 20 63 69 75 64 61 64 20 64 6F   una ciudad do
      0062:  6E 64 65 20 6C 65 20 70 72 6F 70 6F 6E 65  nde le propone
      0070:  6E 20                                      n
~~~

If you think like a human knowing your context and knowing what you expect, you will guess that the length of the initial short event descriptor is incorrect and that the rest is just text that should be part of it. But, if you think like a DVB parser, it's synchronous. There are trailing bytes in the supposed related content descriptor, making it incorrect. So, the incorrect descriptor is the "related content" one, not the short event. Otherwise, how would you guess were the error is? Yes, the descriptor tag 114 and descriptor length 114 are in fact ASCII letters 'tr' from the word 'catastrofe'.

The descriptor is incorrect. So, it cannot be converted. Using the text display, the analysis tries to make it somehow readable. But the XML must be a compilable representation. It cannot be partial. If a section or descriptor is not recognised, it is generated in the generic form.

JuankGF, 2021-03-02:
Thanks for your answer.
I'll be more especific. I'm getting de input TS from a real DVB transmision and then using TSDuck to extract EITs to XML. I understand that DVB parser try to match received data with specific standar structures like descriptors. But in the example I put before the output is generated by TSDuck v.3.19-1516. In recent versions of the tool, the same experiment with the same data shows diferent output (only showing a generic_long_table for the EIT that contain such errors). 
In those cases we lost precision in where is the error and wich events has wrong data. Because we only get a generic_long_table with many sections of hexadecimal values.
My answer is in this case, Why older versions change the content displayed for the same data?
I ask this because not all events in EIT generated were wrong formed, and when tstables utility exports XML in recent versions we lost the oportunity to parse those events. In the code below we see an entire EIT converted to generic_long_table with newer versions of TSDuck loosing all correct events.

`<generic_long_table table_id="0x50" table_id_ext="0x0005" version="4" current="true" private="true">
    <metadata PID="18" first_ts_packet="0" last_ts_packet="621"/>
    <section>
      00 01 00 64 00 51
    </section>
    <section>
      00 01 00 64 00 51
    </section>
    <section>
      00 01 00 64 00 51
    </section>
    <section>
      00 01 00 64 00 51
    </section>
    <section>
      00 01 00 64 00 51
    </section>
    <section>
      00 01 00 64 00 51
    </section>
    <section>
      00 01 00 64 00 51
    </section>
    <section>
      00 01 00 64 00 51
    </section>
    <section>
      00 01 00 64 08 51 00 01 E7 5F 05 20 03 01 47 35
      21 0A 4D 08 65 6E 67 3D 54 49 45 4D 50 4F 20 44
      45 20 43 49 4E 45 20 3A 20 20 4D 41 44 20 4D 41
      58 20 33 2E 20 4D 41 53 20 41 4C 4C 41 20 44 45
      20 4C 41 20 43 55 50 55 4C 41 20 44 45 4C 20 54
      52 55 45 4E 4F C6 53 69 6E 6F 70 73 69 73 3A 20
      54 72 61 73 20 6C 61 20 63 61 74 61 73 74 72 6F
      66 65 20 6E 75 63 6C 65 61 72 2C 20 4D 61 64 20
      4D 61 78 20 63 72 75 7A 61 20 75 6E 20 64 65 73
      69 65 72 74 6F 20 64 6F 6E 64 65 20 70 69 65 72
      64 65 20 73 75 20 63 61 72 61 76 61 6E 61 20 64
      65 20 63 61 6D 65 6C 6C 6F 73 2E 20 4C 6C 65 67
      61 20 61 20 75 6E 61 20 63 69 75 64 61 64 20 64
      6F 6E 64 65 20 6C 65 20 70 72 6F 70 6F 6E 65 6E
      20 63 61 6D 62 69 61 72 73 65 6C 6F 73 20 61 20
      63 61 6D 62 69 6F 20 64 65 20 71 75 65 20 61 74
      61 71 75 65 20 61 6C 20 74 69 72 61 6E 6F 20 64
      65 20 6C 61 20 63 69 75 64 61 64 2E
    </section>
    <section>
      00 01 00 64 08 51
    </section>
    <section>
      00 01 00 64 08 51
    </section>
    <section>
      00 01 00 64 08 51
    </section>
    <section>
      00 01 00 64 08 51
    </section>
    <section>
      00 01 00 64 08 51
    </section>
    <section>
      00 01 00 64 08 51
    </section>
[... 1843 lines cut ...]
      45 20 43 45 52 43 41 20 41 20 53 55 20 47 45 4E
      54 45 2E 0A 0A 00 82 E7 62 22 09 29 00 24 42 20
      59 4D 57 65 6E 67 12 55 54 49 4C 49 53 49 4D 4F
      20 3A 20 49 64 65 61 73 20 40 54 65 6D 61 3A 20
      4C 61 6D 70 61 72 61 73 20 64 65 20 6C 61 74 61
      73 2C 20 6D 61 63 65 74 61 73 20 64 65 20 63 65
      6D 65 6E 74 6F 20 79 20 61 72 72 65 67 6C 6F 73
      20 66 6C 6F 72 61 6C 65 73 0A 00 83 E7 62 22 34
      11 00 24 45 20 33 4D 31 65 6E 67 09 45 75 72 6F
      6D 61 78 78 20 23 54 69 74 75 6C 6F 3A 20 4F 62
      72 61 73 20 61 72 74 69 73 74 69 63 61 73 20 65
      78 63 6C 75 73 69 76 61 73 00 84 E7 62 22 58 56
      00 42 03 20 52 4D 50 65 6E 67 0A 46 41 43 49 4C
      49 53 49 4D 4F 41 46 6C 6F 6F 67 61 74 73 20 A8
      45 6C 20 72 65 6C 6F 6A A8 2C 20 4E 61 74 20 47
      65 6F 20 4C 61 62 2C 20 50 6F 77 65 72 20 70 6C
      61 79 65 72 73 20 A8 54 72 61 6D 70 61 20 64 65
      20 61 72 65 6E 61 A8 00 85 E7 62 23 40 59 00 25
      34 20 48 4D 46 65 6E 67 27 53 45 52 49 45 20 49
      4E 46 41 4E 54 49 4C 20 3A 20 41 56 41 54 41 52
      20 20 20 20 20 20 20 54 45 4D 50 20 32 20 20 20
      1A 43 41 50 20 31 30 20 20 20 20 4C 61 20 62 69
      62 6C 69 6F 74 65 63 61 20 0A 0A
    </section>
  </generic_long_table>`

lelegard, 2021-03-02:
It must be a different table.

Don't forget that XML is a _compilable_ format format for TSDuck. It must be correct. If a table is incorrect, it cannot be represented as XML. The generic section or descriptor is just a generic way to keep incorrect data compilable, to regenerate the original binary. Only the text format (not compilable, not reversible) can give you some information or hints on the error. At least if the binary is not too corrupted.