tsduck/tsduck #733: [Q] HLS and EIT (EPG data)

infoviaplus, 2021-03-04:
Hello!

We are trying to convert some DVB-T channels difficult to being received here by air from a neighbour region into separate HLS streams with exactly the same capabilities as the original DVB-T services (audio, video, subtitles, EPG...). And the last one is giving us a hard time. 

We get an MPEG TS of every channel through a Tvheadend server. VLC opens these HTTP Transport Streams flawlessly, and reads all contents (EPG, subtitles...). Tvheadend doesn't offer HLS output, so we tried with ffmpeg's HLS encoder. But, as you all know and TSDuck's manual warns, ffmpeg removes EIT and other mux data (and even has problems simply copying DVB subtitles). 

With TSDuck, a/v work, subtitles work and program name is read by client (VLC Media Player, as said before). But EPG keeps missing. I tried two command lines. First one simply pipes the TS into TSDuck's HLS encoder: 

` tsp -I http "http://our.stream.ts" | tsp -v -O hls /var/www/c001_.ts -l 10 -p /var/www/c01.m3u8 `

Second one "zaps" a specific Service ID (the only one in the HTTP TS, so is a bit redundant) trying to force EIT preservation, then pipes to HLS encoder:

` tsp -I http "http://our.stream.ts" | tsp -P zap --eit 805 | tsp -v -O hls /var/www/c001_.ts -l 10 -p /var/www/c01.m3u8 `

I read through the manual and I think that, if it's possible to keep EIT when converting from "live" TS to HLS, I tried all the ways. 

Can you give me some advice or tell me why it seems impossible to keep EIT into HLS output (not only with TSDuck)? 

Many thanks in advance, and in any case, congrats por this powerful piece of software.

lelegard, 2021-03-04:
Hi @infoviaplus 

First, please, don't use `tsp -I ... | tsp -P ... | tsp -P ... | tsp -O ...`! This is extremely inefficient. The transport stream goes through many pipes and packets are copied back and forth all the time.

The right way is `tsp -I ... -P ... -P ... -O ...`. There is one single `tsp` common and one single common packet buffer where packets are processed "in place" and are never copied.

Thant being said, your command preserves EIT's indeed, if there are any for the given service. I just tried `tsp -I file ... -P zap --eit 0x1392 -O hls test.ts -l 8` on a broadcast capture file what was hanging out somewhere on the disk and I confirm that the EIT's are present in the output segment files for the selected service.

First, check that EIT's are present in some output HLS segment files: `tstables c001_XXX.ts --pid 18`. It EIT's are present, then maybe VLC does not read EIT's from HLS input. EIT's are a DVB concept and HLS is used for OTT. Maybe VLC does not try to read EIT's in HLS streams.

Second, if you don't find EIT's in the output HLS segment files, make sure that there are EIT's for _that_ service in the input stream: `tsp -I http "http://our.stream.ts" -P tables --pid 18 --tid 0x4E-0x6F --tid-ext 805 -O drop` will display all EIT's for service 805.

infoviaplus, 2021-03-04:
Hi again,  @lelegard. 

First of all, thanks for the quick response and for the notes about piping. I fixed my command lines to use only one `tsp`. I'm not a huge fan of piping, so making everything with one instance is perfect. 

About EIT, I think I found the problem. As you said, there's EIT info (at least, the current/next events) at HLS fragments. But this information isn't placed where it should be. I'll ilustrate this with an example. 

This is the output of `ffmpeg -i "http://our.stream.ts"` (slightly shorted and adapted to GitHub editor):

> Input #0, mpegts, from 'http://our.stream.ts':
&nbsp; Program 805
    &nbsp; &nbsp; Metadata:
     &nbsp; &nbsp; &nbsp;  service_name    : (NAME)
     &nbsp; &nbsp; &nbsp;  service_provider: (PROVIDER)
    &nbsp; &nbsp; Stream #0:1(0x81) (spa): Subtitle: dvb_teletext ([6][0][0][0] / 0x0006)
    &nbsp; &nbsp; Stream #0:2(0xd3): Video: h264  [...] 1920x1080 [SAR 1:1 DAR 16:9], 25 fps, 50 tbr, 90k tbn, 50 tbc
    &nbsp; &nbsp; Stream #0:3(0xd4) (spa): Audio: eac3 ([6][0][0][0] / 0x0006), 48000 Hz, stereo, fltp, 256 kb/s
    &nbsp; &nbsp; Stream #0:4(0xd6) (qaa): Audio: eac3 ([6][0][0][0] / 0x0006), 48000 Hz, stereo, fltp, 128 kb/s
    &nbsp; &nbsp; Stream #0:5(0xd8) (spa): Audio: eac3 ([6][0][0][0] / 0x0006), 48000 Hz, mono, fltp, 64 kb/s (visual impaired) (descriptions)
    &nbsp; &nbsp; Stream #0:6(0x325) (spa): Subtitle: dvb_subtitle ([6][0][0][0] / 0x0006)
    &nbsp; &nbsp; Stream #0:7(0x32f) (qaa): Subtitle: dvb_subtitle ([6][0][0][0] / 0x0006)
  &nbsp; No Program
  &nbsp; &nbsp;   Stream #0:0[0x12]: Data: epg

And this is the output of `ffmpeg -i "http://our.server.with.tsduck/01.m3u8"` (the previous TS converted to HLS): 

>Input #0, hls, from '01.m3u8':
&nbsp;Duration: N/A, start: 89434.629367, bitrate: N/A
&nbsp;Program 0
&nbsp;&nbsp;Metadata:
&nbsp;&nbsp;&nbsp;variant_bitrate : 0
&nbsp;&nbsp;Stream #0:0: Subtitle: dvb_teletext ([6][0][0][0] / 0x0006), 492x250
&nbsp;&nbsp;Metadata:
&nbsp;&nbsp;&nbsp;variant_bitrate : 0
&nbsp;&nbsp;Stream #0:1: Video: h264 [...] 1920x1080 [SAR 1:1 DAR 16:9], 50 fps, 50 tbr, 90k tbn, 50 tbc
&nbsp;&nbsp;Metadata:
&nbsp;&nbsp;&nbsp;variant_bitrate : 0
&nbsp;&nbsp;Stream #0:2: Audio: eac3 ([6][0][0][0] / 0x0006), 48000 Hz, stereo, fltp, 256 kb/s
&nbsp;&nbsp;Metadata:
&nbsp;&nbsp;&nbsp;variant_bitrate : 0
&nbsp;&nbsp;Stream #0:3: Audio: eac3 ([6][0][0][0] / 0x0006), 48000 Hz, stereo, fltp, 128 kb/s
&nbsp;&nbsp;Metadata:
&nbsp;&nbsp;&nbsp;variant_bitrate : 0
&nbsp;&nbsp;Stream #0:4: Audio: eac3 ([6][0][0][0] / 0x0006), 48000 Hz, mono, fltp, 64 kb/s
&nbsp;&nbsp;Metadata:
&nbsp;&nbsp;&nbsp;variant_bitrate : 0
&nbsp;&nbsp;Stream #0:5: Subtitle: dvb_subtitle ([6][0][0][0] / 0x0006)
&nbsp;&nbsp;Metadata:
&nbsp;&nbsp;&nbsp;variant_bitrate : 0
&nbsp;&nbsp;Stream #0:6: Subtitle: dvb_subtitle ([6][0][0][0] / 0x0006)
&nbsp;&nbsp;Metadata:
&nbsp;&nbsp;&nbsp;variant_bitrate : 0
&nbsp;&nbsp;Stream #0:7: Data: epg
&nbsp;&nbsp;Metadata:
&nbsp;&nbsp;&nbsp;variant_bitrate : 0

As you can see, TSDuck gets the EIT data and adds it as a new stream **inside** the program, instead of putting it in the TS but out of the program streams (a/v/subs...). I bet it all on this being the cause of EPG not being recognized. 

I am 90% sure that, a long time ago, I saw somewhere an HLS stream (m3u8) with EPG/EIT included in it. Obviously I've no idea of the software used to create it. 

After all of this, I have a pair of questions: 

1. It would be possible to cut the TS in fragments "as it comes", without remapping streams (EIT particularly)? (maybe TSDuck have or could have some kind of switch for this?)
2. Do you have knowledge of apps/devices that are capable of reading the EPG after being remapped by TSDuck's HLS encoder? I'm not totally sure but my bet is that, although EPG data is kept, none or a few clients are capable of decoding it if EIT isn't in the right place. 

Thanks again!

lelegard, 2021-03-04:
> 1. It would be possible to cut the TS in fragments "as it comes", without remapping streams (EIT particularly)? (maybe TSDuck have or could have some kind of switch for this?)
> 2.  Do you have [...]

TSDuck does not remap EIT's. The `hls` plugin does _nothing_ on EIT, nothing at all. EIT's are DVB tables and DVB leaves no room for EIT remapping. EIT's must be on PID 18 and this PID is "global" to the TS, it is not attached to any service. This is the standard, you cannot change it and TSDuck does not change this.

I do not know how ffmpeg structures its output. It is generic to all kinds of transport. TS is just one of many transport formats for ffmpeg and certainly not the one it handles the best. The output does not show the TS structure of the "streams".

If you use TSDuck tools `tsanalyze` and ` tstables` (or their plugin counterparts `analyze` and ` tables` when used in a `tsp` command), you will see that EIT's are always in PID 18, in the original MPTS input as well as in the output HLS segments. That is the way it should be done.

If VLC requires some other structure to grab EIT's from a TS stream coming from HLS, then if you can describe exactly what VLC is expecting, we may find a way to build it using other plugins.

As a final word, there is no such thing as an "HLS encoder". HLS is nothing more than pieces of TS which are referenced in a play list. The `hls` plugin does not do much. It is just a "pair of scissors" that cuts the input stream in pieces _without any modification_ in the TS. The only piece of "intelligence" in the `hls` plugin is to find clever places where to cut the segments and to build the playlist text file. Nothing very clever in fact....

infoviaplus, 2021-03-04:
Hello again, @lelegard. 

Sorry about my misconception based on the above pasted output. I'll give ffmpeg's opinion about Transport Streams less value in the future (seems that, after years working with it, ffmpeg's fooled me). 

I know that HLS output plugin is not an "encoder" in a strict sense, but I supposed that was a straightforward way of calling it. 

I digged more in this painful EIT issue, with the tools you mentioned and with VLC Media Player itself by turning on debug log (Tools > Messages > level 2: debugging). 

To my extreme facepalm, although EIT events don't appear in the OSD or in their specific window (Tools > Program Guide), they're dumped into the debug log. But there's something wrong: when the EIT dump starts, this message apears dozens of times and seems the key of the problem: 

> ts error: libdvbpsi error (EIT decoder): 'last_section_number' differs whereas no discontinuity has occured

Just in case it's useful, this appears when VLC starts the (failed) EIT decoding, just before the list of events mixed with multiple copies of the above error message: 

> ts debug: EITCallBack called
> ts debug: new EIT service_id=866 version=22 current_next=1 ts_id=99 network_id=8916 segment_last_section_number=1 last_table_id=78

In short: I guess that some inconsistency with "last_section_number" is the origin of the incompatibilities, although I don't know if VLC is too strict complaining about that. 

Could you give me some guidance about that "last_section_number" message? 

Thanks a lot for your help.

lelegard, 2021-03-05:
> I'll give ffmpeg's opinion about Transport Streams less value in the future (seems that, after years working with it, ffmpeg's fooled me).

I do not want to criticize ffmpeg. This is a wonderful tool. It is extremely powerful. Its command line / filter chain structure has influenced the TSDuck design. Its processing of video and audio content in invaluable. But, it has one small defect: the way it processes the MPEG-TS container format is notoriously poor. This is why ffmpeg and TSDuck are often used together in TV systems: ffmpeg processes the video/audio content and TSDuck processes the system layer.

> In short: I guess that some inconsistency with "last_section_number" is the origin of the incompatibilities, although I don't know if VLC is too strict complaining about that.

I do not see anything strange about the last_section_number. As you probably know, this is a generic MPEG section format field. Per DVB compliance, EIT present/following tables always have 2 sections (numbered 0 and 1), hence last_section_number is set to 1 in all EIT p/f sections (table id 78 is "EIT p/f Actual"). The first section contains the "present" event. The second section contains the "following" event.

The continuous HLS stream (i.e. the concatenation of all segments) at the output of the `hls` plugin is exactly identical to the input TS stream at the input of this plugin (except an additional copy or the PAT and PMT at the beginning of each segment). 

There are two possible reasons: Either these EIT's are incorrect from the beginning or VLC does not process the EIT's the same way depending on the input being segmented HLS or continuous TS.

To check this, I suggest to concatenate all HLS segments in one stream (`cat c001_*.ts >file.ts` is enough) and run VLC on that file. Does it exhibit the same error?

lars18th, 2021-03-05:
Hi @infoviaplus ,

I've a lot of experience with ffmpeg and tsduck. So let me clarify a few points that I suspect you have misinterpreted:
- You want to use HLS as a transport layer carrying a regular MPEG2-TS. OK, but then any pids in the original stream _can_ be inside the HLS chunks. That includes EIT and any other PSI/SI tables. And the TSDuck is not removing or filtering any EIT data. So if the source has EIT data, then the output will have it.
- The ffmpeg tool is constructed using modules. So the HLS reader is different from the MPEGTS reader. Even if the HLS contains a MPEG-TS. So for this reason you see different outputs about the content of the file when consuming from the `our.stream.ts` or the `01.m3u8` files. So don't think that TSDuck is changing something.
- The videolan tool perhaps uses a similar approach and uses different HLS reader. From my experience the vlc project is very dirty in some points (even I love it). So I suspect it doesn't want to read EPG from a HLS stream.

Having said this, I want to comment my suggestion for your problem:
- Your use case is to transmit some SPTS (from a MPTS) over the Internet, right?
  - In this case you've two good options to carry the service: SRT or HLS. The first is stream oriented and the second is chunk oriented. First uses UDP and the second TCP. Both work very reliable, and the selection of wich one can be based on your network environment. Futhermore, both are suported in TSDuck.
 - In your use case the source is a TV backend (tvheadend) that provies the service with one stream over one HTTP uri.
 - And the target is... what? A client device? Another TV backend? A roaming user? In different scenerarios you need different solutions:
   -  For example, for a roaming final client using an App (for example VLC in Android) perhaps you need a server that can provide _compatible_ HLS streams. In this case you need to understand if the client App supports what you want (for example EPG data). The current answer is that perhaps videolan doesn't support EPG data over HLS.
   - Another option is to use a transparent layer. In this case you can use TSDuck (or another tool) to translate the HLS stream to a regular MPEG2-TS. And then inject the output to the player. For example in this way:
     - TVH -->  MPEG-TS --> TSDuck --> HLS  ........../Internet/..........  HLS --> TSDuck --> MPEG-TS --> VLC 
     - In the client part you can do something like `tsp -I http http://our.server.with.tsduck/01.m3u8 | vlc -`

However, if you want to use this approach of TSDuck source to HLS and TSDuck HLS reader to MPEG-TS pipes, you'll need to do more tests. I'm not sure if this will work well on-demand. Surely, for continuous streaming it will do, and you just have to be careful with the delay.

Regards.

closed 2026-04-26 by github-actions[bot]